PX4/PX4-Autopilot #23406: Unable to connect Raspberry Pi 4 to mRo Pixhawk flight controller via uXRCE-DDS wired serial connection

Geol-23, 2024-07-15:
### Describe the bug

Hello,
I'm using a Raspberry Pi 4 (Ubuntu 22.04) to communicate with an mRo Pixhawk ( PX4 1.14.3). I've followed the [guide](https://docs.px4.io/main/en/companion_computer/pixhawk_rpi.html) and installed the agent as a standalone on the Pi, I've used the reccommended baud rates and ports (921600, TELEM2 on the FC and UART0 on the Pi), I believe the pins are the correct way around and the wire connections are solid. I've also added the dialout permissions and I'm pretty sure nothing else is running on that port.

To diagnose this I've tried to boot up a seperate FC (same version as the main FC) and the issue persists. I've established communication using mavproxy.py according to the [guide](https://docs.px4.io/main/en/companion_computer/pixhawk_rpi.html). I've also used `screen` to take a look at the data going through /dev/ttyAMA0 and there is activity (albeit it's just invalid characters).

## Logs
#### Raspberry Pi:
`MicroXRCEAgent serial --dev /dev/ttyUSB0 -b 921600 -v 6`
`[1709157626.383849] info | TermiosAgentLinux.cpp | init | running... | fd: 3`
`[1709157626.385726] info | Root.cpp | set_verbose_level | logger setup | verbose_level: 6`

#### Flight Controller
`nsh> uxrce_dds_client status`
`INFO [uxrce_dds_client] Running, disconnected`
`INFO [uxrce_dds_client] Using transport: serial`

I'm still new to the Pixhawk environment so any help is appreciated.
Cheers.

### To Reproduce

1. Install the agent standalone the RPi and configure UART0 per the guide
2. Configure the baud rate (921600), port (TELEM2) and leave everything else default
3. Reboot FC as needed and power it through the micro-usb port to your ground station (I'm using QGroundControl)
4. Assuming the two are wired together and since the client is already running on the FC start the agent using this command `MicroXRCEAgent serial -b 921600 -D /dev/ttyAMA0 -v 6`

### Expected behavior

The guide states that there should be some output on the RPi terminal and the FC terminal but nothing changes for me. When I check the status of the client, it's running but not connected. Same with the output on the terminal. 

### Flight Log

N.A.

### Software Version

1.14.3

### Flight controller

mRo Pixhawk 

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

Using TELEM2 on the FC and the RX, TX and ground pins on the RPi pins 10, 8, 6 respectively.

Geol-23, 2024-07-16:
I've done some more investigating, using a multimeter in AC mode I read the voltage across the TX and GND pin on the FC and got a voltage reading however when I read the voltage across the TX and (any) GND pin on the RPi I read no voltage. I'm confident I installed the agent correctly and set up the RPi correctly but for thoroughness heres my `config.txt ` file:

`[all]`
`kernel=vmlinuz`
`cmdline=cmdline.txt`
`initramfs initrd.img followkernel`

`[pi4]`
`max_framebuffers=2`
`arm_boost=1`

`[all]`
`# Enable the audio output, I2C and SPI interfaces on the GPIO header. As these`
`# parameters related to the base device-tree they must appear *before* any`
`# other dtoverlay= specification`
`dtparam=audio=on`
`dtparam=i2c_arm=on`
`dtparam=spi=on`

`# Comment out the following line if the edges of the desktop appear outside`
`# the edges of your display`
`disable_overscan=1`

`# If you have issues with audio, you may try uncommenting the following line`
`# which forces the HDMI output into HDMI mode instead of DVI (which doesn't`
`# support audio output)`
`#hdmi_drive=2`

`[cm4]`
`# Enable the USB2 outputs on the IO board (assuming your CM4 is plugged into`
`# such a board)`
`dtoverlay=dwc2,dr_mode=host`

`[all]`

`# Enable the KMS ("full" KMS) graphics overlay, leaving GPU memory as the`
`# default (the kernel is in control of graphics memory with full KMS)`
`dtoverlay=vc4-kms-v3d`

`# Autoload overlays for any recognized cameras or displays that are attached`
`# to the CSI/DSI ports. Please note this is for libcamera support, *not* for`
`# the legacy camera stack`
`camera_auto_detect=1`
`display_auto_detect=1`

`# Config settings specific to arm64`
`arm_64bit=1`
`dtoverlay=dwc2`
`enable_uart=1`
`dtoverlay=disable-bt`

Geol-23, 2024-07-26:
Still having the issue, though with more testing I've managed to narrow it down to a software issue on the Raspberry Pi. Using a feedback loop across the TX and RX pins on the Pi, I've managed to send and recieve characters over the serial port. Unfortunately I still can't get the flight controller to communicate with the Raspberry Pi.

I've also tried installing the Agent as a ROS2 package and the issue still persits. However something has changed. With the two disconnected and reading the voltage at the TX and RX pins on the Pi they both read 3.3V. I'm pretty sure the UART protocol is that the TX pin is high by default but would it matter if the RX pin 3.3V?

Also using `screen` or `minicom` and the same feedback loop as before, no characters appear on the terminal when running the agent. This tells me that the agent isn't attempting to handshake with the with the client.

XXLiu-HNU, 2024-09-01:
Maybe you need to modify` theuxrce_dds_client ` parameter in QGC