PX4/PX4-Autopilot #23480: [Bug] sitl_multiple_run for gazebo-classic starts with wrong mav_sys_id

av-jeroen, 2024-08-01:
### Describe the bug

When executing the SITL multiple drones bash script for gazebo-classic multiple drones are being deployed. The Mavlink system Ids should start at 1 and increase for every drone. Instead it starts at 2. 

Before 1.14 this was the way it worked, and I belief this is the way people will expect it to work. 

### To Reproduce

The bug is visible when running with one drone:
```bash
./Tools/simulation/gazebo-classic/sitl_multiple_run.sh
```

But also with multiple
```bash
./Tools/simulation/gazebo-classic/sitl_multiple_run.sh -n 3
```

### Expected behavior

Expected: The MAV_SYS_ID should be 1. 
Actual: The MAV_SYS_ID is is 2. 

### Flight Log

Can't connect, so no log. 

### Software Version

1.14.0 and up

### Flight controller

SITL

### Vehicle type

Multicopter

av-jeroen, 2024-08-02:
Solved by changing the following in `./Tools/simulation/gazebo-classic/sitl_multiple_run.sh`

```
spawn_model ${vehicle_model} $(($n + 1))
```

into

```
spawn_model ${vehicle_model} $n
```

Not sure if this was intended behaviour, maybe certain developers want to run this next to a SIH setup? 

I'll close it for now.

Jaeyoung-Lim, 2024-08-02:
@av-jeroen Please consider creating a PR if you think this is a modification needed to improve the repository.