PX4/PX4-Autopilot #23493: [Bug] Multiple drones MAVROS topic /drone_1/mavros/local_position/odom all to 0

hcws, 2024-08-05:
### Describe the bug

Multiple drones are being simulated using Gazebo. After starting the simulation environment, the coordinates of each drone are all set to 0. However, when I move each drone, their coordinates change. In Gazebo, the coordinates of each drone are different.

### To Reproduce

    1. Drone switched on : in the begining
    Uploaded mission : QGC4.2.8
    Took off  : No
    See error ： No

### Expected behavior

```
/drone_0/mavros/local_position/odom
/drone_1/mavros/local_position/odom
/drone_2/mavros/local_position/odom
```
should diffent.
[image]

### Screenshot / Media

[image]
[image]

### Flight Log

```
 [ INFO] [1722869475.724269800]: Plugin hil initialized
[ INFO] [1722869475.724361343]: Plugin home_position loaded
[ INFO] [1722869475.727141823]: Plugin home_position initialized
[ INFO] [1722869475.727230050]: Plugin imu loaded
[ INFO] [1722869475.730212494]: FCU URL: udp://:14542@localhost:14582
[ INFO] [1722869475.732204695]: udp0: Bind address: 0.0.0.0:14542
[ INFO] [1722869475.732263324]: udp0: Remote address: 127.0.0.1:14582
[ INFO] [1722869475.732295631]: GCS bridge disabled
[ INFO] [1722869475.737673960]: Plugin setpoint_position initialized
[ INFO] [1722869475.737812448]: Plugin setpoint_raw loaded
[ INFO] [1722869475.738241055]: Plugin 3dr_radio loaded
[ INFO] [1722869475.746235224]: Plugin 3dr_radio initialized
[ INFO] [1722869475.746324686]: Plugin actuator_control loaded
[ INFO] [1722869475.751467767]: Plugin actuator_control initialized
[ INFO] [1722869475.752017032]: Plugin imu initialized
[ INFO] [1722869475.752096765]: Plugin landing_target loaded
[ INFO] [1722869475.754096671]: Plugin adsb loaded
[ INFO] [1722869475.758508787]: Plugin adsb initialized
[ INFO] [1722869475.758599494]: Plugin altitude loaded
[ INFO] [1722869475.760336182]: Plugin altitude initialized
[ INFO] [1722869475.760403420]: Plugin cam_imu_sync loaded
[ INFO] [1722869475.761186163]: Plugin cam_imu_sync initialized
[ INFO] [1722869475.761330140]: Plugin camera loaded
[ INFO] [1722869475.763151808]: Plugin camera initialized
[ INFO] [1722869475.763336496]: Plugin cellular_status loaded
[ INFO] [1722869475.765040971]: Plugin setpoint_raw initialized
[ INFO] [1722869475.765176998]: Plugin setpoint_trajectory loaded
[INFO] [1722869475.766340, 0.000000]: Loading model XML from ros parameter model_description
[ INFO] [1722869475.767032424]: Plugin cellular_status initialized
[ INFO] [1722869475.767153522]: Plugin command loaded
[ INFO] [1722869475.768538046]: Plugin landing_target initialized
[ INFO] [1722869475.768636756]: Plugin local_position loaded
[INFO] [1722869475.780229, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
[ INFO] [1722869475.781503332]: Plugin setpoint_trajectory initialized
[ INFO] [1722869475.781651856]: Plugin setpoint_velocity loaded
[ INFO] [1722869475.785055308]: Plugin local_position initialized
[ INFO] [1722869475.785169021]: Plugin log_transfer loaded
[ INFO] [1722869475.785701409]: Plugin command initialized
[ INFO] [1722869475.785799156]: Plugin companion_process_status loaded
[ INFO] [1722869475.786105872]: Plugin setpoint_velocity initialized
[ INFO] [1722869475.786241587]: Plugin sys_status loaded
[... 446 lines cut ...]
```

### Software Version

**QGC** : V4.2.8

**Gazebo**
user@pc:~$ gazebo --version
Gazebo multi-robot simulator, version 11.14.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

Gazebo multi-robot simulator, version 11.14.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

### Flight controller

3DR Iris Quadrotor 

### Vehicle type

Multicopter

XXLiu-HNU, 2024-09-01:
Try `ros topic  echo  /drone_1/mavros/local_position/local`?

hcws, 2024-09-01:
> Try `ros topic echo /drone_1/mavros/local_position/local`?

They are all (0,0,0)

XXLiu-HNU, 2024-09-01:
How did you solve it? We also encountered this problem when doing outdoor cluster experiments. The local position coordinate system of the drone was not uniform.~..~

hcws, 2024-09-01:
> How did you solve it? We also encountered this problem when doing outdoor cluster experiments. The local position coordinate system of the drone was not uniform.~..~

Set offsets