PX4/PX4-Autopilot #23514: Pixhawk 6x Rev8 Bugs

roboticman991999, 2024-08-07:
### Describe the bug

We purchased three Pixhawk 6x Rev 8 with Standard Baseboard V1. The first autopilot already had the PX4 v1.14.2 dev firmware and the second autopilot already had the PX4 v1.15.0 alpha firmware, both of which versions are not stable. TELEM1 port of the autopilots does not work, by changing the MAV_0_CONFIG parameter to TELEM2 and connecting of the telemetry module to TELEM2 port, there is not communication between autopilot and QGroundControl. But by changing the MAV_0_CONFIG parameter to TELEM3 and connecting the telemetry module to TELEM3 port, there is communication.

The value of the Bat1_V_DIV parameter (Battery 1 Voltage divider) cannot be changed from -1 by pressing the Calculate button. By reading its description in the PX4 site, I did not understand how much I should set for it. Also, by resetting the parameter values of the PX4 v1.15.0 alpha firmware to their default values, the corresponding autopilot will have many problems. Fortunately, it will return to its previous state after loading the parameter values of the PX4 v1.14.2 dev firmware to it.

We updated the third autopilot firmware to PX4 v1.14.3 Stable version using QGroundControl. Unfortunately, there are the following problems:
1-     TELEM1 port does not work, unlike the other two firmwares, changing the MAV_0_CONFIG parameter to TELEM2 and TELEM3 and connecting the telemetry module to TELEM2 and TELEM3 ports, problem remains.
2-     The value of the Bat1_V_DIV parameter (Battery 1 Voltage divider) cannot be changed from -1 by pressing the Calculate button. By reading its description in the PX4 site, I did not understand how much I should set for it.
3-     The radio control does not work, we tried with two radios from FrSky and RadioLink, but the message “Please turn on transmitter” is always displayed after trying to calibrate the radio controller. After hours of checking the difference of the parameters of the available firmwares, it seems that the problem is with the RC_INPUT_PROTO parameter and the code related to it, because when we select all of its options and restart all the components, the problem remains. This parameter does not exist in the dev and alpha firmwares
4-     None of the motors rotate, because the necessary signal is not sent to the speed controllers.
5-     Ardupilot firmware cannot be installed on the autopilots. 

### To Reproduce

by using QGC Update the firmware of Pixhawk 6x Rev8 to PX4 v1.14.3 Stable version

### Expected behavior

TELEM1 port can be used. Bat1_V_DIV parameter (Battery 1 Voltage divider) can be changed from -1 by pressing the Calculate button. RC can be calibrated. Motors can rotate. Ardupilot firmware can be installed on the autopilots.

### Screenshot / Media

[image]
[image]
[image]
[image]

### Flight Log

https://logs.px4.io/plot_app?log=c60d91df-a3ec-4927-9e86-dc2d8a00e1e8

### Software Version

PX4 v1.14.3 Stable
PX4 v1.14.2 dev
PX4 v1.15.0 alpha

### Flight controller

Holybro Pixhawk 6x Rev8

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

Matching bound rules

### Additional context

On Holybro's website, it is mentioned that this new autopilot (Pixhawk 6x Rev 8) is compatible with PX4 v1.14.3 stable firmware, but a non-stable version was already loaded on them. Currently, the latest stable firmware of the PX4 autopilot is version v1.14.3, and the previous stable version that can be downloaded on GitHub is v1.13.3, unfortunately both of which are not compatible with these autopilots. There are not even alpha and dev firmwares on GitHub or other sites that we can load them on the third autopilot and work with temporarily until these problems are solved in the next stable version.
Please send the download link of PX4 v1.14.2 dev and PX4 v1.15.0 alpha firmwares through this email. Also advise us how to minimize the possibility of the crash using these unstable firmware and very new hardware. In the end, I hope the necessary feedback would be sent to the PX4 and Ardupilot firmware teams so that they can provide stable firmwares compatible with this hardware as soon as possible.
Thanks

roboticman991999, 2024-08-07:
@dagar please check this bugs
thanks

TedObrien, 2024-08-14:
I've used a few Pixhawk 6x Rev 8 running 1.14.3 but with the cm4 baseboard and not had any of these issues, Although not using an identical setup. Have a few ideas:

1. Telem1 port: 
    - Could you probe vcc and gnd pins on telem1 and make sure its getting 5v?
    - Have you set the correct baud rate for the radio you are using
    - Double check the mavlink instance has been properly setup with `mavlink status` in the mavlink console
 2. Not using this so can't comment.
 3. Are you sure your transmitter is bound to the receiver? What protocol are you using? Is the wiring correct?
 4. Do the motors beep on startup and when the FC has booted? Are you sure you're using the correct protocol (eg. dshot) and your esc's are configured correctly?
 
 I haven't used that baseboard so its possible the issue could be with that.

roboticman991999, 2024-08-26:
Thank you very much.
1-	I installed the px4 v1.13.3 firmware
2-	I enabled the SYS_BL_UPDATE parameter
3-	I entered the text “bl_update /etc/extras/px4_fmu-v6x_bootloader.bin” in the MAVLink Console and the bootloader was updated
4-	I installed the px4 v1.14.3 firmware
The problem was solved.

julianoes, 2024-10-09:
Yes, the SYS_BL_UPDATE param is broken, see:
https://github.com/PX4/PX4-Autopilot/issues/20592
https://github.com/PX4/PX4-Autopilot/pull/23158