PX4/PX4-Autopilot #23528: [Bug] OPEN_DRONE_ID_LOCATION SITL Payload Missing a Byte

amsmith-pro, 2024-08-09:
### Describe the bug

The OPEN_DRONE_ID_LOCATION mavlink message broadcast from the PX4 SITL instance to localhost:14550 is consistently one payload byte short (58 instead of 59), with the correct headers and checksum bytes present. The payload is missing one of the final seven u8 fields, most likely the last field, `timestamp_accuracy`. See analysis below.

### To Reproduce

`make px4_sitl gz_x500` on Ubuntu Jammy

Bind to the 14550/udp port on localhost, filter out the OPEN_DRONE_ID_LOCATION mavlink packet.

I'm using `deser` from https://github.com/mavlink/rust-mavlink to decode the buffer. Of all the ~30 mavlink messages from the PX4 SITL instance, only the OPEN_DRONE_ID_LOCATION message fails to parse.

It fails to parse because it's expecting an enum value of 0-15 for the `timestamp_accuracy` field (the final u8 field in the payload), but since that byte is missing it instead fails to process the first checksum byte, which is usually not a value of 0-15.

Example: (will need to scroll to see whole payload)
```
could not deserialize frame: Invalid enum value for enum type "MavOdidTimeAcc", got 180
payload < 35 5b 40 1c 8a 0a 18 05 ec c4 66 3e 00 00 00 00 8e 95 4b bf 00 ff 7f 47 2c 01 04 00 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 0c 05 01 03 >
checksum < b4 84 >

could not deserialize frame: Invalid enum value for enum type "MavOdidTimeAcc", got 51
payload < 31 5b 40 1c 88 0a 18 05 b0 93 6b 3e 00 00 00 80 fd ca 4a bf 00 ff 7f 47 20 35 0b 00 0b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 0c 05 01 03 > 
checksum < 33 60 >
```
(a for loop iterates through the received udp packet buffer to print these hex values, I manually added 'payload < >' and 'checksum < >' to this markdown to make the end of the payload packet easier to see)

The number of payload bytes above are 58, instead of the expected 59. `MavOdidTimeAcc` is the enum type (0-15) expected for `timestamp_accuracy` the expected final byte of the payload. "Got 180" and "Got 51" correspond with the first checksum bytes of each, 0xb4 and 0x33, indicating that the `timestamp_accuracy` field is simply missing.

According to the mavlink code the final seven fields (a byte each) are supposed to be:
```c
_mav_put_uint8_t(buf, 52, status);
_mav_put_uint8_t(buf, 53, height_reference);
_mav_put_uint8_t(buf, 54, horizontal_accuracy);
_mav_put_uint8_t(buf, 55, vertical_accuracy);
_mav_put_uint8_t(buf, 56, barometer_accuracy);
_mav_put_uint8_t(buf, 57, speed_accuracy);
_mav_put_uint8_t(buf, 58, timestamp_accuracy);
```
Comparing with QGroundControl (listed fields are not in order):
[image]

It appears that speed_accuracy is indeed the final 0x03 byte in the payloads above.

Further analysis:
The header bytes (not shown) are confirmed correct and the two checksum bytes are present. Given that the zero'd out 20-byte `id_or_mac` array appears to be in the correct offset, it then follows that one of the final 7 bytes is being left off, most likely just a truncation of the final byte (`timestamp_accuracy`) when the checksum is added.

This confused me for a bit because QGroundControl is displaying `timestamp_accuracy` as 0 in the MAVLink analyzer, even though the 59th byte doesn't exist in the packet.

However, I see within the qgroundcontrol mavlink v2 lib that this can happen because the entire message is read into a zeroed out buffer that's 64-bit aligned (padded):
```c
//
// qgroundcontrol/libs/mavlink/include/mavlink/v2.0/common/mavlink_msg_open_drone_id_location.h
//
open_drone_id_location->timestamp_accuracy = mavlink_msg_open_drone_id_location_get_timestamp_accuracy(msg);

// where

static inline uint8_t mavlink_msg_open_drone_id_location_get_timestamp_accuracy(const mavlink_message_t* msg)
{   
    return _MAV_RETURN_uint8_t(msg,  58);
}

// where

//
// qgroundcontrol/libs/mavlink/include/mavlink/v2.0/protocol.h
//
#define _MAV_RETURN_uint8_t(msg, wire_offset) (uint8_t)_MAV_PAYLOAD(msg)[wire_offset]

// where

//
// qgroundcontrol/libs/mavlink/include/mavlink/v2.0/mavlink_types.h
//
#define _MAV_PAYLOAD(msg) ((const char *)(&((msg)->payload64[0])))

// ...

typedef struct __mavlink_message {
        // ...
        uint64_t payload64[(MAVLINK_MAX_PAYLOAD_LEN+MAVLINK_NUM_CHECKSUM_BYTES+7)/8];
        // ...
}
```

So the payload being smaller than expected won't get caught by QGC, which will read padding in the payload64 buffer in place of the missing bytes.

### Expected behavior

I expect a 59 byte payload for the OPEN_DRONE_ID_LOCATION mavlink message, with `timestamp_accuracy` present.

It's possible this marks a larger issue for more Mavlink packets; rust-mavlink fails to parse this payload because it's expecting an enum value 0-15 at the final field (`timestamp_accuracy` is an enum), which is missing, and so fails on the checksum byte (not usually within 0-15). It could be that other mavlink message payload bodies are being cut short by a byte as well, but deserializing fine because their final fields are not enum types and so the first checksum byte is being used instead as a value for the final payload field.

### Flight Log

https://logs.px4.io/plot_app?log=f1a8041b-e24f-4dbf-bd85-4c23a40a796e

### Software Version

1.15.0

### Flight controller

SITL

### Vehicle type

Multicopter

dakejahl, 2024-08-22:
https://mavlink.io/en/guide/serialization.html#payload_truncation

If you set `timestamp_accuracy` to a non zero value the payload length will indeed be 59.

closed 2024-08-22 by dakejahl

amsmith-pro, 2024-08-22:
> https://mavlink.io/en/guide/serialization.html#payload_truncation
> 
> If you set `timestamp_accuracy` to a non zero value the payload length will indeed be 59.

Thank you, missed this feature of mavlink v2. Sorry for the long writeup.