PX4/PX4-Autopilot #23589: [Bug] Simple Position Control fails to set Position Setpoint after stick movement

vertiq-luca, 2024-08-21:
### Describe the bug

When MPC_POS_MODE is set to simple position control (0) and the aircraft takes off in position mode the aircraft will hold altitude and position. If the right stick is moved to move the aircraft the position setpoint goes to NaN (as expected). Once the stick is released and the vehicle is in braking mode the setpoint remains as NaN. This prevents the vehicle from holding position and instead just uses 0 velocity as the hold mode which isn't very effective.

### To Reproduce

set MPC_HOLD_MAX_XY to 0
set MPC_POS_MODE to Simple position control

Take off

Move right stick (mode 2 radio) out of the deadzone and then release.

Observe either the PID position mode XY graph and see that the setpoint has disappeared or look at the logs for the position setpoint.

### Expected behavior

I expected the position setpoint to come back when the sticks are released and the vehicle to maintain position.

### Flight Log

https://review.px4.io/plot_app?log=c834a12f-958e-4c93-8d69-4b2b34ca54b6

### Software Version

1.14dev

### Flight controller

ModalAI Voxl2 Mini

### Vehicle type

Other

### Additional context

I chased this down to this line:
https://github.com/PX4/PX4-Autopilot/blob/3478765c310d3d34fc139dbf62594c41da6f85a9/src/modules/flight_mode_manager/tasks/ManualPosition/FlightTaskManualPosition.cpp#L113

I added some printouts in the if else ladder here and found that it was entering this line, but when printing the position setpoint, it is NaN, even though the printed position is a real number. 

It seems like var1.xy() = var2.xy() doesn't work? Altitude seems to work because it's only setting var1(2) = var2(2) instead of using the .xy() function.

MaEtUgR, 2024-09-11:
Sounds familiar: https://github.com/PX4/PX4-Autopilot/pull/22574
Is that change present on your 1.14dev branch?

MaEtUgR, 2024-09-11:
I quickly checked and the fix I linked was backported to 1.14:
https://github.com/PX4/PX4-Autopilot/pull/22578
It went into the point release PX4 1.14.1
Can you help by making sure the source for the firmware you're using actually updates to the point releases which contain critical bugfixes?
[image]

closed 2024-09-11 by vertiq-luca