PX4/PX4-Autopilot #23602: (new) GZ Feature tracker

Jaeyoung-Lim, 2024-08-22:
### Describe problem solved by the proposed feature

This is a feature tracker to achieve feature parity with gazebo classic.

### Describe your preferred solution

The list of features that are missing in the new GZ is as follows:
**PX4 related Features**
- [ ] Gimbal Control
- [ ] Camera Manager / Video Streaming
- [ ] Camera triggering/geotagging
- [ ] Faster than real-time simulation
- [ ] Rotor failures

**Sensors**
- [ ] Airspeed sensors
- [ ] GPS yaw heading simulation

**Actuators**
- [ ] Motor thrust influence of wind

**Simulation Features**
- [ ] Faster than real-time simulation
- [ ] Vehicle view tracking

### Describe possible alternatives

NA

dakejahl, 2024-08-22:
Add optical flow to the sensors list. I can help with creating the plugin architecture, I just need to find some time to do it.

mrpollo, 2024-08-22:
Maybe add switching the MAVSD SITL tests to the new GZ as well. I can take care of it once we have faster than real-time.

dirksavage88, 2024-08-23:
1. Collision prevention/obstacle distance already exists in  PX4 gz with the x500 lidar model: https://github.com/PX4/PX4-Autopilot/blob/1a0f97ebbd67870d4686ee7ebb6d38e11662eabc/src/modules/simulation/gz_bridge/GZBridge.cpp#L808 

2. Catapult is just a Wrench publisher in GZ and a model would purely  be for visual purposes. Easy to implement outside of PX4:
`gz topic -t /world/default/wrench -m gz.msgs.EntityWrench -p 'entity:{name:"x_500",type:MODEL},wrench:{force: {x:32000,y:0,z:8500,}}' -n 100`

3. Airspeed is in GZ garden, but not yet in a PX4 plugin: https://github.com/gazebosim/gz-sensors/blob/gz-sensors7/src/AirSpeedSensor.cc

Multivehicle simulation as in multiple PX4 capable vehicles or just one PX4 vehicle and another controlled externally?

Jaeyoung-Lim, 2024-08-23:
@dirksavage88 

> Collision prevention/obstacle distance already exists in PX4 gz with the x500 lidar model:

I know, but does the test work?

> Catapult is just a Wrench publisher in GZ and a model would purely be for visual purposes. Easy to implement outside of PX4:
gz topic -t /world/default/wrench -m gz.msgs.EntityWrench -p 'entity:{name:"x_500",type:MODEL},wrench:{force: {x:32000,y:0,z:8500,}}' -n 100

We need to write a plugin that does it automatically on arm to make testing more convenient. It doesn't make sense to type that everytime you test it. 

> Airspeed is in GZ garden, but not yet in a PX4 plugin: https://github.com/gazebosim/gz-sensors/blob/gz-sensors7/src/AirSpeedSensor.cc

That is just the sensor implementation. We need to incorporate it into gz bridge and handle the airframe configs for it.

dirksavage88, 2024-08-29:
> @dirksavage88
>
>
> I know, but does the test work?
>
>
> We need to write a plugin that does it automatically on arm to make testing more convenient. It doesn't make sense to type that everytime you test [...]

By test, this is in reference to a unit test of some sort? I am not sure how tests were carried out in gazebo classic.

Myzhe, 2024-10-01:
Maybe add [distance sensors (rangefinders)](https://docs.px4.io/main/en/sensor/rangefinders.html#gazebo-simulation) to the sensor list.

dirksavage88, 2024-10-07:
@Jaeyoung-Lim Please see the following [PR](https://github.com/PX4/PX4-Autopilot/pull/23783) for faster than real time support

hamishwillee, 2024-10-09:
This is very nice!! Note, we need to make sure these things are documented. Feature parity also means "works out of the box", not "we added a distance sensor to the model, but actually it can only be used from ROS if you know what you are doing, maybe ...".

Claudio-Chies, 2024-12-06:
There seems to be a bug regarding the interaction of the Advanced Lift Drag plugin, and the windy world. where the advanced plane gets blown away before takeoff. `make px4_sitl gz_advanced_plane_windy`
might be added to this list?
https://github.com/PX4/PX4-Autopilot/issues/24064

Jaeyoung-Lim, 2024-12-06:
@Claudio-Chies Lets keep bugs away from this thread and focus on missing features. Once the features are ported, bugs should be treated as is

dakejahl, 2024-12-25:
I started on adding plugin support for gz, looking for input
https://github.com/PX4/PX4-Autopilot/pull/24153

martinbaer, 2025-01-22:
Hello, is there any planned support for HITL? [This issue](https://github.com/PX4/PX4-Autopilot/issues/22431) suggested it but it has gone stale. Thanks

Pedro-Roque, 2025-02-19:
@Jaeyoung-Lim rotor plugin, last time I checked, was not bidirectional either. Not sure if this has been fixed meanwhile.

maponarooo, 2025-02-25:
I'm very looking forward gz camera features. Come on....
 
Camera Manager / Video Streaming
Camera triggering/geotagging

Roboticistprogrammer, 2025-03-28:
> **Worlds**
> 
> * [ ]  Ocean
> * [x]  Baylands

Hi @Jaeyoung-Lim, 
Can you have a look at (https://github.com/Roboticistprogrammer/PX4-gazebo-models/blob/ocean-world/worlds/ocean.sdf).
I think some stuff needs to happen at the PX4 stratup script but maybe this could be a good starting point, feel free to let me know how to move forward to add this world file.

PR #23943 (2024-12-02): Add gz model for quadtailsitter
### Solved Problem
Add the quadtailsitter and airspeed sensor to gz simulation

Fixes #23602

### Solution
Allowing airspeed sensor in gz bridge, adding a config file for the quadtailsitter.

As Gazebo garden reaches EOL by Nov 2024 this solution only takes into account sensor msgs from gazebo harmonic (LTS)   

### Changelog Entry

### Test coverage
It was tested on Ubuntu 22.04 using QGroundControl and gazebo Harmonic.

### Context
related PR with results: 

- [ ] https://github.com/PX4/PX4-gazebo-models/pull/65