PX4/PX4-Autopilot #23636: [Bug]  When flying multiple drones outdoors, VehicleLocalPosition is all (0, 0, 0)

XXLiu-HNU, 2024-09-01:
### Describe the bug

**BackGround:**
We use RTK for UAV swarm location. But we don't know **how to use the same coordinate for each UAV**. Becase each UAV treats the starting position as the coordinate origin. So, we want set the local_position by calculating the latitude and longitude height of the UAV.

### To Reproduce

1. Correctly set up the drone hardware and base station.
2. Communicate via DDS `sudo MicroXRCEAgent serial /dev/ttyUSB0 -b 300000 `
3. Each drone performs the same steps
4. Print drone location`ros2 topic echo /px4_*/fmu/out/vehicle_local_position`
5. The position of each drone is (0, 0, 0), which seems to be because when performing EKF2, the starting position is set to the origin by default. 

### Expected behavior

Then we want all drones to have the same coordinate system

### Software Version

V1.15

### Flight controller

cuav x7

### Vehicle type

Multicopter

matteodelseppia, 2024-09-07:
Why not simply using latitude/longitude coordinates then?

XXLiu-HNU, 2024-09-08:
> Why not simply using latitude/longitude coordinates then?

To tell you the truth, we're all newbies and don't know much about PX4. How should we set it up? What parameters need to be modified

matteodelseppia, 2024-09-08:
Just use /px4_*/fmu/out/vehicle_global_position

bresch, 2024-09-09:
I agree with @XXLiu-HNU that using lat/lon/alt is probably the best but an alternative solution that works well if the drone are not flying too far from each other and stay near their startup location (e.g.: drone show) is to manually set the origin of their local frame to the same coordinates using https://mavlink.io/en/messages/common.html#SET_GPS_GLOBAL_ORIGIN

closed 2024-09-09 by bresch

XXLiu-HNU, 2024-09-10:
> I agree with @XXLiu-HNU that using lat/lon/alt is probably the best but an alternative solution that works well if the drone are not flying too far from each other and stay near their startup location (e.g.: drone show) is to manually set the origin of their local frame to the same coordinates using https://mavlink.io/en/messages/common.html#SET_GPS_GLOBAL_ORIGIN
> 
> I'll close the issue as this is not a bug.

I have try, but it seem not supported, We use ROS2 galactic and the version of PX4 is 1.15. Here is the output:
[image]

bresch, 2024-09-13:
@XXLiu-HNU That's unfortunate. The message is for sure supported in PX4, it's probably something on the mavros side. Unfortunately I can't really help you with mavros.