PX4/PX4-Autopilot #23705: [Bug] GZ_BRIDGE Check GZ_SIM_RESOURCE_PATH is set correctly.

Varchasvin10, 2024-09-21:
### Describe the bug

I ran the gazebo sitl simulation model make px4_sitl gz_x500_depth and the gz-bridge doesnt seem to be working. I killed all gz servers running and still this issue persists. 

[image]

Im using ROS2 Humble on Ubuntu 22.04

### Vehicle type

None

dirksavage88, 2024-09-21:
I can fully replicate this as well, even though I am setting the sim resource path manually the error still persists

dirksavage88, 2024-09-21:
It has to do with changes to the gz bridge adding a scene service and setting a default follow distance to the airframe. I am using this patch to get around it for now. 

```
diff --git a/src/modules/simulation/gz_bridge/GZBridge.cpp b/src/modules/simulation/gz_bridge/GZBridge.cpp
index 29d19fe111..b4cc4a27c0 100644
--- a/src/modules/simulation/gz_bridge/GZBridge.cpp
+++ b/src/modules/simulation/gz_bridge/GZBridge.cpp
@@ -155,17 +155,17 @@ int GZBridge::init()
 			}
 
 			std::string scene_info_service = "/world/" + _world_name + "/scene/info";
-			bool scene_created = false;
-
-			while (scene_created == false) {
-				if (!callSceneInfoMsgService(scene_info_service)) {
-					PX4_WARN("Service call timed out as Gazebo has not been detected.");
-					system_usleep(2000000);
-
-				} else {
-					scene_created = true;
-				}
-			}
+			// bool scene_created = false;
+
+			// while (scene_created == false) {
+			// 	if (!callSceneInfoMsgService(scene_info_service)) {
+			// 		PX4_WARN("Service call timed out as Gazebo has not been detected.");
+			// 		system_usleep(2000000);
+			//
+			// 	} else {
+			// 		scene_created = true;
+			// 	}
+			// }
 
 			gz::msgs::StringMsg follow_msg{};
 			follow_msg.set_data(_model_name);
```

Jaeyoung-Lim, 2024-09-22:
@dirksavage88 Which version of gazebo are you using?

dirksavage88, 2024-09-22:
> @dirksavage88 Which version of gazebo are you using? Can you check if the service exist? The fact that the error message only appears once seems that the service call succeeds in the end no?

```
Gazebo Sim, version 8.6.0
I see now, yes it does eventually create the scene. The px4 model follow view angle is offset though, which made me suspect something was wrong with the scene/view angle and the error was causing it...So the view angle was just jarring enough and coupled with the error was just confusing.

Varchasvin10, 2024-09-23:
> It has to do with changes to the gz bridge adding a scene service and setting a default follow distance to the airframe. I am using this patch to get around it for now.
>
> ```
> diff --git [...]

Replicated this and it shows a new gz_bridge error

[image]

snDevX, 2024-09-23:
What ubuntu distro are you in? In 24.04 I get the same error. 

I noticed that `make px4_sitl` leaves the directory `~/PX4/PX4-Autopilot/build/px4_sitl_default/build_gz` empty which I think its where the problem stems.

Jaeyoung-Lim, 2024-09-23:
@Varchasvin10 @snDevX The problems you are experiencing is different with @dirksavage88 

The problem is that gz is not being built or started. I would check whether you can run gz

Varchasvin10, 2024-09-23:
> @Varchasvin10 @snDevX The problems you are experiencing is different with @dirksavage88
> 
> The problem is that gz is not being built or started. I would check whether you can run gz

gz sim is running in my system but the terminal shows gz_bridge warning and when i run an offboard example to control the uav, the gazebo simulation does not respond and the odom values also were not published.

[image]

with the offboard example,

[image]

stw124, 2024-09-27:
I have the same problem. Has anyone solved the problem?

Jaeyoung-Lim, 2024-09-28:
@Varchasvin10  The warning has nothing to do with your problems in offboard. Can you fly the vehicle using QGC?

@stw124 What do you mean by the "same"? It is not clear what the problem is in this thread.

Varchasvin10, 2024-09-30:
> @Varchasvin10 The warning has nothing to do with your problems in offboard. Can you fly the vehicle using QGC?
> 
> @stw124 What do you mean by the "same"? It is not clear what the problem is in this thread.

Even with QGC i cannot arm the uav, it gets disarmed and the teleop doesnt work.

Jaeyoung-Lim, 2024-09-30:
@Varchasvin10 sounds like your problem is not related to the one reported here

Varchasvin10, 2024-10-03:
It is, the gz_bridge error is not letting the messages be published to the gazebo

Jaeyoung-Lim, 2024-10-03:
@Varchasvin10 I don't think the error is related to the message not being published. You need to elaborate on why you have come to this conclusion.

JustinHsuNctu, 2024-10-04:
I have the same problem when I download the latest version. But when I return to the old version which I downloaded on June 11 ,2024, it's OK! I can control the UAV in Gazebo.
Post this message for reference.

usmanNoor5, 2024-10-06:
same issue

usmanNoor5, 2024-10-07:
what is the solution does any one know

octavioaguila, 2024-10-09:
Same issue :(

dakejahl, 2024-10-09:
@Jaeyoung-Lim I think people are conflating problems here. The error messages about the `/gui` services being unavailable are valid, I see it everytime. The issue is that the service calls are happening before the services are available. I can fix it by adding a 1 second sleep after `callEntityFactoryService` and `callSceneInfoMsgService`. This might vary across computers though so a better solution would be to check if the service is available before trying to call it.

snDevX, 2024-10-10:
> @Jaeyoung-Lim I think people are conflating problems here. The error messages about the `/gui` services being unavailable are valid, I see it everytime. The issue is that the service calls are [...]

tell us which files you changed

dakejahl, 2024-10-10:
@snDevX GZBridge.cpp

To be clear though... this error is harmless. It just means the `/gui/follow` service fails because the world isn't ready (defaulting the camera view to follow the drone). If your GZ sim is not working at all it's a different issue and unrelated to the "service called timed out" error

NaShoval, 2024-10-14:
Did anyone find a solution? I have ubuntu 24.04 and did not have this issue until now.
`ake px4_sitl gz_advanced_plane
[0/1] cd /home/hawk/GitDir/PX4-Autopilot/build/px4_sitl_default/src/modules/simulation/gz_br...4_SIM_MODEL=gz_advanced_plane /home/hawk/GitDir/PX4-Autopilot/build/px4_sitl_default/bin/px4

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=4008
INFO  [param] selected parameter default file parameters.bson
INFO  [param] importing from 'parameters.bson'
INFO  [parameters] BSON document size 460 bytes, decoded 460 bytes (INT32:18, FLOAT:5)
INFO  [param] selected parameter backup file parameters_backup.bson
INFO  [dataman] data manager file './dataman' size is 7872608 bytes
INFO  [init] Gazebo simulator
INFO  [init] starting gazebo with world: /home/hawk/GitDir/PX4-Autopilot/Tools/simulation/gz/worlds/default.sdf
WARN  [init] PX4_GZ_MODEL_POSE not set, spawning at origin.
INFO  [gz_bridge] world: default, model name: advanced_plane_0, simulation model: advanced_plane
INFO  [gz_bridge] Requested Model Position: 0,0,0,0,0,0
INFO  [gz_bridge] Model position z is less or equal 0.0, moving upwards
gz sim -g: symbol lookup error: /snap/core20/current/lib/x86_64-linux-gnu/libpthread.so.0: undefined symbol: __libc_pthread_init, version GLIBC_PRIVATE
ERROR [gz_bridge] Service call timed out. Check GZ_SIM_RESOURCE_PATH is set correctly.
WARN  [gz_bridge] Service call timed out as Gazebo has not been detected.
ERROR [gz_bridge] Service call timed out: /gui/follow
ERROR [gz_bridge] Service call timed out: /gui/follow/offset
INFO  [lockstep_scheduler] setting initial absolute time to 3280000 us
INFO  [commander] LED: open /dev/led0 failed (22)
WARN  [health_and_arming_checks] Preflight Fail: Airspeed selector module down
WARN  [health_and_arming_checks] Preflight Fail: ekf2 missing data
WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station
INFO  [airspeed_selector] No airspeed sensor detected. Switch to non-airspeed mode.
INFO  [uxrce_dds_client] init UDP agent IP:127.0.0.1, port:8888
INFO  [tone_alarm] home set
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] partner IP: 127.0.0.1
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280`

Jaeyoung-Lim, 2024-10-14:
@NaShoval Is there something that didn't work?

NaShoval, 2024-10-15:
> @NaShoval Is there something that didn't work?

When I run "make" in the PX4's directory, my gazebo does not initialise automatically.
When I manually initialise it, and than closing the gazebo window - the first window (that I expected PX4 to open), suddenly appears.
I just update to ubuntu 24.04 and did not have this issue until now..

dirksavage88, 2024-11-04:
> @snDevX GZBridge.cpp
>
> To be clear though... this error is harmless. It just means the `/gui/follow` service fails because the world isn't ready (defaulting the camera view to follow the drone). If [...]

The default camera view is a little jarring since it's at an offset angle to the vehicle body:
[image]

@Jaeyoung-Lim is it possible to get this fixed (just the timeout) since it still results in a degraded user experience? I understand it is a red herring for other errors users are seeing and commenting on, but fixing the core issue is still pretty important.

benbaarber, 2024-12-17:
Same issue here on ubuntu 24.04 when running `make px4_sitl gz_x500`:

```
ERROR [gz_bridge] Service call timed out. Check GZ_SIM_RESOURCE_PATH is set correctly.
WARN  [gz_bridge] Service call timed out as Gazebo has not been detected.
```

And gazebo launches exactly like @dirksavage88 most recent comment

I tried running the same command first in an ubuntu:22.04 docker container after running `Tools/setup/ubuntu.sh`, got the same error.

Jaeyoung-Lim, 2024-12-18:
@benbaarber Please consider making a pull request for the fix

mylongword, 2024-12-19:
Same issue here on ubuntu 24.04 when running PX4_GZ_WORLD=lawn    make px4_sitl gz_x500:

ERROR [gz_bridge] Service call timed out. Check GZ_SIM_RESOURCE_PATH is set correctly.
ERROR [gz_bridge] Task start failed (-1)
ERROR [init] gz_bridge failed to start and spawn model

The Patch by @dirksavage88 worked (I don't understand why).
To avoid another timeout warning on my Nitro 5, I also changed, in the same file,
if (_node.Request(service, req, 1000, rep, result)) {
to
if (_node.Request(service, req, 5000, rep, result)) {

ckjunaid, 2024-12-24:
> @benbaarber Please consider making a pull request for the fix
sir i do face this problem , the problem is we launched 3 terminal one for simulation from px4 and dds communication and keyboard operation .
actually when i click the space it should arm but not responding to it but can arm and fly using qgc iam following the ark electronics git ....

ckjunaid, 2024-12-24:
[
[image]
](url)

surajsurve4511, 2025-01-01:
ss

surajsurve4511, 2025-01-01:
I also faced the error , I had tried solving the error with  many ways but the GZBridge is not properly build in new versions of px4.
Shall we try to switch and use gazebo - classic , for that we need to downgrade to ubuntu 20

Jaeyoung-Lim, 2025-01-01:
@surajsurve4511 As mentioned above, this is not an error

surajsurve4511, 2025-01-01:
But as of I tried of fetching Lidar sensor data with ros I was unable to get the data in ros2 node ..!
Can you just guide me to get lidar sensor data to ros2 node ,I want the data to be fetched in a python script.!

TheMightiestCarrot, 2025-01-21:
Guys! i have found a simple adhoc solution!
Run Gazebo in headless mode first, wait a bit and then start px4:

```
gz sim ~/PX4-Autopilot/Tools/simulation/gz/worlds/default.sdf & GZ_PID=$!

sleep 5

PX4_SYS_AUTOSTART=4002 PX4_GZ_MODEL_POSE="268.08,-128.22,3.86,0.00,0,-0.7" PX4_GZ_MODEL=x500_depth ./build/px4_sitl_default/bin/px4
```

Or you can just start gazebo in one terminal:
```gz sim ~/PX4-Autopilot/Tools/simulation/gz/worlds/default.sdf```
and px4 in a new one:
```PX4_SYS_AUTOSTART=4002 PX4_GZ_MODEL_POSE="268.08,-128.22,3.86,0.00,0,-0.7" PX4_GZ_MODEL=x500_depth ./build/px4_sitl_default/bin/px4```

dakejahl, 2025-01-21:
It's fixed in the PR https://github.com/PX4/PX4-Autopilot/pull/24164

PR #24164 (2025-01-21): gz: increase timeout for service request
Closes #23705
Closes #23809
closed 2025-01-21 by dakejahl