PX4/PX4-Autopilot #23714: [Bug] GPS does not show up for PixRacer Pro in v1.15.0

sssagara, 2024-09-24:
### Describe the bug

GPS module (Holybro Neo-M8N) doesn't seem to be recognized when PixRacer Pro is flashed to v1.15.0. GPS Status does not show up as valid in QGC main page and GPS_RAW_INT message is also not present in MAVLink Inspector. 

When flashed with v1.14.3 with the exact same setup and params, GPS module is immediately recognized and GPS status and GPS_RAW_INT message are present.

### To Reproduce

1. Flash PixRacer Pro to v1.15.0 (tried via both QGC firmware tab and building from source)
[image]
2. GPS status shows up as invalid on QGC main page
[image]
3. GPS_RAW_INT message not present in MAVLink Inspector
[image]
4. GPS still does not show up even when drone is placed outdoors with clear skies

### Expected behavior

Expected to see GPS status on QGC main page and GPS_RAW_INT message in MAVLink Inspector

### Flight Log

https://review.px4.io/plot_app?log=a0c84de9-c359-4f8c-b4b1-4331c02232ba

### Software Version

HW arch: MRO_PIXRACERPRO
PX4 git-hash: 30e763b6780061d70a14894e3e8b06e6a656f9b8
PX4 version: Release 1.15.0 (17760511)
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: 5d74bc138955e6f010a38e0f87f34e9a9019aecc
Build datetime: Sep 23 2024 14:27:51
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 00060000000035363233313251020046001c
MCU: STM32H7[4|5]xxx, rev. V

### Flight controller

mRo PixRacer Pro

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

1. GPS module plugged into GPS port of PixRacer Pro
2. GPS_1_CONFIG set to GPS 1
3. SER_GPS1_BAUD set to Auto

### Additional context

When flashed with v1.14.4 with the setup and params unchanged and the same, the GPS module is immediately recognized after flashing. Tried figuring out if the GPS module was faulty by testing it with a Pixhawk 4. When Pixhawk 4 is flashed with either v1.14.4 or v1.15.0, the same GPS module is immediately recognized and GPS_RAW_INT messages present in MAVLink Inspector

v1.14.4 PixRacer Pro
[image]
[image]
[image]

bresch, 2024-09-24:
Found that `dmesg` reports 
```
Starting GPS Port on /dev/ttyS3
Conflicting config for /dev/ttyS3
```

This is apparently because `SEP_PORT1_CFG` is also set to GPS 1 by default so it conflicts with `GPS_1_CONFIG`.

Solution for you @sssagara is to set `SEP_PORT1_CFG` to "disabled" and reboot.

sssagara, 2024-09-24:
> Found that `dmesg` reports
> 
> ```
> Starting GPS Port on /dev/ttyS3
> Conflicting config for /dev/ttyS3
> ```
> 
> This is apparently because `SEP_PORT1_CFG` is also set to GPS 1 by default so it conflicts with `GPS_1_CONFIG`.
> 
> Solution for you @sssagara is to set `SEP_PORT1_CFG` to "disabled" and reboot.

Yes! That solved it! Thank you for the help and really appreciate the quick reply!

closed 2024-09-24 by sssagara