PX4/PX4-Autopilot #23727: [Bug] Dronecan Firmware Updating is Broken on Main

AlexKlimaj, 2024-09-25:
### Describe the bug

Firmware updating on 1.15 works but is broken on main after the Dronecan submodule update. Nodes fail the update then the update restarts and fails again indefinitely. 

```
Online nodes (Node ID, Health, Mode):
         71 CRIT       SW_UPD

uavcan: cycle time: 71419 events, 4965837us elapsed, 69.53us avg, min 10us max 72605us 488.892us rms
uavcan: cycle interval: 71419 events, 2893.50us avg, min 19us max 266795us 2696.179us rms
nsh> INFO  [uavcan:71:Boot] Es
ERROR [uavcan:71:Boot] Pf
INFO  [uavcan:71:Boot] Es
ERROR [uavcan:71:Boot] Pf
INFO  [uavcan:71:Boot] Es
```

### Vehicle type

None

julianoes, 2024-10-08:
This is updating via the PX4 SD card? I wasn't aware of that method and always used the gui tool with a USB to CAN tool.

AlexKlimaj, 2024-10-08:
The main way I've always been updating firmware over CAN has been from the PX4 SD card. Most customers use this method as well since not many people have USB to CAN adapters.

julianoes, 2024-10-09:
Makes sense. I will have to test this and fix it.

closed 2024-10-17 by dagar