PX4/PX4-Autopilot #23740: [Bug] Yaw estimate error with Holy bro kakute h7 v1.3 with Px4 1.15 stable

airsimdevdd, 2024-09-30:
### Describe the bug

i flashed my kakute  h7 with px4 1.15  and connected a holy bro m9n gps with compass to it via sda scl its yaw estimate is very unstable and jumps off by 180 degrees and i get many errors in qgc regarding it like yaw estimate error , heading not stable , it is yawing in stablized mode too 

### To Reproduce

1. power on th drone 
2. connect it to qgc  
3. u can see heading estimate error and heading jumps off suddenly 

### Expected behavior

heading must be stable not jump by a large scale

### Software Version

PX4 1.15 stable

### Flight controller

holy bro kakute v1.3

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

compass is connected to sda scl port of fc

federicociresola, 2024-10-04:
Have you tried using the previous stable version 1.14.4? Just to understand if there is a problem with the PX4 version or your setting

bresch, 2024-10-09:
The log is missing a lot of topics. Are you sure the compass is correctly connected? Check the output of `listener vehicle_magnetometer` in the mavlink shell of QGC

airsimdevdd, 2024-11-07:
> The log is missing a lot of topics. Are you sure the compass is correctly connected? Check the output of `listener vehicle_magnetometer` in the mavlink shell of QGC

yes it is published ,still im getting yaw estimate error  and heading suddenly starting moving if the drone is stable

airsimdevdd, 2024-11-07:
> Have you tried using the previous stable version 1.14.4? Just to understand if there is a problem with the PX4 version or your setting

i initially tried 1.14.3 then moved 1.15.1  facing same isssue on both

closed 2026-03-04 by github-actions[bot]