PX4/PX4-Autopilot #23756: [Bug] fixed-wing airspeed and wind estimation problem in SITL

TigerWuu, 2024-10-01:
### Describe the bug

PX4 software-in-the-loop calculates the airspeed using differential pressure in ***SensorAirspeed.cpp*** in the ***simulation*** module. However, SensorAirspeed.cpp generates the fake differential pressure by **body velocity**, it will induce some airspeed calculating and wind estimation problems when the wind comes from the x-axis of the body frame  
[image]
I think a more feasible modification is using the wind triangle concept, obtaining the ground speed from uorb topic and true wind value from gazebo plugin, and calculating the true airspeed to generate the differential pressure. What do you guys think?

### To Reproduce

1. Follow the steps in https://docs.px4.io/main/en/ros2/user_guide.html to install PX4 autopilot, ROS2 and gazebo sim.
2. Open a standard_vtol in the windy.sdf 
3. Set the wind magnitude as (0, -2, 0) using the *wind* tag in the sdf file.
    <wind><linear_velociy>0 -2 0</linear_velociy></wind> 
4. Arm the drone and transition to the fixed wing mode, fly straightly to the north.
5. Open the plotjuggler to monitor the wind estimation value in the ***/fmu/out/wind***

### Expected behavior

the ***windspeed_east*** and ***windspeed_north*** value should converge to real value (0, -2)
[image]

### Software Version

[image]

### Vehicle type

Hybrid VTOL

Jaeyoung-Lim, 2024-10-01:
@TigerWuu Feel free to open a pull request for the fix

TigerWuu, 2024-10-25:
@Jaeyoung-Lim Hi, I have fixed the wind estimation problem the following is the result. Now the estimations look much more normal.
[image]

I modified two repositories to fix this problem, ***gz-sim*** and ***PX4-Autopilot***. The current information flow will be like this
[image]

* gz-sim : I added a new plugin **WindEffectPub.cc** referring to the **WindEffect.cc**, which will publish the wind information to **/world/$myworld$/wind_info** topic
* PX4-Autopilot : I modified the **gz_bridge.cpp** to receive the wind information from GZ and publish to a new message **wind_gz**, then the **Sensor _airspeed_sim.cpp** will receive the **wind_gz** message and **vehicle_local_position** to calculate the true airspeed from wind triangle.

However, I have the following problem with PR, I am currently working on 
* PX4 V1.14
* GZ sim garden 7.9.0

cause I run into the same issue as #23705,  if I use the latest version of px4 and gz sim
* PX4 V1.15
* GZ sim harmonic 8.6.0

what should I do? should I wait until this bug fixed then apply my modification into the latest version of px4 and gz?

Jaeyoung-Lim, 2024-10-29:
@TigerWuu Thanks for looking into the problem.

You would have to reapply your modification to the latest 'main' of the PX4 repository to apply your changes. Otherwise your changes would not get applied in future versions