PX4/PX4-Autopilot #23800: [Bug] Latest parameter meta data generation is causing failures in QGC validation

DonLakeFlyer, 2024-10-10:
### Describe the bug

Output from QGC says it all:
```
PX4ParameterMetaDataLog: Invalid max value, name: "VTQ_TELEM_IDS_1"  type: 5  max: "4294967295"  error: "Value must be within 0 and 2147483647"
PX4ParameterMetaDataLog: Invalid value for bitmask, bit: 31
PX4ParameterMetaDataLog: Invalid max value, name: "VTQ_TELEM_IDS_2"  type: 5  max: "9223372036854775807"  error: "Value must be within 0 and 2147483647"
```

These parameters are INT32s. Note the metadata for max which is larger than an INT32.

And then for `VTQ_TELEM_IDS_2`:
```
PX4ParameterMetaDataLog: Invalid value for bitmask, bit: 32
PX4ParameterMetaDataLog: Invalid value for bitmask, bit: 33
...
```

This is trying to set bits which are not available on an INT32.

Also since these are both bit mask parameters these should really be unsigned.

### To Reproduce

Boot latest daily build QGC

### Vehicle type

None

DonLakeFlyer, 2024-10-10:
Also FYI: Although parameters theoretically support [u]int64's. QGCs px4 parameter metadata parsing code assumes 32 bits only for bit masks. That said switching to support 64 bits doesn't seem like a great idea since it's had zero testing. I don't think any current params are 64 bit.

DonLakeFlyer, 2024-10-10:
@dagar Any reason bit masks can be UINT32? That would fix the max problem.

DonLakeFlyer, 2024-10-11:
A unit test which sanity checks the metadata would be nice so that this doesn't make it all the way to QGC before it's discovered. Look at https://github.com/mavlink/qgroundcontrol/blob/master/src/FirmwarePlugin/PX4/PX4ParameterMetaData.cc#L78 to see how QGC validates it for examples to test.

DonLakeFlyer, 2024-10-12:
Also a simple option is to remove min/max from the output bitmask metadata. In reality since this is a signed int they are both being set incorrectly.

sickopickle, 2025-03-08:
Hi I'm having a similar issue

`PX4ParameterMetaDataLog: Invalid max value, name: "VTQ_TELEM_IDS_1"  type: 5  max: "4294967295"  error: "Value must be within 0 and 2147483647"
PX4ParameterMetaDataLog: Invalid value for bitmask, bit: 31`

Seems to be for IDS_1 not IDS_2. I am trying to load a custom world from [asv_wave_sim](https://github.com/srmainwaring/asv_wave_sim). Everything functions and loads in but I'm not able to launch the drone. Below is my full terminal log

```
qt.multimedia.ffmpeg: Using Qt multimedia with FFmpeg version 7.1 LGPL version 2.1 or later
qt.multimedia.ffmpeg: Available HW decoding frameworks:
qt.multimedia.ffmpeg:      cuda
qt.multimedia.ffmpeg: Available HW encoding frameworks:
qt.multimedia.ffmpeg:      cuda
PX4ParameterMetaDataLog: Invalid max value, name: "VTQ_TELEM_IDS_1"  type: 5  max: "4294967295"  error: "Value must be within 0 and 2147483647"
PX4ParameterMetaDataLog: Invalid value for bitmask, bit: 31
PX4ParameterMetaDataLog: Invalid max value, name: "VTQ_TELEM_IDS_1"  type: 5  max: "4294967295"  error: "Value must be within 0 and 2147483647"
PX4ParameterMetaDataLog: Invalid value for bitmask, bit: 31
PX4ParameterMetaDataLog: Invalid max value, name: "VTQ_TELEM_IDS_1"  type: 5  max: "4294967295"  error: "Value must be within 0 and 2147483647"
PX4ParameterMetaDataLog: Invalid value for bitmask, bit: 31
Error loading text-to-speech plug-in "speechd"
qgc.audio.audiooutput: Failed to set the TTS engine.
VideoReceiverLog: Stop called on empty URI
VideoReceiverLog: Stop called on empty URI
qt.positioning.geoclue2: Unable to start the client: "org.freedesktop.DBus.Error.AccessDenied" "Geolocation disabled for UID 1000"
qgc.positionmanager.positionmanager: QGCPositionManager::_setPositionSource(QGCPositionSource)::<lambda(QGeoPositionInfoSource::Error)> QGeoPositionInfoSource::AccessError
PX4ParameterMetaDataLog: Invalid max value, name: "VTQ_TELEM_IDS_1"  type: 5  max: "4294967295"  error: "Value must be within 0 and 2147483647"
PX4ParameterMetaDataLog: Invalid value for bitmask, bit: 31
qgc.audio.audiooutput: AudioOutput not initialized. Call init() before using say().
ComponentInformationTranslationLog: Locale  "en_US"  not found in json
ComponentInformationTranslationLog: Locale  "en_US"  not found in json
ParameterManagerLog: Attemping load from cache
ParameterManagerLog: Parameters loaded from cache /home/pickle/.config/QGroundControl.org/ParamCache/1_1.v2
FirmwarePluginLog: Unable to parse version info from file "https://api.github.com/repos/PX4/Firmware/releases"

```

DonLakeFlyer, 2025-03-10:
All of those are just warnings and shouldn't causes any problems with flight.