PX4/PX4-Autopilot #23809: [Bug] Issue in gazebo opening after running "make"

NaShoval, 2024-10-14:
### Describe the bug

When I run "make" in the PX4's directory, my gazebo does not initialise automatically.
When I manually initialise it, and than closing the gazebo window - the first window (that I expected PX4 to open), suddenly appears. 
I just update to ubuntu 24.04 and did not have this issue until now.. 

### Flight Log

```
make px4_sitl gz_advanced_plane
[0/1] cd /home/hawk/GitDir/PX4-Autopilot/build/px4_sitl_default/src/modules/simulation/gz_br...4_SIM_MODEL=gz_advanced_plane /home/hawk/GitDir/PX4-Autopilot/build/px4_sitl_default/bin/px4

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=4008
INFO  [param] selected parameter default file parameters.bson
INFO  [param] importing from 'parameters.bson'
INFO  [parameters] BSON document size 460 bytes, decoded 460 bytes (INT32:18, FLOAT:5)
INFO  [param] selected parameter backup file parameters_backup.bson
INFO  [dataman] data manager file './dataman' size is 7872608 bytes
INFO  [init] Gazebo simulator
INFO  [init] starting gazebo with world: /home/hawk/GitDir/PX4-Autopilot/Tools/simulation/gz/worlds/default.sdf
WARN  [init] PX4_GZ_MODEL_POSE not set, spawning at origin.
INFO  [gz_bridge] world: default, model name: advanced_plane_0, simulation model: advanced_plane
INFO  [gz_bridge] Requested Model Position: 0,0,0,0,0,0
INFO  [gz_bridge] Model position z is less or equal 0.0, moving upwards
gz sim -g: symbol lookup error: /snap/core20/current/lib/x86_64-linux-gnu/libpthread.so.0: undefined symbol: __libc_pthread_init, version GLIBC_PRIVATE
ERROR [gz_bridge] Service call timed out. Check GZ_SIM_RESOURCE_PATH is set correctly.
WARN  [gz_bridge] Service call timed out as Gazebo has not been detected.
ERROR [gz_bridge] Service call timed out: /gui/follow
ERROR [gz_bridge] Service call timed out: /gui/follow/offset
INFO  [lockstep_scheduler] setting initial absolute time to 3280000 us
INFO  [commander] LED: open /dev/led0 failed (22)
WARN  [health_and_arming_checks] Preflight Fail: Airspeed selector module down
WARN  [health_and_arming_checks] Preflight Fail: ekf2 missing data
WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station
INFO  [airspeed_selector] No airspeed sensor detected. Switch to non-airspeed mode.
INFO  [uxrce_dds_client] init UDP agent IP:127.0.0.1, port:8888
INFO  [tone_alarm] home set
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] partner IP: 127.0.0.1
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280 
```

### Software Version

Gazebo harmonic 8.6.0

### Vehicle type

Other

Wangzy321, 2024-11-06:
The same problem. How can it be solved?

Wangzy321, 2024-11-06:
[image]
How should it be resolved?

leechor, 2025-01-03:
> ![1](https://private-user-images.githubusercontent.com/83379250/383450667-48079198-24c2-45a1-8ee6-3c2439c1973f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU4OTU5NTYsIm5iZiI6MTczNTg5NTY1NiwicGF0aCI6Ii84MzM3OTI1MC8zODM0NTA2NjctNDgwNzkxOTgtMjRjMi00NWExLThlZTYtM2MyNDM5YzE5NzNmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTAzVDA5MTQxNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU2MjhmNmNlZDQ5ZGFhMmQ2MDUzZDFkZGFiNDI2OWMzZDdjNWU0MDkzYjIxZDljNzUxYzBkMzhjYWVkNzRjNDQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.wwcHoXeoS5aokOmoFB_ajw6wF98pjIMqmwUT855-oag) How should it be resolved?

have you solved it ?

dakejahl, 2025-01-03:
@leechor try https://github.com/PX4/PX4-Autopilot/pull/24164

PR #24164 (2025-01-21): gz: increase timeout for service request
Closes #23705
Closes #23809
closed 2025-01-21 by dakejahl