PX4/PX4-Autopilot #23817: Which estimator(s) is being used out of all (PX4 | Pixhawk 6C/6X/6X Pro)

MuhammadBilal1, 2024-10-16:
### Describe the bug

There are multiple **Accel/Gyro** sensors available in the different editions of the Pixhawk (6C/6X/6X Pro).

### How these multiple sensors are being used?

- **Sensor fusion**
  - If the sensor fusion is being done, what happens to the bad data if there is any?
- **Only single sensor with good data**
  - If only single instance is being used, how to identify which sensor is being used?
  - If only single instance is being used, how the sensor is switched during the flight if that data goes bad?

How the priority of the sensors is set?
How do I know which sensor is being used in the flight from the flight log?

[image]

### Flight Log

Pixhawk 6X Pro:
https://review.px4.io/plot_app?log=9bc0accd-c0d1-4b60-bd06-51df47df33b9

Pixhawk 6C:
https://review.px4.io/plot_app?log=3c848853-7a08-48aa-b412-4c4974c7f548

[image]

[image]

### Software Version

PX4 1.14.3
PX4 1.15.0

### Flight controller

Pixhawk

### Vehicle type

Multicopter

AlexisTM, 2024-10-16:
It seems you want a discussion, in which case it may be better to drop a message on Discord or Discourse.

MuhammadBilal1, 2024-10-16:
> It seems you want a discussion, in which case it may be better to drop a message on Discord or Discourse.

Done. Any help regarding this will be appreciated.
https://discord.com/channels/1022170275984457759/1022186188255285399/1296123280599679086

mrpollo, 2024-10-16:
You should find answers to your questions here https://docs.px4.io/main/en/advanced_config/tuning_the_ecl_ekf.html

closed 2026-03-04 by github-actions[bot]