PX4/PX4-Autopilot #23823: [Bug] VTOL will crash in return mode after velocity control via offboard mode

vegetable-to-kou-foot, 2024-10-17:
### Describe the bug

Hi, we are testing the VTOL drone using PX4 1.14.3+MAVROS 1.19.0+Ubuntu 20.04, but we found that after offboard velocity control in fixed wing mode, the drone crashed when switched to return mode.

Based on the logs, we found that the fixed-wing motor's RPM was constant in return mode, which could lead to the drone not having enough lift and thus crashing. As a comparison, drone wont crash in return mode after position control, which makes us wonder.

We tested this in version 1.13.2, 1.14.0 and 1.14.3, where 1.14 had this issue, but 1.13 was fine. I think some logic error may happened in the 1.14 update, but due to specific sensor requirements, we can only use the version above 1.14. I sincerely want to know how to deal with this issue.

There are our logs and runnable primary code.

Velocity control, return and crash in real flight: https://logs.px4.io/plot_app?log=f2a737ec-4436-4806-92e6-f8fc320d4b4c
[image]

The oscillations of the TECS controller do not seem to be related to this problem, since we can still reproduce the problem in SITL after solving the oscillations.
Crash in SITL: https://logs.px4.io/plot_app?log=56f79dc6-ca1a-44b5-a0dd-dcb45118fbe8
[image]

Position control, return and crash in real flight: https://logs.px4.io/plot_app?log=473e2a04-2c2e-42df-9c6c-3dd0ca746df1
[image]

Runnable primary code: https://paste.ubuntu.com/p/bhxjVzG2Vv/

Thank you!

### To Reproduce

1. Drone switched on
2. Took off
3. Switched to fixed-wing
4. Switched to offboard mode
5. Sent velocity control command
6. Stop velocity control, drone exited offboard mode and activated failsafe mode(set to return mode)
7. Fixed-wing motor’s RPM was constant
8. Crashed

### Expected behavior

Fixed wing motor's  RPM should be controlled to vary.

### Software Version

NuttShell (NSH) NuttX-11.0.0
nsh> ver all
HW arch: PX4_FMU_V5
HW type: V5006000
HW version: 0x006
HW revision: 0x000
PX4 git-hash: 1dacb4cdef2d7145754fc788fa8dc482eed74b40
PX4 version: Release 1.14.3 (17695743)
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: de41e7feaeffaec3ce65327e9569e8fdb553ca3d
Build datetime: May 30 2024 03:29:42
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 00020000000036323635303351130044003b
MCU: STM32F76xxx, rev. Z

### Flight controller

CUAV V5 nano

### Vehicle type

Hybrid VTOL

swarm5, 2024-10-17:
OMG, so glad I found this issue. I also found the same problem in my testing. I think this is a big risk and should attract the attention of the development team.

swarm5, 2024-10-17:
@Jaeyoung-Lim

kcx1207, 2024-10-17:
I also found this issue before, so I gave up on the 1.14 version of the firmware. Specifically, in the gazebo SITL, after stopping sending the offboard speed command, the drone switches to RETURN mode and the motor speed is constant and then crashes!
https://review.px4.io/plot_app?log=e13517bd-8831-495d-b2ec-500abbfe4ec5
P.S. If the offboard speed command is close to the home point, the drone will quickly enter multi-rotor mode and land without crashing!
https://review.px4.io/plot_app?log=fb9c89d9-a981-4181-9f93-8fb9f1906634

Jaeyoung-Lim, 2024-10-17:
@vegetable-to-kou-foot @kcx1207 @swarm5 This is not a bug. The documentation clearly states that velocity setpoints are not supported in offboard mode for px4: https://docs.px4.io/main/en/flight_modes/offboard.html#fixed-wing

Making velocity setpoints work are not that hard, but it needs to be tested and implemented in the `fw_pos_control`