PX4/PX4-Autopilot #23842: [Bug] Unable to get publish topic in companion computer RasPi

willian-li, 2024-10-23:
### Describe the bug

I was successfull connect ROS2 and PX4 via serial by the guide [[https://docs.px4.io/main/en/companion_computer/pixhawk_rpi.html](https://github.com/PX4/PX4-Autopilot/issues/url)](url)

Opened the agent,It seems normal connected ,but only created subscriber

```
 MicroXRCEAgent serial --dev /dev/serial0 -b 921600
[1729656300.229474] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1729656300.230241] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1729656610.665654] info     | Root.cpp           | create_client            | create                 | client_key: 0x00000001, session_id: 0x81
[1729656610.665872] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x00000001, address: 1
[1729656610.823344] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x00000001, participant_id: 0x001(1)
[1729656610.935502] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x800(2), participant_id: 0x001(1)
[1729656610.935869] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x800(4), participant_id: 0x001(1)
[1729656610.938791] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x800(6), subscriber_id: 0x800(4)
[1729656611.050873] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x801(2), participant_id: 0x001(1)
[1729656611.051157] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x801(4), participant_id: 0x001(1)
[1729656611.053431] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x801(6), subscriber_id: 0x801(4)
[1729656611.159530] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x802(2), participant_id: 0x001(1)
[1729656611.159772] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x802(4), participant_id: 0x001(1)
[1729656611.162041] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x802(6), subscriber_id: 0x802(4)
[1729656611.268209] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x803(2), participant_id: 0x001(1)
[1729656611.268516] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x803(4), participant_id: 0x001(1)
[1729656611.270824] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x803(6), subscriber_id: 0x803(4)
[1729656611.376358] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x804(2), participant_id: 0x001(1)
[1729656611.376607] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x804(4), participant_id: 0x001(1)
[1729656611.378905] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x804(6), subscriber_id: 0x804(4)
[1729656611.483236] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x805(2), participant_id: 0x001(1)
[1729656611.483476] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x805(4), participant_id: 0x001(1)
[1729656611.485679] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x805(6), subscriber_id: 0x805(4)
[1729656611.618511] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x806(2), participant_id: 0x001(1)
[1729656611.618804] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x806(4), participant_id: 0x001(1)
[1729656611.621017] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x806(6), subscriber_id: 0x806(4)
[1729656611.731680] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x807(2), participant_id: 0x001(1)
[1729656611.731914] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x807(4), participant_id: 0x001(1)
[1729656611.734131] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x807(6), subscriber_id: 0x807(4)
[1729656611.839739] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x808(2), participant_id: 0x001(1)
[1729656611.839966] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x808(4), participant_id: 0x001(1)
[1729656611.842125] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x808(6), subscriber_id: 0x808(4)
[1729656611.950880] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x809(2), participant_id: 0x001(1)
[1729656611.951113] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x809(4), participant_id: 0x001(1)
[1729656611.953234] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x809(6), subscriber_id: 0x809(4)
[1729656612.060988] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x80A(2), participant_id: 0x001(1)
[1729656612.061322] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x80A(4), participant_id: 0x001(1)
[1729656612.063867] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x80A(6), subscriber_id: 0x80A(4)
[1729656612.173254] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x80B(2), participant_id: 0x001(1)
[1729656612.173565] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x80B(4), participant_id: 0x001(1)
[1729656612.176007] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x80B(6), subscriber_id: 0x80B(4)
[1729656612.280689] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x80C(2), participant_id: 0x001(1)
[1729656612.280930] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x80C(4), participant_id: 0x001(1)
[1729656612.283367] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x80C(6), subscriber_id: 0x80C(4)
[1729656612.390301] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x80D(2), participant_id: 0x001(1)
[1729656612.390607] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x80D(4), participant_id: 0x001(1)
[1729656612.393019] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x80D(6), subscriber_id: 0x80D(4)
[... cut ...]
OS git-hash: d140f96627bd55edd24060cb90fe4c55ed3b9efd
Build datetime: Oct 22 2024 13:05:44
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 0002000000003234333830395119002c0041
MCU: STM32F76xxx, rev. Z

### Flight controller

CUAV V5 nano

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

PX4 TELEM2 Pin	RPi GPIO Pin
UART5_TX (2)	RXD (GPIO 15 - pin 10)
UART5_RX (3)	TXD (GPIO 14 - pin 8)
GND (6)	Ground (pin 6)

ziwei-gao, 2024-11-28:
Hello, I also encountered the same problem. How did you solve it? Thanks

willian-li, 2024-11-28:
I changed my Flight controller from CUAV V5 nano to Pixhawk 6x.

> Hello, I also encountered the same problem. How did you solve it? Thanks

closed 2024-11-28 by willian-li

ziwei-gao, 2024-11-28:
> I changed my Flight controller from CUAV V5 nano to Pixhawk 6x.
> 
> > Hello, I also encountered the same problem. How did you solve it? Thanks

thanks