PX4/PX4-Autopilot #23852: [Bug] Height estimation error in SITL with v1.15.0

goth42378, 2024-10-25:
### Describe the bug

I was trying to do the gazebo simulation using the ROS2 individual motor control message open in PX4, but in the process I found that the altitude estimation of the drone was flawed. When the drone was stationary on the ground, the altitude estimation sometimes jumped to more than 1.5 m and the speed jumped to 0.5 m/s.
I tried to replace the PX4 version, but since the individual controller was not available until after V1.15.0, I tested v1.15.1 v1.15.0 and v1.15.0 beta. All these versions had similar problems.
I don't know how to fix it. Has anyone ever had a problem like this?

### To Reproduce

1. Run the simulition with command `make px4_sitl gazebo-classic`
2. See the height estimate at the ground station via the mavlink inspecter

### Expected behavior

The height and the vertical velocity should be close to zero.

### Screenshot / Media

[image]

### Vehicle type

Multicopter

kkcqh, 2024-12-03:
I have also encountered the same problem. Have you solved it?

goth42378, 2024-12-04:
> I have also encountered the same problem. Have you solved it?

Nope. I rolled back the project to 742d237.

Hotchnerz, 2025-01-21:
I have a similar issue as well on SITL w/ gazebo classic using v1.15.0 - v1.15.3. Currently using ROS2 Foxy on Ubuntu 20.04.

[image]

Just sending a pos setpoint to the drone and leaving the simulation for a couple of minutes will result in the simulated drone faltering in height and sink towards the ground. Altitude estimation of the drone is flawed similar to above. Additionally, When reviewing the logs, the local pos estimate for Z does not reflect the drone on the ground.

[image]

Currently fixed by rolling to v1.14 for SITL w/ gazebo classic.