PX4/PX4-Autopilot #23874: Support two or more optical flows to obtain better position information for indoor areas

serkanMzlm, 2024-10-31:
### Describe problem solved by the proposed feature

When flying a drone in an indoor environment, many errors occur in the x and y axes. I think the reason for this is that 1 optical flow is not enough and in this case it is calculating its position incorrectly. I think that different optical flow data should be combined for a more stable position mode flight in indoor spaces.

### Describe your preferred solution

Using multiple optical flows simultaneously

### Describe possible alternatives

optik akış veya IMU tarafında ayarlamalar

mrpollo, 2025-02-14:
I think this has been discussed in the past, but we don't currently have any bandwidth to work on this, we would gladly guide you on how to do this if you have the time to contribute a PR.

serkanMzlm, 2025-02-17:
@mrpollo I would like to work on this topic in my free time. However, if you can guide me on the path I should follow, I would like to start developing in this area.

afwilkin, 2025-02-18:
@serkanMzlm We are working on this. If you would like to message me, I would like to understand your background and see if we can join forces. 

Please feel free to email me at andrew.wilkins@ascendengineer.com

serkanMzlm, 2025-02-19:
@afwilkin,  I am currently considering using VO for better positioning. By providing the VO output data to PX4, I believe a more stable position estimation can be achieved in indoor systems.

closed 2025-02-20 by serkanMzlm