PX4/PX4-Autopilot #23875: [Bug] Irregular frequency of rate_ctrl when px4io driver is running

Igor-Misic, 2024-10-31:
### Describe the bug

I've notice irregular calls in wq:**rate_ctrl** when **px4io** drive is running. 
For example, **control_allocation** is expect to run `2.5 [ms]` loop but every `1 [s]` it can be seen that is not case. 

Here is plot where it can be seen that oscillations are more than double with 1 Hz frequency .
In the first part px4io is running, while in the second part I stopped px4io with stop command and there are not spikes anymore.
[image]

I've used control_allocation, but it is the same behavior for all loops that use wq:**rate_ctrl**.

### To Reproduce

Use this code to reproduce:
https://github.com/PX4/PX4-Autopilot/tree/test-rate_ctrl_performance

### Software Version

v1.14.4 and main

### Flight controller

fmu-v5x

### Vehicle type

None

dakejahl, 2024-11-05:
Which hardware? I just tested on pixhawk 5x and cube orange and don't see the issue. Do you need to be armed or have anything else plugged in?

Igor-Misic, 2024-11-05:
@dakejahl thanks for trying to reproduce and interesting observation. I used a Pixhawk fmu-v5x, only connected device is RC at PX4IO with the SBUS protocol. 

I also test with fmu-v6x and can't reproduce. It could be related only to STM32F7 boards.

closed 2026-02-17 by github-actions[bot]