PX4/PX4-Autopilot #23880: shutdown command is not completely killing the processes.

rajeev-gupta-bashrc, 2024-11-01:
### Describe problem solved by the proposed feature

In the gazebo simulation the shutdown command doesn't kills the gz server, and the gazebo gui remains open with the terminal being unresponsive. If we further kill the background processes by ctrl+c then gazebo doesn't open for the next time. We need to restart the pc (this restarting issue is related to gazebo/gz-server if we don't properly kill the gz-server). 

### Describe your preferred solution

I've added a new "kill command in px4 to kill the entire process. 

### Describe possible alternatives

We can also make changes in the shutdown command itself, but for our current purpose, I found the "kill" command more straight forward and easy.

rajeev-gupta-bashrc, 2024-11-01:
Solved using this:
https://github.com/PX4/PX4-Autopilot/pull/23881/commits/598e921788ea7243c11afc31876e92f035515bd1

dakejahl, 2024-11-05:
why not just use ctrl+c?

rajeev-gupta-bashrc, 2024-11-06:
@dakejahl actually we wanted to automate the launch and shut down process, that's why ctrl+c wasn't an option for us. We feed the kill command and it kills the process.
Thank you for looking into it. If you can suggest some changes, we welcome you.

dakejahl, 2024-11-06:
@rajeev-gupta-bashrc if you're automating the launch/shutdown you can just kill the process from your own scripts? Ctrl+C is just sending SIGINT. I don't think adding a pkill -f to the cpp code is the right place to do it.