PX4/PX4-Autopilot #23904: [Bug] "Operation timeout, aborting transfer" alert on flight controller board with no SD card slot

HB-SP-P, 2024-11-07:
### Describe the bug

When PX4 does not detect an SD card present, it will give a popup alert on QGC: "Operation timeout, aborting transfer". 
The alert also happens to board like Kakute H7 mini, which does not have a SD card slot, and the alert is preventing user from configuring the flight controller.
This issue only happens on version 1.15 and later.

### To Reproduce

1. Update to latest stable or master PX4 firmware.
2. Wait for QGC connect to flight controller board without SD card slot.

### Expected behavior

No "Operation timeout, aborting transfer" alert

### Software Version

1.15.1 stable

### Flight controller

Kakute H7 mini

### Vehicle type

None

JOKERBING, 2024-11-14:
I also encountered the same problem using the main version

JOKERBING, 2024-11-19:
I have solved this problem. By configuring the defconfig file, I used FLASH on the SPI bus, turned off VNet MMCSD_SDIO, and turned on VNet MMCSD_SPI.

HB-SP-P, 2024-11-20:
I tried the lates stable release on the Kakute H7 mini, and the warning no longer appears. I guess it is fixed?

nb-programmer, 2024-11-30:
It seems to happen to me constantly since v1.15.x with the `px4_fmu-v3_default` firmware image, regardless of whether SD card present or not. I can't save any config and overall nothing seems to work.

I had to downgrade to `v1.14.2` image by building it myself (not sure if v1.14.4 works).

For anyone's reference (just following the [CI pipeline](https://github.com/PX4/PX4-Autopilot/blob/968732477803be8a73edd13a29c8ab71458eb354/.ci/Jenkinsfile-compile#L188)):
1. `docker run -it --rm px4io/px4-dev-nuttx-focal:2022-08-12`
2. `cd ~ && git clone --recurse-submodules --branch v1.14.2 https://github.com/PX4/PX4-Autopilot && cd PX4-Autopilot/`
3. `make distclean; git clean -ff -x -d .`
4. `git fetch --tags`
5. `ccache -s`
6. `make px4_fmu-v3_default`

Flash the `/root/PX4-Autopilot/build/px4_fmu-v3_default/px4_fmu-v3_default.px4` file using "Custom file" option in QGroundControl.