PX4/PX4-Autopilot #23911: [Bug] offboard control send force/torque the UAV jittering

ZCXSNdie, 2024-11-08:
### Describe the bug

when I used mavros topic-actuator_controls to send 3D force and torque massage to a fully actuated UAV , found the body twitches irregularly , especially on pitch. I have changed the frequency of posting topics from 200 to 400HZ , but I don't feel it has any effect.

### To Reproduce

1.UAV takes off in position control mode ,switch to offboard mode 

### Expected behavior

Hardware issuses have been ruled out, as using low-level P-PID control for attitude is very stable. The expected effect is stable.

### Screenshot / Media

[image]

### Flight Log

 The flight log is as follows :https://review.px4.io/plot_app?log=a1c7da7c-7128-4278-a518-32777c865fed.

### Software Version

px4 1.11

### Vehicle type

Multicopter

ZCXSNdie, 2024-11-08:
@Jaeyoung-Lim I have learned a lot from your answer before , and I would greatly appreciate it if you could reply to me .

qtweng, 2024-11-22:
Likewise, I have observed twitching in manual mode when publishing ActuatorServos and OffboardControlMode.

Jaeyoung-Lim, 2024-11-25:
@ZCXSNdie How does the hardware setup look like? Sending a mavros topic at 400Hz on the companion computer doesn't mean that the autopilot is receiving the message at 400Hz. 

Note that the rate control onboard the vehicle is running at almost 1kHz. If you do actuator controls offboard, it can be very sensitive to any jitter or delay from the sensor information being sent from the autopilot to the commands being sent to the autopilot.

ZCXSNdie, 2024-11-26:
> @ZCXSNdie How does the hardware setup look like? Sending a mavros topic at 400Hz on the companion computer doesn't mean that the autopilot is receiving the message at 400Hz.
>
> Note that the rate [...]

Thanks for your reply. The hardware uses NX, the serial port uses telem2, and the baud rate is 921600. The frame uses a six-rotor drone with blades tilted 30°. The hardware PX4 version is CUAV V5+.