PX4/PX4-Autopilot #23938: [Bug] modifications on print_status() in ControlAllocator have no effect

JaimeParker, 2024-11-13:
### Describe the bug

after revising custom print in [ControlAllocator.cpp](https://github.com/PX4/PX4-Autopilot/blob/v1.13.0/src/modules/control_allocator/ControlAllocator.cpp#L670) like this:

```c++
if (_actuator_effectiveness != nullptr) {
		PX4_INFO("Effectiveness Source: %s", _actuator_effectiveness->name());
	}

	// Print current effectiveness matrix
	for (int i = 0; i < _num_control_allocation; ++i) {
		const ActuatorEffectiveness::EffectivenessMatrix &effectiveness = _control_allocation[i]->getEffectivenessMatrix();

		if (_num_control_allocation > 1) {
			PX4_INFO("Instance: %i", i);
		}

		PX4_INFO("  custom print test.");
		PX4_INFO("  Effectiveness.T =");
		effectiveness.T().print();
		PX4_INFO("  minimum =");
		_control_allocation[i]->getActuatorMin().T().print();
		PX4_INFO("  maximum =");
		_control_allocation[i]->getActuatorMax().T().print();
		PX4_INFO("  actuator_trim =");
		_control_allocation[i]->getActuatorTrim().T().print();
		PX4_INFO("  control_trim =");
		_control_allocation[i]->getControlTrim().T().print();
		PX4_INFO("  Configured actuators: %i", _control_allocation[i]->numConfiguredActuators());
	}
```

and defining functions in [ControlAllocation.hpp](https://github.com/PX4/PX4-Autopilot/blob/6823cbc4140e29568f00e1211ae60e057adb1a1f/src/modules/control_allocator/ControlAllocation/ControlAllocation.hpp):

```c++
        const matrix::Vector<float, NUM_ACTUATORS> &getActuatorTrim() const { return _actuator_trim; }
	const matrix::Vector<float, NUM_AXES> &getControlTrim() const { return _control_trim; }
```

then,

* run `make clean`
* run `make px4_sitl gazebo`
* revise the [mavros_posix_sitl.launch](https://github.com/PX4/PX4-Autopilot/blob/6823cbc4140e29568f00e1211ae60e057adb1a1f/launch/mavros_posix_sitl.launch#L14)

```xml
<arg name="vehicle" default="iris_ctrlalloc"/>
```

and run `roslaunch px4 mavros_posix_sitl.launch`

then in the px4 terminal, run `commander takeoff` and `control_allocator status`

will got no new added message printed, even the string.

### To Reproduce

* using PX4 1.13.0
* revise ControlAllocator.cpp and ControlAllocation.hpp like above
* revise the launch file to use iris allocation model
* roslaunch
* run control_allocator status to check

### Expected behavior

at least the added:

```c++
PX4_INFO("  custom print test.");
```

should be printed

### Screenshot / Media

[image]

[image]

### Software Version

PX4 1.13.0

### Flight controller

px4 sitl gazebo

### Vehicle type

Multicopter

### Additional context

if u think this should not be a [bug] issue, welcome to discuss in the [forum](https://discuss.px4.io/t/why-do-modifications-to-the-print-status-function-in-controlallocator-have-no-effect/42409)

dakejahl, 2024-11-18:
The code isn't getting recompiled. Try deleting the build directory 
```
make clean && rm -rf build
```
make sure you're running the same code that you're building, that you don't accidentally have two px4 repositories downloaded.

JaimeParker, 2024-11-18:
@dakejahl Thanks for that! I double checked that I have only one PX4 repo, and run

```shell
make clean && rm -rf build
```

the whole build folder is removed, and the next `make px4_sitl gazbeo` took longer time, but it still failed, without any custom message printed.

dakejahl, 2024-11-18:
then your changes must not be getting saved, check `git status`

JaimeParker, 2024-11-18:
@dakejahl weird, it do has modificaiton

[image]

actually I just want to check the variables in [ControlAllocationPseudoInverse allocate](https://github.com/PX4/PX4-Autopilot/blob/v1.13.0/src/modules/control_allocator/ControlAllocation/ControlAllocationPseudoInverse.cpp#L180)

```c++
// Allocate
_actuator_sp = _actuator_trim + _mix * (_control_sp - _control_trim);
```

for multirotor, I checked that there is no set for `_actuator_trim` and `_control_trim`, so all zeros vector, and I run a python script to validate the control allocator process:

```python
import numpy as np

# Constants
NUM_AXES = 6
NUM_ACTUATORS = 16
THRUST_Z = 5  # Index for Z-thrust in control allocation scale

# Fixed rotor axis (upwards in NED frame)
axis = np.array([0, 0, -1])

# Rotor parameters
rotor_positions = np.array([
    [0.13, 0.22, 0.0],  # Rotor 0
    [-0.13, -0.20, 0.0],  # Rotor 1
    [0.13, -0.22, 0.0],  # Rotor 2
    [-0.13, 0.20, 0.0],  # Rotor 3
])

moment_ratios = np.array([0.12, 0.12, -0.12, -0.12])
thrust_coefficients = np.ones(4) * 6.925

# Initialize the effectiveness matrix
effectiveness_matrix = np.zeros((NUM_AXES, NUM_ACTUATORS))

# Populate the effectiveness matrix
for i, (position, km, ct) in enumerate(zip(rotor_positions, moment_ratios, thrust_coefficients)):
    thrust = ct * axis
    moment = ct * np.cross(position, axis) - ct * km * axis

    effectiveness_matrix[:3, i] = moment  # Rows 0-2: moments
    effectiveness_matrix[3:, i] = thrust  # Rows 3-5: thrusts

# Display the effectiveness matrix
print("Effectiveness Matrix:")
print(effectiveness_matrix[:6, :6])

print("\n Effectiveness Matrix.T:")
print(effectiveness_matrix[:6, :6].T)

# Compute the pseudo-inverse (mix matrix)
mix_matrix = np.linalg.pinv(effectiveness_matrix)

print("\n Mix Matrix:")
print(mix_matrix[:6, :6])

control_allocation_scale = np.ones(NUM_AXES)

# Update control allocation matrix scale
def update_control_allocation_matrix_scale(mix_matrix, control_allocation_scale):
    # Roll and pitch scaling
    roll_torque_nonzero = np.abs(mix_matrix[:, 0]) > 1e-3
    pitch_torque_nonzero = np.abs(mix_matrix[:, 1]) > 1e-3

    roll_norm_scale = np.sqrt(np.sum(mix_matrix[roll_torque_nonzero, 0] ** 2) / max(1, np.sum(roll_torque_nonzero) / 2))
    pitch_norm_scale = np.sqrt(np.sum(mix_matrix[pitch_torque_nonzero, 1] ** 2) / max(1, np.sum(pitch_torque_nonzero) / 2))

    control_allocation_scale[0] = max(roll_norm_scale, pitch_norm_scale)
    control_allocation_scale[1] = control_allocation_scale[0]

    # Yaw scaling
    control_allocation_scale[2] = np.max(np.abs(mix_matrix[:, 2]))

    # Thrust scaling (X, Y, Z)
    for axis_idx in range(3):
        thrust_nonzero = np.abs(mix_matrix[:, 3 + axis_idx]) > 1e-3
        norm_sum = np.sum(np.abs(mix_matrix[thrust_nonzero, 3 + axis_idx]))
        control_allocation_scale[3 + axis_idx] = norm_sum / max(1, np.sum(thrust_nonzero))

# Normalize control allocation matrix
def normalize_control_allocation_matrix(mix_matrix, control_allocation_scale):
    for i in range(NUM_AXES):
        if control_allocation_scale[i] > 1e-3:
            mix_matrix[:, i] /= control_allocation_scale[i]

    mix_matrix[np.abs(mix_matrix) < 1e-3] = 0.0

# Perform update and normalization
update_control_allocation_matrix_scale(mix_matrix, control_allocation_scale)
normalize_control_allocation_matrix(mix_matrix, control_allocation_scale)

# Display the updated control allocation scale
print("\nUpdated Control Allocation Scale:")
print(control_allocation_scale)

# Display the normalized mix matrix
print("\nNormalized Mix Matrix:")
print(mix_matrix[:6, :6])

torque_thrust = np.array([1.0, 0.0, 0.0, 0.0, 0.0, 0.5])

actuator_sp_px4 = mix_matrix @ torque_thrust
print("\n Actuator setpoint for PX4:")
print(actuator_sp_px4)

actuator_sp_px4_valid = actuator_sp_px4[:4]

force_px4 = np.zeros(4)
for i in range(4):
    force_px4[i] = (abs(actuator_sp_px4_valid[i]) * 1e4 + 1e3) ** 2 * 5.84 * 1e-6 * 9.8 / 1000
print("\n Force setpoint PX4:")
print(force_px4)
print("Total force setpoint PX4:")
print(np.sum(force_px4))
```

and got

```
 Actuator setpoint for PX4:
[-0.93773277 -0.06226723 -0.06226723 -0.93773277  0.          0.
  0.          0.          0.          0.          0.          0.
  0.          0.          0.          0.        ]

 Force setpoint PX4:
[6.16325284 0.1506956  0.1506956  6.16325284]
Total force setpoint PX4:
12.62789688888889
```

first, I didn't see a `abs(x)` for `actuator_sp`, they will be clipped by default [min, max] = [0, 1].

second, if I use abs(actuastor_sp) for setting pwm for rotor, the torque direction is totally reversed.

JaimeParker, 2024-11-18:
it works only in 

```
make px4_sitl gazebo_iris_ctrlalloc
```

[image]

if using 

```
roslaunch px4 mavros_poxis_sitl.launch
```

all custom messages will not be printed.

sfuhrer, 2025-01-06:
So the issue seems that when doing the roslaunch it doesn't use the current state of your checked out PX4 code but some pre-compiled version? Or what conclusions did you get to?