PX4/PX4-Autopilot #23954: [Bug] Opening Mavlink shell causes uxrce_dds_client to crash

JaySKZ, 2024-11-16:
### Describe the bug

The drone is able to initialize both mavlink and uxrce_dds_client fine. However, the act of opening mavlink shell in QGC (simply pressing "Enter" on the mavlink command line without typing anything) causes uxrce_dds_client to disconnect, and not be able to connect again.

### To Reproduce

1. boot up drone
2. start mavlink on *serial* using command (in rCS or on nsh command line) `mavlink start -d /dev/ttyS4 -b 115200 -x`
3. start uxrce_dds_client on *ethernet* using command (in rCS or on nsh command line) `uxrce_dds_client start -t udp -h 192.168.0.2 -p 8888`
4. uxrce_dds_client starts, time sync converge successfully
5. in QGC, open MAVLink Console and press enter a few times (nsh acknowledges mavlink shell opened)
6. uxrce_dds_client disconnects, re-initializes but never reconnects
7. issue can be replicated when both mavlink and uxrce_dds_client are on serial, and when mavlink is on ethernet and uxrce_dds_client is on serial

### Screenshot / Media

[image]
[image]

### Software Version

PX4 version: 1.15.0 0 (17760256) with custom board files

### Flight controller

Custom hardware

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

MAVLink is routed on serial device /dev/ttyS4 (TELEM2)
UXRCE_DDS_CLIENT is on ethernet

AlexisTM, 2024-11-17:
Does it happen with both on Ethernet? Could it be caused by RAM usage? (We are very high by default)

JaySKZ, 2024-11-18:
Hi @AlexisTM  thank you for your response! Yes, the same issue does happen with both on Ethernet. With both on Ethernet, the act of opening Mavlink shell actually brings down the entire mavlink connection, and the drone does not recover unless the MCU is reset (even calling `mavlink stop` then `mavlink start` does not allow it to connect with QGC again). Again, this is only happening IF mavlink shell is opened in QGC, otherwise the drone behaves fine and both mavlink and uxrce_dds_client behaves nominally.

I checked RAM usage and it is under the limit throughout the entire operation. We are using the STM32H745IIK6 MCU which has quite a bit of RAM.

AlexisTM, 2024-11-18:
Do you have a STLink or JTAG debugger to find the crash?
Could it be related to not calling `mavlink boot_complete` after the mavlink restart?

JaySKZ, 2024-11-18:
Hi @AlexisTM we do indeed have STLink connected and we do call `mavlink boot_complete`

The issue is solved now! The problem was that we already had a nsh console open for debug, and opening mavlink shell called `netinit_bringup()` again in `nsh_consolemain.c` which seems to disconnect uxrce_dds_client. I guess PX4 assumes that at any point only one console would be open which is probably true in almost all cases, but since we are bringing up completely custom hardware we've kept a debug console open over UART.

The fix currently is a bit of hardcode in `nsh_consolemain.c` such that `netinit_bringup()` only gets called once. Would it make sense for there to be a more general solution implemented?

Thank you for your speedy responses!

AlexisTM, 2024-11-18:
I knew starting another nsh in the same nsh link crashed, but I thought it was just an incorrect usage.
I suppose not being able to start two NSH is still a bug. (as we are allowed to start them)

closed 2024-11-23 by JaySKZ