PX4/PX4-Autopilot #23956: EKF is complex more and more

garlinplus, 2024-11-17:
### Describe problem solved by the proposed feature

EKF is reconstructed to eskf,but it's performance maybe worse than before.
When this is no position and velocity fusion mode and there is motion acceleration,the attitude get more fluctuation and the convergence rate is slower.And the code is become more and more.

### Describe your preferred solution

Reconstruct it and tune parameters of it

### Describe possible alternatives

no

bresch, 2024-11-19:
> When this is no position and velocity fusion mode and there is motion acceleration, the attitude get more fluctuation and the convergence rate is slower

Could you give a bit more details please? 
Do you maybe have a log? 
Are you using 1.15? 
Do you mean that the attitude drifts when flying in stabilized mode without GNSS fusion? 
Are you using a mag?

Many things changes, not only the derivation. The error-state filter is far superior and is unlikely to be the cause of the issue. We removed a ton of small hacks that were over-constraining the natural evolution of the EKF, maybe one of those was actually useful in some specific cases.

dagar, 2024-11-20:
Sharing logs for comparison would be quite helpful.

closed 2026-01-31 by github-actions[bot]