PX4/PX4-Autopilot #23963: [Bug] PX4 doesn't have a default compiler anymore

PetervdPerk-NXP, 2024-11-18:
### Describe the bug

#23869 removes the installation of GCC 9.3.1
Instead install OS provided GCC.
Now we're all using different compilers

| Host         | Compiler                          |
|--------------|-----------------------------------|
| Ubuntu 24.04 | GCC 13.2                          |
| Ubuntu 22.04 | GCC 10.3                          |
| Ubuntu 20.04 | GCC 9-2029-q4                     |
| CI           | Logs show GCC 9.3.1 |
| Any machine that used old ubuntu.sh           | GCC 9.3.1 |

https://launchpad.net/ubuntu/+source/gcc-arm-none-eabi

### Vehicle type

None

dagar, 2024-11-18:
What I would like to do going forward is have the official/supported compiler version that we use for CI and to produce QGC flashable builds is to follow the current Ubuntu LTS. 

Unofficially I would like to keep things working back to the previous LTS version. With newer CI resources we can now afford to do a few builds across a larger matrix of distros + versions. However if/when someone hits an issue we need to point them at the supported (current LTS) environment.

PetervdPerk-NXP, 2024-11-18:
Wouldn't it just be easier to select a compiler that everyone should use?
I think you can install the ARM release of the GCC 13.2 on Ubuntu 22.04 as well.
So that would ensure that everyone is using GCC 13.2.

Reason I'm bringing this up because on a clean Ubuntu 24.40 it will use GCC 13.2 which fails to compile on various targets.

Also do we really want to go into runtime errors caused by different compiler versions?

dagar, 2024-11-18:
We're still choosing a version that CI and production builds will use and what I will strongly recommend people use. The difference is we won't make a mess of your system trying to force an exact version of something. 

Another piece to be aware of is that it's more than just a specific GCC version. There are different edge cases with cmake versions, python pain, and then other random tools. 

The point of this is to try and be more tolerant of a wider array of setups. There have always been many people using Ubuntu versions in between an LTS, or Fedora, RHEL, Arch, etc, etc. They hit super minor things like compiler warnings or other annoyances, some of them get frustrated and waste a lot of time, others end up hacking these fixes in private and power through. Let's try to minimize the barrier to entry for PX4 development, because whether we like it or not people are doing it regardless.

> Reason I'm bringing this up because on a clean Ubuntu 24.04 it will use GCC 13.2 which fails to compile on various targets.

It didn't happen the way I would have preferred, but things are in flux at the moment and will soon be reconciled with everything on 24.04 with packaged GCC (clean simple setup script, default px4-dev container, CI and QGC builds). On the side we'll have a minimal set of builds across all Ubuntu versions (and ideally other distros next).

PetervdPerk-NXP, 2024-11-18:
> We're still choosing a version that CI and production builds will use and what I will strongly recommend people use. The difference is we won't make a mess of your system trying to force an exact version of something.

Okay that sounds reasonable to me, it's just that at the moment it's quite vague hence this issue whereas before we were fixed on GCC 9.3.1.

PetervdPerk-NXP, 2024-11-18:
Some compilation errors with GCC 13.2 triggering on werror i.e.
```
/home/peter/src/PX4-Autopilot/platforms/common/spi.cpp: In function 'int px4_find_spi_bus(uint32_t)':
/home/peter/src/PX4-Autopilot/platforms/common/spi.cpp:90:59: error: the address of 'px4_spi_buses' will never be NULL [-Werror=address]
   90 |         for (int i = 0; ((px4_spi_bus_t *) px4_spi_buses) != nullptr && i < SPI_BUS_MAX_BUS_ITEMS; ++i) {
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors
[536/1335] Building CXX object platforms/common/uORB/CMakeFiles/uORB.dir/uORB.cpp.obj
ninja: build stopped: subcommand failed.
make: *** [Makefile:232: px4_fmu-v4] Error 1
peter@SMW015329:~/src/PX4-Autopilot$
```

alexcekay, 2024-11-20:
Just a quick note. I have been using GCC 13 since months as I did not want to use the old GCC 9.
Apart from the `spi.cpp` problem described by @PetervdPerk-NXP I did not have to do any changes.

GCC 13 also has the advantage of using less FLASH for the same build.
For example https://github.com/Auterion/PX4_firmware_private/actions/runs/11930030958/job/33250019827?pr=2707 fails with GCC9, but runs fine with GCC13 locally as it uses 10K less FLASH.

MaEtUgR, 2024-11-20:
> Some compilation errors with GCC 13.2 triggering on werror i.e.

https://github.com/PX4/PX4-Autopilot/pull/23994

PetervdPerk-NXP, 2025-01-03:
@dagar can we pick a new number/version of GCC going forward? 

It seems that V6X-RT has some weird startup issues with the newer GCC compilers (10.3, 13.2) whereas GCC 9.3.1 works fine. I want to look into this issue, however I don't want to debug all potential compiler versions, would be nice to bump the version of GCC and match CI/Releases with that.

alexcekay, 2025-01-08:
Just wanted to quickly add a few more comparisons between the GCC9 used by the CI and a local GCC13.3.1.
I think it would really help us with the FLASH problems we are facing.

**FLASH (v5x)**
Here is the output from bloaty (`bloaty -s vm -d sections v5x-gcc13.elf -- v5x-gcc-ci.elf`).
```
    FILE SIZE        VM SIZE    
 --------------  -------------- 
  +0.2%      +7  +0.2%      +7    .data
  -9.4%      -5  [ = ]       0    .ARM.attributes
  -9.2%      -7  [ = ]       0    .comment
  -3.3% -64.4Ki  [ = ]       0    .debug_abbrev
  +2.0% +3.20Ki  [ = ]       0    .debug_aranges
  +2.6% +12.1Ki  [ = ]       0    .debug_frame
  -2.7%  -737Ki  [ = ]       0    .debug_info
   +15%  +650Ki  [ = ]       0    .debug_line
  [NEW]    +423  [ = ]       0    .debug_line_str
  [DEL] -5.35Mi  [ = ]       0    .debug_loc
  [NEW] +3.46Mi  [ = ]       0    .debug_loclists
  [DEL] -1.25Mi  [ = ]       0    .debug_ranges
  [NEW]  +579Ki  [ = ]       0    .debug_rnglists
  +2.5% +86.4Ki  [ = ]       0    .debug_str
   +11%     +23  [ = ]       0    .shstrtab
 -102.2% -27.8Ki  [ = ]       0    [4 Others]
  [ = ]       0  [DEL]     -52    [ELF Header]
  [ = ]       0  -0.1%     -64    .bss
  [ = ]       0  [DEL]    -128    [ELF Program Headers]
  [DEL] -31.8Ki  [DEL] -31.8Ki    [LOAD #1 [RWX]]
  -2.1% -41.2Ki  -2.1% -41.2Ki    .text
  -5.8% -2.73Mi  -3.5% -73.3Ki    TOTAL
```

**FLASH (v6x)**
Here is the output from bloaty (`bloaty -s vm -d sections v6x-gcc13.elf -- v6x-gcc-ci.elf`).
```
    FILE SIZE        VM SIZE    
 --------------  -------------- 
  -9.4%      -5  [ = ]       0    .ARM.attributes
  -9.2%      -7  [ = ]       0    .comment
  -3.4% -64.6Ki  [ = ]       0    .debug_abbrev
  +2.1% +3.20Ki  [ = ]       0    .debug_aranges
  +2.6% +11.8Ki  [ = ]       0    .debug_frame
  -2.8%  -731Ki  [ = ]       0    .debug_info
   +15%  +618Ki  [ = ]       0    .debug_line
  [NEW]    +423  [ = ]       0    .debug_line_str
  [DEL] -5.14Mi  [ = ]       0    .debug_loc
  [NEW] +3.32Mi  [ = ]       0    .debug_loclists
  [DEL] -1.19Mi  [ = ]       0    .debug_ranges
  [NEW]  +553Ki  [ = ]       0    .debug_rnglists
  +2.5% +85.2Ki  [ = ]       0    .debug_str
   +10%     +22  [ = ]       0    .shstrtab
  +0.7% +4.25Ki  [ = ]       0    .strtab
  -3.3% -19.3Ki  [ = ]       0    .symtab
  +5.0%     +40  [ = ]       0    [ELF Section Headers]
 -93.9% -68.6Ki  [ = ]       0    [Unmapped]
  [ = ]       0  -0.1%     -64    .bss
  -2.1% -39.4Ki  -2.1% -39.4Ki    .text
  -5.9% -2.67Mi  -2.0% -39.4Ki    TOTAL
```

**CPU LOAD (v5x)**
Test setup: v5x running on the bench and armed.
```
GCC9 max CPU load:  0.749
GCC13 max CPU load: 0.724
```
The results are very similar here, indicating that the FLASH savings are not gained by a space-time tradeoff.

mrpollo, 2025-04-30:
As discussed on the dev call today (April 30th, 2025)  once we properly fork the v1.16 release, we are going to be updating to a new strategy to manage GCC dependency.

1. Set gcc to whatever the ubuntu LTS ships via apt-get
2. Update containers and CI to use the containers with updated GCC
3. Append build instructions on our docs to call out what we the setup is for building stable releases via CI

closed 2026-02-13 by mrpollo