PX4/PX4-Autopilot #23965: PX4 SITL Tests Improvements

mrpollo, 2024-11-18:
We need to improve the current SITL tests. The current architecture is becoming stale and needs work, resulting in flaky tests and developers ignoring GitHub actions altogether.

TODO
- Update architecture to use the latest dependencies, Base OS, PX4, and MAVSDK
- Migrate to the latest Gazebo (not classic)
- Document how SITL tests work in Github Actions

mrpollo, 2024-11-21:
Yes, @MaEtUgR, we are changing gears here and started working on fixing the tests after we fixed the noisy issues that were hiding the real problems

dagar, 2024-11-21:
This one looks real, during RTL there's a bogus "Pilot took over using sticks" and we have a fly away.

standard_vtol: 'RTL with Mission Landing': failed
https://github.com/PX4/PX4-Autopilot/actions/runs/11957851318/job/33335973614

https://logs.px4.io/plot_app?log=e0673b3c-8514-4390-b3ee-d00a317b17e3

[image]

MaEtUgR, 2024-11-21:
A shot in the dark. Probably not the issue but might give us more insights also if it only rules out one thing: https://github.com/PX4/PX4-Autopilot/pull/24017

MaEtUgR, 2024-11-21:
@dagar Regarding pilot took over I'm confused. Plotjuggler doesn't show this message (event?). As expected there's no valid stick input and also never a sticks moving flag. The mode never changes to Position or indicates switching to Position/Altitude is not possible but that would be what "Pilot took over" stands for.
[image]

mrpollo, 2025-05-07:
Relevant https://github.com/PX4/PX4-Autopilot/pull/24237