PX4/PX4-Autopilot #24033: [Bug] WARN  [health_and_arming_checks] Preflight Fail: ekf2 missing data

anhducad1111, 2024-11-25:
### Describe the bug

[image]
it my first setup with px4 simulation https://docs.px4.io/main/en/sim_gazebo_gz/
cd ~
git clone https://github.com/PX4/PX4-Autopilot.git --recursive
bash ./PX4-Autopilot/Tools/setup/ubuntu.sh
cd PX4-Autopilot/
make px4_sitl

Installed ROS 2
installed and ran Micro XRCE-DDS Agent & Client
Built ROS 2 Workspace
Installed MAVSDK

### Expected behavior

ready to takeoff

### Flight Log

[0/1015] git submodule src/drivers/gps/devices
[8/1015] git submodule src/modules/mavlink/mavlink
[11/1015] git submodule src/modules/uxrce_dds_client/Micro-XRCE-DDS-Client
[1014/1015] cd /home/ad/PX4-Autopilot/build/...PX4-Autopilot/build/px4_sitl_default/bin/px4

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=4002
INFO  [param] selected parameter default file parameters.bson
INFO  [param] selected parameter backup file parameters_backup.bson
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 4002
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG0_PRIO: curr: -1 -> new: 50
  CAL_MAG1_ID: curr: 0 -> new: 197644
  CAL_MAG1_PRIO: curr: -1 -> new: 50
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
INFO  [dataman] data manager file './dataman' size is 7872608 bytes
INFO  [init] Gazebo simulator
INFO  [init] gazebo already running world: default
WARN  [init] PX4_GZ_MODEL_POSE not set, spawning at origin.
INFO  [gz_bridge] world: default, model name: x500_depth_0, simulation model: x500_depth
INFO  [gz_bridge] Requested Model Position: 0,0,0,0,0,0
INFO  [gz_bridge] Model position z is less or equal 0.0, moving upwards
INFO  [lockstep_scheduler] setting initial absolute time to 2190908000 us
INFO  [commander] LED: open /dev/led0 failed (22)
WARN  [health_and_arming_checks] Preflight Fail: ekf2 missing data
WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station
INFO  [tone_alarm] home set
INFO  [uxrce_dds_client] init UDP agent IP:127.0.0.1, port:8888
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] partner IP: 127.0.0.1
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2024-11-25/10_05_03.ulg
INFO  [logger] Opened full log file: ./log/2024-11-25/10_05_03.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> WARN  [health_and_arming_checks] Preflight Fail: Attitude failure (roll)
WARN  [health_and_arming_checks] Preflight Fail: Attitude failure (roll)
INFO  [uxrce_dds_client] synchronized with time offset 1732526911822399us
INFO  [uxrce_dds_client] successfully created rt/fmu/out/battery_status data writer, topic id: 19
INFO  [uxrce_dds_client] successfully created rt/fmu/out/estimator_status_flags data writer, topic id: 82
INFO  [uxrce_dds_client] successfully created rt/fmu/out/failsafe_flags data writer, topic id: 88
INFO  [uxrce_dds_client] successfully created rt/fmu/out/manual_control_setpoint data writer, topic id: 137
INFO  [uxrce_dds_client] successfully created rt/fmu/out/position_setpoint_triplet data writer, topic id: 181INFO  [uxrce_dds_client] successfully created rt/fmu/out/sensor_combined data writer, topic id: 213
INFO  [uxrce_dds_client] successfully created rt/fmu/out/timesync_status data writer, topic id: 236
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_land_detected data writer, topic id: 265
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_attitude data writer, topic id: 252
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_control_mode data writer, topic id: 259
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_global_position data writer, topic id: 260
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_gps_position data writer, topic id: 262
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_local_position data writer, topic id: 266
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_odometry data writer, topic id: 271
INFO  [uxrce_dds_client] successfully created rt/fmu/out/vehicle_status data writer, topic id: 276
INFO  [uxrce_dds_client] time sync converged
WARN  [timesync] time jump detected. Resetting time synchroniser.
WARN  [uxrce_dds_client] time sync no longer converged
INFO  [uxrce_dds_client] time sync converged

### Software Version

newest 25/11/2024 

### Vehicle type

Multicopter

### Additional context

source ~/px4-venv/bin/activate

cd ~/Micro-XRCE-DDS-Agent
MicroXRCEAgent udp4 -p 8888

source ~/.bashrc

cd ~/PX4-Autopilot

make px4_sitl gz_x500_depth
make clean

ros2 rubashrcn ros_gz_image image_bridge /camera

source ~/px4-venv/bin/activate
cd ~/PX4-ROS2-Gazebo-YOLOv8
python uav_camera_det.py

ishivarana, 2024-11-26:
This is not a bug. I came across the same problem, which appears due to a wrong installation of Gazebo. For example, either both Harmonic and Garden are installed, or mismatched versions of Gazebo are present on your system.

To resolve this, install the correct version of Gazebo if it's not present by running the following command:

```
 sudo apt install gz-garden
```
Then, install the correct version of the Gazebo-ROS bridge using the command:

```  
sudo apt install ros-humble-ros-gzgarden
```
Next, uninstall any incorrect version of Gazebo that is installed on your system. For example:
```
 sudo apt remove gz-harmonic 
 sudo apt autoremove
```
Finally, reboot your system. The missing EKF2 problem should now be resolved.

anhducad1111, 2024-11-27:
[image]
now  i have this instead
https://github.com/monemati/PX4-ROS2-Gazebo-YOLOv8
I do this step by step
and ur guide
pxh> WARN  [health_and_arming_checks] Preflight Fail: Compass Sensor 0 missing
im already source ~/.bashrc

anhducad1111, 2024-11-27:
I realized that if I follow https://github.com/monemati/PX4-ROS2-Gazebo-YOLOv8/tree/main
to step
PX4_SYS_AUTOSTART=4002 PX4_GZ_MODEL_POSE="268.08,-128.22,3.86,0.00,0,-0.7" PX4_GZ_MODEL=x500_depth ./build/px4_sitl_default/bin/px4
it have WARN  [health_and_arming_checks] Preflight Fail: Compass Sensor 0 missing

but when i run in make px4_sitl gz_x500_depth_windy
it normal
i tried to install uninstall garden or harmonic, tried to use 1 of them but cant fix that

dagar, 2024-11-27:
For some of these things it's a bit of a race condition where commander really shouldn't be complaining about "ekf2 missing data" immediately. I'll try to adjust it to silence false positives.

antoniosteiger, 2025-10-23:
I am also seeing:

`WARN [health_and_arming_checks] Preflight Fail: ekf2 missing data`
and
`WARN  [health_and_arming_checks] Preflight Fail: No connection to the ground control station`

The latter repeats when I try to start the drone. I am seeing this with the following px4 command:

`make px4_sitl gz_x500`

This is on ros2-jazzy with gazebo harmonic and px4 version 1.16.0. Gazebo opens correctly, the drone model spawns. But the px4_ros_com example does not get the drone to lift off. I just get: 

` INFO [tone_alarm] notify negative`.

Does px4_sitl require a GCS to be running? I do not remember reading about this in the docs. Also uXRCE DDS Client and Agent work fine, lots of topics are being registered.