PX4/PX4-Autopilot #24042: [Bug] RC Input Not Recognized by Pixhawk 6X Running PX4 Pro v1.15.2

Turki-01, 2024-11-26:
### Describe the bug

I am trying to connect my RC system (FrSky X9 Lite S and Archer GR8) to the Pixhawk 6X flight controller with PX4 Pro v1.15.2. The receiver is bound to the transmitter using the ACCESS protocol, and I can see a valid SBUS signal on the oscilloscope at the RC port, but QGroundControl fails to recognize the RC input signal during calibration. Additionally, using the MAVLink console command rc_input status shows that RC input is not running. Manually starting the service still results in the system "searching for signal," despite a visible SBUS signal.

### To Reproduce

1. Power on the Pixhawk 6X connected to a Pixhawk Mini Baseboard.
2. Bind the FrSky X9 Lite S transmitter with the FrSky Archer GR8 receiver (using ACCESS protocol).
3. Connect the receiver's SBUS OUT and Ground pins to the RC port's SBUS IN and Ground on the Pixhawk Mini Baseboard.
4. Power the receiver with 5V from a BEC.
5. Open QGroundControl, flash the firmware (PX4 Pro v1.15.2), and attempt to calibrate the radio in the "Radio" tab.
6. Observe no signal during calibration.
7. Test the RC port with an oscilloscope to verify an SBUS signal.
8. Use the MAVLink console to execute rc_input status and observe the state as "not running" or "searching for signal."

### Expected behavior

The RC signal from the Archer GR8 receiver should be recognized by the Pixhawk 6X, and calibration in QGroundControl should proceed as expected.

### Screenshot / Media

[image]
[image]
[image]
[image]
[image]
[image]

### Flight Log

No flight log is available as the issue occurs during initial setup.

### Software Version

PX4 Pro v1.15.2

### Flight controller

Pixhawk 6X with Pixhawk Mini Baseboard

### Vehicle type

Multicopter

### How are the different components wired up (including port information)

- Receiver (FrSky Archer GR8): 
  - Powered via 5V BEC.
  - SBUS OUT connected to SBUS IN on the RC port of the Pixhawk Mini Baseboard.
  - Ground connected to Ground on the RC port.

- Oscilloscope:
   - Connected to RC port's SBUS IN and Ground pins to verify signal presence.

### Additional context

- Tested the system with another receiver (FrSky RX6R) and simulated a PPM signal with Arduino Uno. Both cases produced valid signals on the oscilloscope, but no recognition by the Pixhawk 6X.

- Used MAVLink console commands to check the RC port:
  - rc_input status initially returned "not running."
  - Starting the RC input service on ttyS5 showed "searching for signal," despite a valid SBUS signal.

- Verified hardware functionality and firmware updates:
  - Receiver firmware: Archer 2.1.14
  - Transmitter firmware: ISRM-X 2.2.10

I suspect the issue might be related to configuration or firmware handling of the RC port. Any guidance or resolution would be greatly appreciated.

julianoes, 2025-04-09:
Do you have PX4IO enabled?

Check this issue:
https://github.com/PX4/PX4-Autopilot/issues/21388

KarthiAru, 2025-06-15:
I'm facing the same issue. More details here - https://discord.com/channels/1022170275984457759/1022185721450213396/1383434246688866345

@julianoes The `SYS_USE_IO` does seem to be deprecated from v.1.15.0. Is there any other config changes required to make this work?

KarthiAru, 2025-06-15:
Issue got resolved for me. 
What happened was that I had copied a reference board folder and made some changes to it a while ago, but I don't remember which version it was based on. Recently, I was working with both v1.15.4 and the latest main branch, so the files in the src folder were likely out of sync, which probably caused the issue. This time, I started fresh from v1.15.4, and everything worked correctly.