PX4/PX4-Autopilot #24060: No response from 192.168.2.115 with STM32H765II, with LAN8742

Abner0907, 2024-12-02:
### Describe the bug

@davids5 @dagar @julianoes   hello, can you help me with the private debug with LAN8742 ETHERNET. 
I have use a private board which is similar as px4_fmu_v6x_default.
now my code is rebased on the v1.12.3.  
I am appreciate that you can tell me how to debug the reason or how to fix the issue. thank you very much.
The lan8742  pcb sch is 
[image]

now, I can get the information 
>nsh> ifconfig
eth0	Link encap:Ethernet HWaddr 2e:89:fa:c3:ff:ff at UP
	inet addr:192.168.2.115 DRaddr:192.168.2.1 Mask:255.255.255.0
lo	Link encap:Local Loopback at UP
	inet addr:127.0.0.1 DRaddr:127.0.0.1 Mask:255.0.0.0

however, when I ping myself 192.18.2.115, it will tell me "No response from s1tm32_txavail_work: ifup: 1"
**the result is outputs**:
>No response from stm321_txavail_work: ifup: 1
sendto_eve9nthandler: flags: 4000
sendto_event2.handler: Send1 ICMP request
sendto_request: Out6going ICMP packet length: 84 (84)
sendto_eventhan8dler: Resuming
stm32_transmit: d_len: 98 d_buf: 0x240085e0 txhead: 0x2.24008fc0 tdes3: 00000000
stm32_transmit: txhead: 0x24008f60 txtail: 0x24008fc0 inflight: 1
stm32_getreg: 40029134->0000d3c2
stm32_putreg: 40029134<-0000d3.c3
stm32_getreg: 40029160->00008405
11stm32_getreg: 40029160->000084055
stm32_getreg: 40029134->0000d3c3
stm32_putreg: 40029160<-00000001
stm32_freeframe: txhead: 0x24008f60 txtail: 0x24008fc0 inflight:: 1
stm32_freeframe: txtail: 0x24008 ifc0 des0: 240085e0 des2: 80000062 des3: 30000000
stm32_getreg: 40029134->0000d3c3
stm32_putreg: 40029134<-0000d3c3
stm32_getreg: 40029134->0000d3c3
stm32_putreg: 40029c134<-0000d3c2
stm32mp_putreg: 40029160<-0000800_0
stm32_getreg: 40029160->00000404
sstm32_getreg: 40029160->00000404
stm32_getreg: 40029134->0000d3c2
eq=6 time=1000 ms

**FYI: I have log the startup log:**

>igmp_initialize: IGMP initializing
stm32_ethinitialize: intf: 0
stm32_ifdown: Taking the network down
stm32_getreg: 58024480->00000000
stm32_putreg: 58024480<-00008000
stm32_putreg: 58024480<-00000000
stm32_getreg: 40029000->00000001
stm32_putreg: 40029000<-00000001
stm32_getreg: 40029000->00000001
stm32_getreg: 40029000->00000000
stm32_getreg: 40029000->00000000
stm32_putreg: 40029000<-00001000
stm32_getreg: 40029004->00000000
stm32_putreg: 40029004<-00001001
igmp_devinit: IGMP initializing dev 0x240082dc
igmp_grpalloc: addr: 010000e0 dev: 0x240082dc
igmp_addmcastmac: Adding: IP 020000e0
igmp_mcastmac: IP: 020000e0 -> MAC: 01005e000002
stm32_addmac: MAC: 01:00:5e:00:00:02
stm32_getreg: 40028010->00000000
stm32_putreg: 40028010<-00800000
stm32_getreg: 40028008->00000000
stm32_putreg: 40028008<-00000404
igmp_addmcastmac: Adding: IP 010000e0
igmp_mcastmac: IP: 010000e0 -> MAC: 01005e000001
[... 165 lines cut ...]
stm32_putreg: 40028090<-00000000
stm32_putreg: 40028050<-00000000
stm32_ethconfig: Enable normal operation
stm32_macaddress: eth0 MAC: 2e:89:fa:c3:ff:ff
stm32_putreg: 40028300<-0000ffff
stm32_putreg: 40028304<-c3fa892e
stm32_getreg: 40028000->00008100
stm32_putreg: 40028000<-00008102
stm32_getreg: 40028d00->0007001a
stm32_putreg: 40028d00<-0007001b
stm32_getreg: 40028000->00008102
stm32_putreg: 40028000<-00008103
stm32_getreg: 40029104->00200010
stm32_putreg: 40029104<-00200011
stm32_getreg: 40029108->002004c0
stm32_putreg: 40029108<-002004c1
stm32_putreg: 400280b4<-00007038
stm32_putreg: 40029134<-0000d3c3
stm32_getreg: 40029160->00000000
stm32_putreg: 40029160<-00000102
NuttShell (NSH) NuttX-10.2.0

### Vehicle type

None

julianoes, 2024-12-02:
As you seem to be creating your own board, and on top of that are on an old PX4 version, I don't think this is a bug in PX4 as such, so I'm closing this issue.

You can message any of the [PX4 consultants](https://px4.io/community/consultants/) to help you with this, or ask in Discord or discuss.px4.io instead.

closed 2024-12-02 by julianoes

Abner0907, 2024-12-02:
@julianoes  tks, solved.