PX4/PX4-Autopilot #24073: [Bug] Gazebo Classic geotagged_cam does not output images on survey mission

tnedev, 2024-12-04:
### Describe the bug

As confirmed in the [PX4-SITL_gazebo-classic issue](https://github.com/PX4/PX4-SITL_gazebo-classic/issues/1059#issuecomment-2516063926)

Using the **Gazebo Classic** with **PX4 1.15.2** and **plane_cam** does not save survey images while doing **survey mission**. 

Manual trigger with camera capture button in QGC saves images to _build/px4_sitl_default/src/modules/simulation/simulator_mavlink/frames/_ .

The bug is confirmed by @hamishwillee and reviewed by @Jaeyoung-Lim who concluded that is not  **PX4-SITL_gazebo-classic** bug but **PX4-Autopilot** issue. 

### Software Version

I'm using PX4 1.15.2 and Gazebo Classic under Ubuntu 20 & Ubuntu 22.

### Vehicle type

None

Jaeyoung-Lim, 2024-12-04:
@tnedev Please post a flight log and instructions to reproduce

tnedev, 2024-12-05:
> @tnedev Please post a flight log and instructions to reproduce

Added them. It's a 400MB log, so I uploaded it to Wetransfer.

Jaeyoung-Lim, 2024-12-06:
I was able to reproduce the problem, and while the camera does trigger, the gazebo plugin fails to grab the mavlink stream coming out of the vehicle during the survey.

closed 2024-12-06 by Jaeyoung-Lim