PX4/PX4-Autopilot #24104: [Bug] Errors compiling v1.15 and main on Ubuntu 20.04

gsoga, 2024-12-13:
### Describe the bug

Ubuntu 20.04
PX4 versions tested: 1.14, 1.15, main
Platform: px4_fmu-v6x

Results:

- v1.14 with `make px4_fmu-v6x_default`: compiles ok

- v1.15 and main with `make px4_fmu-v6x_default`: error related with septentrio

    > PX4-Autopilot/src/drivers/gnss/septentrio/sbf/messages.h:101:10: error: 'septentrio::sbf::Header::id_number' is too small to hold all values of 'enum class septentrio::sbf::BlockID' [-Werror]
  101 |  BlockID id_number: 13;

- I am interested in trying the new rover implementation and I have tested `make px4_fmu-v6x_rover` with v1.15 and with main and I got an additional error:

    > PX4-Autopilot/src/modules/ekf2/EKF/ekf.h:56:10: fatal error: ekf_derivation/generated/state.h: No such file or directory
   56 | #include <ekf_derivation/generated/state.h>

    I checked the directory and it was generated. Again I tried `make px4_fmu-v6x_rover` and got the first error:
    > PX4-Autopilot/src/drivers/gnss/septentrio/sbf/messages.h:101:10: error: 'septentrio::sbf::Header::id_number' is too small to hold all values of 'enum class septentrio::sbf::BlockID' [-Werror]
  101 |  BlockID id_number: 13;

I see that the error related with septentrio has been mentioned in:
https://discuss.px4.io/t/error-septentrio-id-number-is-too-small-to-hold-all-values-of-enum-class-septentrio-blockid-werror-101-blockid-id-number-13/42652
https://discuss.px4.io/t/error-septentrio-id-number-is-too-small-to-hold-all-values-of-enum-class-septentrio-blockid-werror/42255

### Vehicle type

None

Quintin-Z, 2025-02-06:
I also encountered the same problem， It can be solved by upgrading the arm-none-gcc toolchain.
use the script，./Tools/setup/ubunut.sh

samirertt, 2025-02-08:
> I also encountered the same problem， It can be solved by upgrading the arm-none-gcc toolchain. use the script，./Tools/setup/ubunut.sh

nope ./Tools/setup/ubuntu.sh doesnt change anything bro any other idea ?

samirertt, 2025-02-09:
alright it worked bro thanks had to delete the folder at build