PX4/PX4-Autopilot #24141: [Bug] Strange cpu load in arm64 navio2

Lqs66, 2024-12-20:
### Describe the bug

I'm getting the following error on a Raspberry Pi 3b+ on ARM64 with navio2 on board:

Sensor timeout and high cpu load 
```
pxh> sensors status
selected gyro: 4457218 (1)
validator: best: 1, prev best: 1, failsafe: YES (3 events)
sensor #0, prio: 50, state: TOUT ECNT
	val:  -0.0003, lp:  -0.0003 mean dev:  -0.0000 RMS:   0.0032 conf:   0.0000
	val:   0.0045, lp:   0.0033 mean dev:   0.0000 RMS:   0.0016 conf:   0.0000
	val:   0.0039, lp:   0.0006 mean dev:  -0.0000 RMS:   0.0164 conf:   0.0000
sensor #1, prio: 50, state: OK
	val:   0.8032, lp:   0.8044 mean dev:   0.0017 RMS:   0.0646 conf:   1.0000
	val:  -0.0002, lp:  -0.0002 mean dev:   0.0003 RMS:   0.0532 conf:   1.0000
	val:   0.0013, lp:   0.0013 mean dev:   0.0006 RMS:   0.0269 conf:   1.0000

selected accel: 4457218 (1)
validator: best: 1, prev best: 1, failsafe: YES (3 events)
sensor #0, prio: 50, state: TOUT ECNT
	val:   0.1547, lp:   0.1386 mean dev:  -0.0000 RMS:   0.0135 conf:   0.0000
	val:  -0.2858, lp:  -0.2880 mean dev:  -0.0000 RMS:   0.0714 conf:   0.0000
	val:  -9.5208, lp:  -9.2159 mean dev:  -0.0003 RMS:   0.2704 conf:   0.0000
sensor #1, prio: 50, state: OK
	val:   0.1532, lp:   0.1357 mean dev:   0.0002 RMS:   0.0112 conf:   1.0000
	val:   0.0955, lp:   0.1025 mean dev:   0.0002 RMS:   0.0093 conf:   1.0000
	val:  -9.8962, lp:  -9.8894 mean dev:  -0.0170 RMS:   0.2841 conf:   1.0000

[vehicle_magnetometer] selected MAG: 4522498 (0)
validator: best: 0, prev best: 0, failsafe: NO (0 events)
sensor #0, prio: 50, state: OK
	val:   0.0128, lp:   0.0098 mean dev:  -0.0001 RMS:   0.0051 conf:   1.0000
	val:  -0.3526, lp:  -0.3538 mean dev:   0.0001 RMS:   0.0056 conf:   1.0000
	val:  -0.0650, lp:  -0.0722 mean dev:   0.0000 RMS:   0.0046 conf:   1.0000
MAG 4522498 EN: 1, offset: [0.000 0.000 0.000], scale: [1.000 1.000 1.000], Internal

[vehicle_air_data] selected BARO: 4028169 (0)
validator: best: 0, prev best: 0, failsafe: NO (0 events)
sensor #0, prio: 75, state: OK
	val: 97236.0000, lp: 97242.3594 mean dev:  -0.0021 RMS:   3.5231 conf:   1.0000
	val:  42.0900, lp:  42.0833 mean dev:   0.0094 RMS:   0.0092 conf:   1.0000
	val: 346.1333, lp: 345.5956 mean dev:  -0.0078 RMS:   0.3032 conf:   1.0000
BARO 4028169 EN: 1, offset: 0.000, Ext
```

[image]

### Vehicle type

None

dagar, 2025-01-15:
Can you configure htop to show thread names?

dagar, 2025-01-15:
If you share the output of `uorb top` that might also be a clue.

jk213509, 2025-04-06:
I wonder if this is related to #19358