PX4/PX4-Autopilot #24145: [Bug] quad takeoff with flow only (no gps) does not work

dakejahl, 2024-12-20:
### Describe the bug

Quads with optical flow can't seem to reliably takeoff (get stuck on the ground) and if they do manage to takeoff end up at the wrong altitude. It seems that the Local Position Z estimate is being miscalculated. The drone will be sitting on the ground with props spinning, and the local position Z continues to increase.

I've opened a PR here in hopes that someone can help test this
https://github.com/PX4/PX4-Autopilot/pull/24144

### To Reproduce

```
make px4_sitl_default gz_x500_flow_aruco
```
Issue a takeoff command
```
pxh> commander takeoff
pxh> INFO  [commander] Armed by internal command	
INFO  [tone_alarm] arming warning
INFO  [navigator] Using default takeoff altitude: 2.5 m	
INFO  [commander] Takeoff detected
```
Watch the vehicle stay on the ground

### Expected behavior

The vehicle should takeoff to 2.5m

### Flight Log

https://review.px4.io/plot_app?log=40b7d570-33af-4225-ac11-927feea2aa94

### Software Version

https://github.com/PX4/PX4-Autopilot/pull/24144

### Vehicle type

Multicopter

patrikpordi, 2024-12-21:
I experienced the same on hardware (dexi quads with ARK Pi6X Flow), if the setpoint is below ~2m the drone is not able to takeoff, when it does it overshoots and then slowly wanders down almost to the ground with some random jumps.
Drone is not able to takeoff:
https://review.px4.io/plot_app?log=8c200263-a86e-40f5-882c-67180bf274ff
https://review.px4.io/plot_app?log=24083308-d27c-455c-a004-ffce9d693c28

Drone is not able to hold altitude:
https://review.px4.io/plot_app?log=515fd8e7-8d49-4860-961a-7d6925bca648
https://review.px4.io/plot_app?log=e9a25a26-9fdb-4d4f-b47b-6ead5d60c23f
https://review.px4.io/plot_app?log=3fbafe76-7a5e-49fd-b80d-6e06f1a7ec43

dakejahl, 2024-12-21:
Turns out this issue is due to the X500 model in sim spawning 0.25m above the ground and dropping into place. The physical issue @patrikpordi is reporting looks the same, but must be different.

Here is the fix for sim, this solves the issue I reported above
https://github.com/PX4/PX4-gazebo-models/pull/72

closed 2025-01-15 by dakejahl