PX4/PX4-Autopilot #24149: terrain following in fixed-wing mode

xdwgood, 2024-12-23:
### Describe problem solved by the proposed feature

Currently PX4 does not support terrain following in fixed-wing mode? Does anyone know what I should do to support terrain following in fixed-wing mode? thank you

### Describe your preferred solution

no

### Describe possible alternatives

no

Jaeyoung-Lim, 2024-12-23:
@xdwgood Due to the dynamic constraints of fixed-wing vehicles, terrain following is not that trivial.

Therefore,  I have implemented a terrain navigation method, which maintains a certain distance to the terrain.

You can find the repository here: https://github.com/ethz-asl/terrain-navigation

xdwgood, 2024-12-24:
> Hi @xdwgood
> 
> Did you mean something similar to these images? If so, you can set it up in the QGC, but you must enter the climb/descent rate that corresponds to your UAV performance.
> 
> ![ter2](https://private-user-images.githubusercontent.com/12409718/398175653-b7a75cc3-b74c-4621-8e77-fda8a0681daf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUwMDQ5NTIsIm5iZiI6MTczNTAwNDY1MiwicGF0aCI6Ii8xMjQwOTcxOC8zOTgxNzU2NTMtYjdhNzVjYzMtYjc0Yy00NjIxLThlNzctZmRhOGEwNjgxZGFmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjI0VDAxNDQxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY3NjdhMTkzM2MyNWJiMGEwMGQ0MWZlZDg1ODZmZjI0NjlhMTk0NWI5YjUwZTMwNGI3MDFiN2VmNDE1YTI0OTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.AUPfRI0E3FDdg-oH7-YZENgzFcVi22IqtuVy6-QNIic)
> 
> ![ter2](https://private-user-images.githubusercontent.com/12409718/398175542-052cdf74-ce68-43ca-88ba-8613cee9e81d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUwMDQ5NTIsIm5iZiI6MTczNTAwNDY1MiwicGF0aCI6Ii8xMjQwOTcxOC8zOTgxNzU1NDItMDUyY2RmNzQtY2U2OC00M2NhLTg4YmEtODYxM2NlZTllODFkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMjQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjI0VDAxNDQxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU0NWMxYjJiNTg2NTYxYTY0ZjllZDg5ZjhmMDRjZTE1ZmZmODE5MGJlMGQ0M2ZmMzNlZDM3YTUxMWY4ZTBmY2QmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.JZrvJIJEBC_II67RnxVqW3BUvPibdrwBYOQIRwa_UjI)

I want to achieve is that the fixed-wing flight automatically adjusts the flight altitude according to the terrain height measured by downwards lidar.

xdwgood, 2024-12-24:
@Jaeyoung-Lim  Thanks! I'll have a look.

closed 2025-06-03 by xdwgood