PX4/PX4-Autopilot #24176: add hobbywing dronecan compatibility

Mdea290, 2025-01-07:
### Describe problem solved by the proposed feature

I tried to use Hobbywing H6M, but they don't work with can communication. In ardupilot there is an option under dronecan to select Hobbywing esc which is missing here.
Is it possible to add this feature? it is really frustrating to have a dronecan esc that can provide all the telemetry and not be able to use it.

### Describe your preferred solution

hobbywing driver implementation?

### Describe possible alternatives

no

Mdea290, 2025-01-07:
We have already contacted them, but it seems they are not that interested in fixing the problem, which is a real shame because Ardupilot can support them.
I thought that some experienced people could port the Hobbywing esc code from Ardupilot to PX4 with almost no effort.

dagar, 2025-01-15:
If anyone with this hardware is interested in implementing support please let me know and I can walk you through what's needed.