PX4/PX4-Autopilot #24183: [Bug] drone overturned after takeoff

xdwgood, 2025-01-08:
### Describe the bug

This is our fourth flight, and compared to the previous three, we have added shock absorbers

### Flight Log

https://logs.px4.io/plot_app?log=35c1e0f2-d9ac-4ea5-8cf6-72feb0fd0d09

### Vehicle type

None

xdwgood, 2025-01-08:
[image]

I suspect that the slow acceleration process of the motor may have caused problems during the takeoff phase.

closed 2025-01-09 by xdwgood

bresch, 2025-01-09:
> we have added shock absorbers

That's probably the issue. Shock absorbers are physical low-pass filters so they add phase lag to the system and can make the rate loop unstable.