PX4/PX4-Autopilot #24187: RP2350 support

vidma, 2025-01-08:
### Describe problem solved by the proposed feature

Raspberry Pico 2 seem to be very good/performant, popular, simple (need very few external PCB components) and cheap CPU

### Describe your preferred solution

- upgrade NuttX (or backport - started backporting from unreleased master branch)
- current state:
  * NuttX backported on top of current `10.3.2+` PX4/NuttX branch
  * compiles
  * linking kinda works, get uf2
  * still need to debug why it does not boot

### Describe possible alternatives

- wait for NuttX release and do full NuttX upgrade/backporting?

### Additional context

continuation of https://github.com/PX4/PX4-Autopilot/pull/18083

vidma, 2025-01-17:
P.S. seems PX4 is booting up already, but still need to debug more what's happening there (no USB serial/mavlink yet):
will make a WIP PR once I getting slightly more functional.

---

[image]

https://blog.flyuav.lt/raspberry-pico-2-a-cheap-flight-controller/

akhodeir, 2025-01-17:
I am crossing my fingers till you finish 👍 
Good job

vidma, 2025-01-17:
btw, meanwhile, @akhodeir do you or someone else have some clue about these (preliminarily it seems it kinda works when SMP disabled):
- *SMP/IRQ:* `g_current_regs[CONFIG_SMP_NCPUS]` / CURRENT_REGS. what are these? and how is it supposed to be defined in case SMP is enabled vs disabled?
- *Linker scripts:* any LD script pecularities special to PX4? in memory_map.ld or also called script.ld/flash.ld . w/o SMP, seem to work I guess
- *gdb and stuff for PX4:* I still need to setup some better debugging env, then this damn USB serial / mavlink / nsh doesn't seem to be possible to connect to. gdb is doing its work, but it would be very nice to see the PX4 log messages somehow when USB serial do not work. any clues? e.g. maybe we have some existing helper which would redirect PX4/Nuttx log messages to `gdb` ?

----

for now, I probably, temporarily, hacked it like that (will need further fixes/cleanup I guess):

IRQ/SMP/CURRENT_REGS
----

```diff
diff --git a/arch/arm/src/rp23xx/rp23xx_irq.c b/arch/arm/src/rp23xx/rp23xx_irq.c
index a3baff0d6c..e8a66ede85 100644
--- a/arch/arm/src/rp23xx/rp23xx_irq.c
+++ b/arch/arm/src/rp23xx/rp23xx_irq.c
+#ifndef CONFIG_SMP
+// FIXME nasty hack: make it compile if CONFIG_SMP is disabled
+// FIXME: in case SMP disabled, what should be the value here? {0}?
+volatile uint32_t *g_current_regs[CONFIG_SMP_NCPUS] = {0};
+#endif
+
 #if defined(CONFIG_SMP) && CONFIG_ARCH_INTERRUPTSTACK > 7
 /* In the SMP configuration, we will need custom interrupt stacks.
  * These definitions provide the aligned stack allocations.
@@ -76,6 +81,12 @@ const uint32_t g_cpu_intstack_top[CONFIG_SMP_NCPUS] =
 };
 #endif /* defined(CONFIG_SMP) && CONFIG_ARCH_INTERRUPTSTACK > 7 */

+/* This is the address of the  exception vector table (determined by the
+ * linker script).
+ */
+
+extern uint32_t _vectors[];
+
```

```diff
diff --git a/arch/arm/include/irq.h b/arch/arm/include/irq.h
index 1fac0e1e55..2508a36b15 100644
--- a/arch/arm/include/irq.h
+++ b/arch/arm/include/irq.h
@@ -89,10 +89,16 @@ extern "C"
 /* For the case of architectures with multiple CPUs, then there must be one
  * such value for each processor that can receive an interrupt.
  */
-
+//#ifdef CONFIG_SMP
 EXTERN volatile uint32_t *g_current_regs[CONFIG_SMP_NCPUS];
 #define CURRENT_REGS (g_current_regs[up_cpu_index()])

+//#else
+// FIXME: when CONFIG_SMP not defined, I commented/disabled these lines out, as:
+// up_cpu_index() macro was conflicting with up_cpu_index() in nuttx/arch/arm/src/rp23xx/rp23xx_cpuindex.c
+//and maybe something g_current_regs defined /arch/arm/src/rp23xx/rp23xx_irq.c
+//volatile FAR uint32_t **g_current_regs;
+//#define CURRENT_REGS (g_current_regs[up_cpu_index()])
+//#endif
+
 /****************************************************************************
  * Public Function Prototypes
  ****************************************************************************/
@@ -116,7 +122,8 @@ EXTERN volatile uint32_t *g_current_regs[CONFIG_SMP_NCPUS];
 #ifdef CONFIG_SMP
 int up_cpu_index(void);
 #else
-#  define up_cpu_index() (0)
+int up_cpu_index(void);
+//  define up_cpu_index() (0)
 #endif
```

LD / Linker script:
----

```diff
+++ b/arch/arm/src/rp23xx/rp23xx_cpuindex.c
@@ -30,11 +30,9 @@
 #include "hardware/rp23xx_sio.h"

 #ifdef CONFIG_ARCH_HAVE_MULTICPU
...

int up_cpu_index(void)
{
  return getreg32(RP23XX_SIO_CPUID);
}

+// !!! NOTE/FIXE: this looks fishy because:
+//  - ifdef CONFIG_ARCH_HAVE_MULTICPU 
+//  - and closing  comment mentions CONFIG_SMP
#endif /* CONFIG_SMP */
```

memory_map.ld or also called script.ld/flash.ld:
```
...

MEMORY
{
    FLASH(rx) : ORIGIN = 0x10000000, LENGTH = 4096k
    RAM(rwx) : ORIGIN =  0x20000000, LENGTH = 512k
    SCRATCH_X(rwx) : ORIGIN = 0x20080000, LENGTH = 4k
    SCRATCH_Y(rwx) : ORIGIN = 0x20081000, LENGTH = 4k
}

+/* added EXTERN */
+EXTERN(_vectors)
ENTRY(_stext)
+/* where is _entry_point defined in ? what's the difference from  _stext? */
-/*ENTRY(_entry_point)*/
+/*EXTERN(bootStr)*/

SECTIONS
{
    .flash_begin : {
        __flash_binary_start = .;
    } > FLASH

    ...

    .text : {
        __logical_binary_start = .;
        _stext = ABSOLUTE(.);

        KEEP (*(.vectors))

        LONG(0xffffded3)
        LONG(0x10210142)
        LONG(0x000001ff)
        LONG(0x00000000)
        LONG(0xab123579)

        KEEP (*(.binary_info_header))

        ...

        . = ALIGN(4);
        /* init data */
+        /* vidma: added _sinit */
+        _sinit = ABSOLUTE(.);
        PROVIDE_HIDDEN (__init_array_start = .);
        KEEP(*(SORT(.init_array.*)))
        KEEP(*(.init_array))
        PROVIDE_HIDDEN (__init_array_end = .);
+        /* vidma: added _einit */
+        _einit = ABSOLUTE(.);

        ...

    } > FLASH

   ...
```

akhodeir, 2025-01-17:
those are really advanced questions. 

I advise you to ask here : https://forums.raspberrypi.com/viewforum.php?f=143&sid=291ab055d458508bb26aadeb341f19da
The forum is really helpful.

vidma, 2025-03-27:
still plenty of weird stuff to fix, but seems like I'm moved forward quite a bit just recently.

**Status:**

- [x] compiler flags / .LD file & linker issues: probably fixed ✅ 
  * had math functions (like uldivmod) linked to be in RAM, but code with zeros -> stuck or FAULT in clock_configure
- [x] Nuttx starts ✅ 
- [x] dmesg over gdb/emgdb: https://auterion.github.io/embedded-debug-tools/emdbg/debug/gdb.html#installation
- [x] currently using newest nuttx branch
  - [x] minor patches to make newest  nuttx code to work with PX4
  - [ ] if needed could port back to older px4-nuttx branch, but for now sticking to original nuttx, in case some new commits would need to be ported to RP235x.
- [ ] Serial/UART 🔴 
  - [ ] Mavlink over serial /dev/ttyS0
  - [ ] NSH over serial /dev/ttyS0 : not working yet ?
- [ ] USBACM serial: 🔴  ? not tested recently
- [ ] SPI: IMU + CS + interrupt - need a fix (config?)
- [ ] I2C not tested yet

---

[image]

[image]

akhodeir, 2025-03-28:
you are doing great job man !! 

regarding MAVlink over USB, did you consider using Pico SDK? it has "tusb.h" which can make your life easier.

vidma, 2025-03-28:
> regarding MAVlink over USB, did you consider using Pico SDK? it has "tusb.h" which can make your life easier.

yeah pico-sdk would be cool, but:
- aren't we kinda locked to Nuttx interfaces/drivers ?
- there's quite a bit of low level init code all around in nuttx/px4, so we'd need to make sure it doesn't conflict with what pico-sdk does under the hood

also (in my limited understanding of PX4) there's a zillion places which use serial & mavlink in a certain way, e.g.:
`mavlink start -d /dev/ttyACM0` and  `cdcacm_autostart start` in rcS init scripts, and `cdcacm_autostart` also contains the mavlink start command...

Btw, low level serial output seems to work (`rp23xx_lowsetup(); showprogress('a');`  in`rp23xx_start.c`), up to a certain point, where it is initialised with a higher level something, which applies more config and possibly uses interrupts of serial TX/RX, after that: RX still works but TX do not, no matter what config combinations I try so far (y/n for CONFIG_USE_EARLYSERIALINIT, CONFIG_SERIAL_TERMIOS, CONFIG_USE_SERIALDRIVER, ...):

[image]

---

regarding USB serial /CDACM:
- it seems init scripts are not called or stops at earlier point, so /dev/ttyACM0 is not created
- USB VBUS sensing is suspicious (thinks it's off), might be one of reasons
- with some config and manual commands in console I was able to obtain /dev/ttyACM0 visible inside PX4, but not on PC yet, something like this:
```
cdcacm_autostart status
cdcacm_autostart start
ls /dev/
serdev
ls /dev/
cdcacm_autostart status
mavlink start -d /dev/ttyACM0 # here mavlink crashes, not yet sure why
```

[image]

vidma, 2025-04-01:
still plenty of weird things/bugs all around,  however  now `mavlink start -d /dev/ttyS0` works already (partially)! 🥳 

[image]
[image]

vidma, 2025-04-03:
something inside `rp23xx_boardinitialize()` seemed to interfere with serial TX, still to debug exactly what it was, a config issue or code/bug. there's still plently of other things that don't work however.

akhodeir, 2025-04-06:
I can help you to build a custom board for testing your build. 

I already build something for Pi_4 : https://github.com/akhodeir/Pi-Fly

if you are interested for co-operation, please leave me your contact.

vidma, 2025-05-04:
after a few more fixes, noticed a weird/interesting glitch - peripherals start working ( *ONLY* ) after `reboot into bootloader` and then reboot back to PX4/Nuttx via `picotool reboot` (no need to reflash):
- [x] I2C IMU - start receiving events
- [x] CRSF RC-IN (UART1)
- [x] ttyACM0 USB serial. unstable, but kinda works!
```
./Tools/mavlink_shell.py /dev/cu.usbmodem01
➜  PX4-Autopilot git:(f/rp2350-wip-nuttx-master) ✗ ./Tools/mavlink_shell.py /dev/cu.usbmodem01
Connecting to MAVLINK...
nsh> Hello, World!!
```
- [x] hrt timer (not sure if only after such reboot sequence, but possibly)
- [ ] to retest GPS. possibly affected as well

I suspect a hrt timer here as USB things started improving after fixing the TIMER register addresses specific for rp2350 (vs 2040).

[image]