PX4/PX4-Autopilot #24188: [v1.16 Release Candidate] - Flight Testing & Flight Issues (log archive)

mrpollo, 2025-01-08:
### Describe the bug

## :exclamation: About
💡 For the upcoming v1.16 release, we are asking the Community for flight tests. This issue will be used for keeping track of all the tests from the Community.

If you're unsure about upgrading, please read the preliminary release notes on the PX4 docs. Pay close attention to the Upgrade Guide before deciding if v1.16 is right for you.

- [v1.16 Release Notes](https://docs.px4.io/main/en/releases/1.16.html)

## Pending Issues
- 

## ❓ Why we need flight testing
🌏 Thousands of people around the world depend on the stable release of PX4.
🖐️ Before every major release, as much testing as possible is needed to ensure that our release actually works!
😯 Simply said, **more testing = better release**
🤗 We would like to encourage every community member to join us in improving PX4!

## 🤷 How you can help with flight testing
😱 Even if you have never contributed a flight testing report, fear not! Please follow the steps below for the details of how you can contribute to flight testing.

1. **Select** a category from the "flight testing matrix" below
1. **Flash** the firmware (documented below)
1. **Test**
1. **Upload** the log (documented below)
1. **Report** the test in this issue thread!

### :bookmark_tabs:  Flight testing matrix
📖 **Here are the table of things we need tests for.**
This will constantly get updated, and your contribution to testing the missing slots (tests needed) will be greatly appreciated!

| Title | Description | Vehicle/Hardware requirements | Testing procedure | Test logs | Related issue/posts |
|---|---|---|---|---|---|
| General | By testing your usual setup, we want to capture any regression / weird behaviors you may notice | Any | Test what you usually use for your personal usage, and report it to the comment below |  |  |

### How to flash v1.16 beta firmware
[image]

1. Open QGC
3. Click the Q icon on upper right > Vehicle Setup > Firmware (Step 1 in picture)
4. Connect your flight controller to the computer
5. When the Firmware setup window pops up, click "Advanced settings" checkbox
6. Select "Beat Testing (beta)" from the drop down (Step 2 in picture)
7. Click "Ok"

This should automatically flash & reboot your flight controller with the latest v1.16 beta release!

### How to share your log
First, retreive the log from the flight controller in QGC Q icon > Analyze Tools > Log Download, and find the log you want to download.

[image]

Then:
1. Go to https://logs.px4.io/
2. Click on "Flight Report" tab (this allows public viewing of the log in the [browse page](https://logs.px4.io/browse))
3. Add the ULog file
5. Fill in the description, feedback, and video URL (if available)
6. Click "Upload"
7. Copy and paste the link of the log, and share it to this thread

When posting on this thread, please mention what kind of test it was (especially if it is one of the testing matrix options) & general info behind it!

## Questions?
Feel free to ask any questions regarding flight testing in the #dev-team channel in the [discord server](https://discord.gg/dronecode)!

### Vehicle type

None

xdwgood, 2025-01-08:
https://github.com/PX4/PX4-Autopilot/issues/24183

afwilkin, 2025-01-16:
No control in stabilized mode. the drone was not reacting to my movement commands. The drone started to slide back, I would pitch forward, and the drone would not react in accordance. Tried to then pull back in case the controls were flip flopped somehow, nothing happened. Crash

https://review.px4.io/plot_app?log=fb2ea683-bfaa-449b-991f-58eae53c97dd
https://review.px4.io/plot_app?log=35493241-0b0a-42c1-9a49-3bbc1670e2ca

afwilkin, 2025-01-24:
Wanted to post this here as well. Pixhawk4 with arkflow running 1.16 

Video: https://www.loom.com/share/141ee5ee32f64e519b5aaf798f495941?sid=f34c75a7-4a02-4fea-be0c-6a24d3742b98
Logs:
https://review.px4.io/plot_app?log=4f5c42ac-5696-44e7-8a57-03a033661189
https://review.px4.io/plot_app?log=bb955f4b-f2d9-49d1-b96a-0ba4e06be0a6

I fly in position mode with joysticks instead of RC that i did in my initial testing

bresch, 2025-01-24:
@afwilkin Thanks for the tests. 

Could you please also add a quick overview of the setup (looking at the log it's a multirotor with optical flow and range finder, no GNSS) and what's the outcome of the flight test from your perspective (e.g.: all good, height oscillation, position drift lower/higher than 1.15).

afwilkin, 2025-01-29:
Apologies for the delay, I thought this was in the other PR we were working on. I just updated my comments above

afwilkin, 2025-01-29:
> From [@PX4/testflights](https://github.com/orgs/PX4/teams/testflights)
> 
> No control in stabilized mode. the drone was not reacting to my movement commands. The drone started to slide back, I would pitch forward, and the drone would not react in accordance. Tried to then pull back in case the controls were flip flopped somehow, nothing happened. Crash
> 
> https://review.px4.io/plot_app?log=fb2ea683-bfaa-449b-991f-58eae53c97dd https://review.px4.io/plot_app?log=35493241-0b0a-42c1-9a49-3bbc1670e2ca
> 
> Drone that was being used (quadcopter, vtol, etc...): quadcopter x500 Flight controller (holybro pixhawk 6c, cube orange, etc): pixhawk 4 Radio (SBUS, sik telemetry, etc): sbus Controller (joysticks, RC): RC Position (GPS, VIO, Flow): GPS ESC (dshot, pwm, etc): PWM

Re - Stabilization Issue, 

What caused the crash in my earlier post MIGHT have been more of a heading issue than a broken stabilization flight mode or RC issue. I 
1. reflashed 1.15.2 on the vehicle
2. flew in stabilized over RC and did not see any issues
3. Flashed 1.16.0 alpha
4. **Recalibrated the RC** 
5. Reran the same tests
6. Did not see an issue, BUT I did see points where the vehicle would randomly yaw without my input

This makes me think there might be something about the mag that is feeding in to the heading that is causing some inconsistencies between 1.15.2 and 1.16.0

afwilkin, 2025-01-29:
Following the tests outlined [here](https://docs.px4.io/main/en/test_cards/mc_01_manual_modes.html#arm-and-take-off)

[Arm and Take-off](https://docs.px4.io/main/en/test_cards/mc_01_manual_modes.html#arm-and-take-off)
Video: https://www.loom.com/share/6ab405b7b71d426d8a55fa5b50d906de
Log: https://review.px4.io/plot_app?log=1809ab53-0576-4952-83f9-96762e040c66

[Flight](https://docs.px4.io/main/en/test_cards/mc_01_manual_modes.html#flight)

### Stabilized: 
Video: https://www.loom.com/share/6ab405b7b71d426d8a55fa5b50d906de
Log: https://review.px4.io/plot_app?log=1809ab53-0576-4952-83f9-96762e040c66

### Altitude:
Video: https://www.loom.com/share/35cc4a5185074605b092a9548852556a?sid=3b0b36de-7eb3-4010-9cb2-84272b707644
Log: https://review.px4.io/plot_app?log=5161578f-2dd4-45b5-aacc-bfc18d68b1fc

### Position
Video: https://www.loom.com/share/ebb87c86ba1c4ca59a090509b13493af?sid=f41d0e77-c455-4ce3-a6a9-68b8f61735ec
Log: https://review.px4.io/plot_app?log=1ded9991-d23a-4409-9c23-e5fa144621e7

[Landing](https://docs.px4.io/main/en/test_cards/mc_01_manual_modes.html#landing)
Video: https://www.loom.com/share/2fe0d6627fc742d189a0a98e27732082?sid=c54ebc9f-447e-4abf-875a-355a7c609141
Logs:
https://review.px4.io/plot_app?log=335c5220-8ed4-478d-9153-0e7efbeb9836
https://review.px4.io/plot_app?log=6317f09e-1f3d-4110-8760-360ac2f590a1

afwilkin, 2025-01-29:
[Create and upload mission](https://docs.px4.io/main/en/test_cards/mc_02_full_autonomous.html#create-and-upload-mission)
Video: https://www.loom.com/share/5b6b365647914e6fb5c8e4d7861ef276?sid=7dcd119e-902b-4ec1-a000-06fa4de410f3
Log: https://review.px4.io/plot_app?log=2babfc02-e180-4651-944d-4b83b57f72ab

[Arm and Take-off](https://docs.px4.io/main/en/test_cards/mc_02_full_autonomous.html#arm-and-take-off)
Video: https://www.loom.com/share/5b6b365647914e6fb5c8e4d7861ef276?sid=e7e0b888-0570-4e45-b302-035b7dcef28d
Log: https://review.px4.io/plot_app?log=de383900-edbc-4cb9-b881-491820895eb1

MaEtUgR, 2025-01-30:
I had a look at your original log. RC input data looks fine. Translation to setpoints also looks high level ok, tracking is not good and I think this is because the yaw rate and hence also the yaw goes unstable:
[image]

Could it be that your vehicle's motor mounts on the arm are not rotated the right way or even rotated slightly the wrong way? That could explain the lack of yaw authority even if the propeller turn direction and motor assignment is correct. For reference the motors should be ever so slightly (2-5°) tilted like indicated on my drawing (assuming props turn in default "cleaning" direction otherwise everything is the opposite way):
[image]

MaEtUgR, 2025-01-30:
I found proof for my hypothesis with the yaw authority. Two motors are basically idling and the other two carry all the weight and still the yaw turns in the wrong direction. Please verify the airframe according to my scetch. I'm happy to explain if it's not clear.
[image]
Note that in the later logs you posted it's slightly better but the issue clearly still persists:
[image]

afwilkin, 2025-02-10:
> I found proof for my hypothesis with the yaw authority. Two motors are basically idling and the other two carry all the weight and still the yaw turns in the wrong direction. Please verify the [...]

Yes, thank you for finding that. When one of our UAV's crashed the arm got pushed in a bit, causing it to be shorter than the others and inducing this weird behavior. We fixed the issue and continued our flight testing. 

Appreciate the find!

afwilkin, 2025-02-10:
[### Failsafe Testing](https://docs.px4.io/main/en/test_cards/mc_04_failsafe_testing.html)

Description of vehicle:
Holybro s500
Flight controller: Pixhawk 4
Radio: Sbus + Sik telemetry
Controller: RC
Position: GPS
ESC: PWM

### [Auto Manual Mix](https://docs.px4.io/main/en/test_cards/mc_03_auto_manual_mix.html)

**Mission**
Result: Pass
Log: https://review.px4.io/plot_app?log=07e64eb7-20e5-4e81-8e0c-c1c6f4fd889f
Comments: Did mission well

**Takeoff in position mode, flip to mission mode, RTL**
Result: Pass
Log: https://review.px4.io/plot_app?log=9f9d4ed9-6f4f-4dc1-b2d9-84311ecbd8c7 , https://review.px4.io/plot_app?log=67332627-897e-413a-88b6-7032d2009e8f
Comments: nothing really wrong. No issue

**RC Loss**
Result: Pass 
Log: https://review.px4.io/plot_app?log=4ecae679-853c-4899-82ff-b5575a17f0c7

**Datalink Loss**
Result: Pass
Log: https://review.px4.io/plot_app?log=7bdac691-a08a-4dfa-b24e-e9220eab0fbc , https://review.px4.io/plot_app?log=5c8ce834-65bc-410b-aa34-c035fd98f25f
Comments: It felt longer than 10 seconds from unplugging the radio, to the drone activating the failsafe. I then realized that the drone will wait 10 seconds to declare there is no connectivity, and then wait 5 more seconds to activate it.... which might be misleading. On the first log we plugged the SIK radio back in, but QGC did not reconnect to drone until we hit the "Disconnect" button in QGC, reinitializing the SIK radio and making a connection.  This is expected behavior as we have seen it in past versions of PX4. So, no issue. In the second log we hit the "Disconnect" button, reacquired the UAV, and took position control as expected. 

**Battery Failsafe**
Result: Pass
Logs: https://review.px4.io/plot_app?log=e689fe73-41c2-440a-86a5-d03d46289086 , https://review.px4.io/plot_app?log=ba880f55-03f4-4288-9e3f-8017a1f6944e
Comments: The first log we flew in position mode until the battery failsafe was hit, and the drone went into RTL mode. we then took over manual control, and waited until the emergency level was hit. The drone then went into land mode. In the second log the cold caused the battery discharge to fall a bit more...rapidly and then drone never had a change to activate RTL, and went straight into land. As it should.

DroneWuKong, 2025-05-05:
Dump of logs from testing out on the V6XRT

Description of vehicle:
Holybro X500 v2
Flight controller: PX4_FMU_V6XRT
Radio: Sik telemetry 915mhz
Controller: RC
Position: GPS
ESC: PWM
Battery: 4s 3300mah

Unit was updated via QGC v4.4.4

In the firmware update menu I chose "beta"

QGC states firmware is 1.16.0beta not "RC"

During Battery Failsafe testing, "issues" arise when setting failsafe levels and logic in handling voltage sag to activate critical failsafe.  In a low battery situation, the time before battery warning, battery critical and emergency landing can be triggered if batteries have higher than normal voltage sag. 

Flight Review Logs:

https://www.loom.com/share/922707bc7df947c788db3dc703641c16?sid=5c8f77ca-f2c3-4c78-9faa-08e1a1b24be0

 Log_10
 https://review.px4.io/plot_app?log=e3e1f958-d031-44d7-a407-a398d2c0280a
 Log_11
 https://review.px4.io/plot_app?log=82c53194-b4bd-4e58-a813-052cc24df78d
 Log_12
 https://review.px4.io/plot_app?log=39fd63ff-4756-4ffa-83c1-efb28acadb36
 Log_13
 https://review.px4.io/plot_app?log=178fcb07-1cde-4d1e-94e6-189bff8cdae8

https://www.loom.com/share/699e22195e6b49f4808a3dcee9928d0c?sid=70f92797-4cab-479b-afd0-eb5dcc1134d9

 Log_14
 https://review.px4.io/plot_app?log=613e0535-ef9a-4b6d-ba35-a44f7b8566b4

https://www.loom.com/share/41a255f4ebb54c2fa9f4b4ba3fa33e07?sid=4f94293d-00ab-4f59-88fb-426709c17db9

 Log_15
 https://review.px4.io/plot_app?log=fb03dbbc-a8ef-4217-8e8f-37929c5f5908

https://www.loom.com/share/824bc26e965e46a1a37cc0b5c014aeed?sid=da4a8413-b773-4233-a34e-8e9aa107f796

 Log_16
 https://review.px4.io/plot_app?log=4a2dea9c-f580-41db-acfd-dce392ad24cd
 Log_17
 https://review.px4.io/plot_app?log=772b0a55-b668-4bb8-8bf4-e185832b00b2

https://www.loom.com/share/0508de1f1a8c474480b9c74840eb5923?sid=8108834d-0d00-4f37-8545-c7d988e72d9c

 Log_18
 https://review.px4.io/plot_app?log=7eed0801-351b-42c5-9e0d-d03b41cd4dc0
 Log_19
 https://review.px4.io/plot_app?log=c4ca70c9-cb41-48fd-96aa-b17c16d2608e

https://www.loom.com/share/4a5a20029b504c05b2038b994607449e?sid=8d16d715-7cef-4555-92cd-9bf36d07ccdc

 Log_20	
 https://review.px4.io/plot_app?log=3b424515-553e-462a-8d37-5cd66b96cc05

afwilkin, 2025-05-09:
### About the Vehicle
Description of vehicle:

Holybro X500 v2
Flight controller: PX4_FMU_V6C
Radio: Sik telemetry 915mhz
Controller: RC
Position: GPS
ESC: dshot
Battery: 4s 3300mah
Setup

QGC: v4.4.4
PX4: v1.16.0-rc1 *
Notes regarding firmware: In the firmware update menu I chose "beta", QGC states firmware is 1.16.0beta not "RC"

### Testing Results

**Flight Review Logs:**

**Flight 1**

- https://review.px4.io/plot_app?log=2623fc00-a4db-46c5-a6c7-b2f444481ad9

**Flight 2**

- https://review.px4.io/plot_app?log=b7288560-05c7-4c14-a8cb-4cf628eb713b

dirksavage88, 2025-05-14:
We definitely should have holybro 6x and arkv6x logs added to the test matrix. @afwilkin we could cover the holybro 6x and Ark does their own internal testing.

afwilkin, 2025-05-14:
Agreed

TeddyZaremba, 2025-05-14:
Starling 2 Max
Flight controller: VOXL2
Radio: Doodle Radio
Controller: Joystick
Position: GPS
ESC: VOXL ESC Mini 4-in-1
Battery: 2s 3000mah

Setup
QGC v4.4.4
PX4 v1.16.0 (RC)

**Manual Modes**

Arm Takeoff in Position 
https://www.loom.com/share/1db805470a6a45728db6f3755ee81711?sid=4ebb863e-0e57-4b68-b332-07788233cae7
https://review.px4.io/plot_app?log=9d49d359-fba2-4663-b90a-449d80b18703

Altitude
https://www.loom.com/share/973410ad28a14cec8850d229bf850c68?sid=b41712d1-023f-4ef2-974b-1ed5fbba7bf8
https://review.px4.io/plot_app?log=fb255bef-0949-454c-8ade-7d24c50b1e30

Position Flight Mode / Precision Land
https://www.loom.com/share/ea83c73f2f514ffd9b7796aff0b2edad?sid=46096409-79c4-4ad6-916e-57b2d1fd02e0
https://review.px4.io/plot_app?log=162e8f7d-2019-47b0-aa60-7c6c221f575a

Notes: It shot straight up when I switched to precision land

**Autonomous**

Upload mission and mission mode
https://www.loom.com/share/68544cea286a4ca39210f766e46b9a40?sid=ccd80231-edab-4130-822d-8f2d75459481
https://review.px4.io/plot_app?log=d1323d81-79f3-4ed9-b678-0a754b9f728a

**Failsafe's**

Lost joystick Failsafe
https://www.loom.com/share/944f2256a15e43b99972c175f2ae63f5?sid=3dcd977b-1e49-4fdb-a2fc-178e3e3f5ffd
https://review.px4.io/plot_app?log=559300e0-c9de-4377-a683-a49b3f51d1e1

Takeoff in Position / Flip to RTL
https://www.loom.com/share/a7011c05b4e04e0b868bc366c56716f9?sid=38e1747c-6f8d-4f03-a109-947471b81252
https://review.px4.io/plot_app?log=ce64e93f-295e-4acf-908f-efc764d65a67

Battery Failsafe
https://www.loom.com/share/82b6b571c9314fc0878238b5099b0de3?sid=5493614e-ef05-4f84-8f44-c5dd084d7749
https://review.px4.io/plot_app?log=31142be3-ad19-4350-aad7-9aa45ac37907

Datalink Loss
https://www.loom.com/share/d8e6a7ffa0754ada8f81d757d894c12e?sid=054a203c-6f88-4bb0-83be-f22a919e298b
https://review.px4.io/plot_app?log=f05e4310-9216-4f76-b7c9-e58ba7fc15c0

mrpollo, 2025-05-14:
Thanks @TeddyZaremba

dirksavage88, 2025-05-21:
Holybro X500
Flight controller: Holybro 6X
Radio: Holybro sik telemetry
Controller: Radiomaster Pocket (ELRS)
Position: Ark GPS
ESC: Holybro
Battery: 4s 6AH

Setup
_QGC v5.0_
_PX4 v1.16 (RC1) w/ rc_common enabled_

**Manual Modes**

Manual Flight Mode
https://www.loom.com/share/d8338b21bd5c42f88ef8ed5b86afb2ab?sid=0a594bd1-96a2-4226-afc4-d7ffa5f51d0d
https://logs.px4.io/plot_app?log=644e1285-88d2-4274-b995-fd530b7c1a2b

Altitude Flight Mode
https://www.loom.com/share/a5af5bc7956e4f51b638672af927b1f8?sid=17a836a5-39ce-482a-a826-51303ca6c876
https://logs.px4.io/plot_app?log=8f74689c-a091-4494-9e1b-db47ba187143

Position Flight Mode 
https://www.loom.com/share/c39e8c13c72248168fc36aca43ac4d5a?sid=347c6494-4d03-4214-ad5e-3ea18ed2c48f
https://logs.px4.io/plot_app?log=f2aea0fd-66ed-4506-8420-e56cc073a800

**Autonomous**

takeoff/land
https://www.loom.com/share/a542c996ab294c73bd895dcacaf4f706?sid=d42218c6-656b-4ed9-b56e-df4bf330352e
https://logs.px4.io/plot_app?log=5a1959ea-efef-4749-85a2-07f79905ec94

mission mode
https://www.loom.com/share/11c70095b1f944af937ae123b9349c20?sid=068a4cee-6723-4044-92d4-bc5397a01987
https://logs.px4.io/plot_app?log=4a07f9e5-9d80-498f-b6b6-738e850a5c83

**Failsafe**

Datalink Loss
https://www.loom.com/share/fa127c7171ec434d92b9134a62c984d9?sid=601b87c1-4316-4d4c-aae4-6d1cce1bcbb1
https://logs.px4.io/plot_app?log=3dccbe33-6d77-462a-8c3b-a0de45e78252

RC Loss
https://www.loom.com/share/abcc4123804e4b95b543a5637e67517f?sid=10e855eb-cb13-4b32-8a57-d9c7d1267033
https://logs.px4.io/plot_app?log=9255d9b3-0e96-4afd-9e87-ddf9509acc0c

Battery Failsafe
https://www.loom.com/share/27972729ebec44b7ae901cab02e3ea41?sid=40eeddee-f6ce-4014-bb02-e2e7998673ee
https://logs.px4.io/plot_app?log=3bdc98b4-2f53-4237-8f51-e71a9f870913

Notes: Unplugging the holybro telem module resulted in QGC crashing (@DonLakeFlyer)

DroneWuKong, 2025-05-22:
Flight Testing of the Cube Orange Plus

Tests
Manual mode - ok
Position mode - nope
Mission mode - nope

Observations
Was having issues with the GPS not being seen, I replaced with an old BN220, but still did not see GPS. I was able to test Attitude and Manual flight. But under the default settings the throttle is a bit abrupt after crossing mid stick. Not having some fine control at that point or at least some longer ramp up would have helped me not bobble the takeoff.

I also did notice that there was no defaulted battery cell count, and when i did choose 4 cell, I had to re calc the voltage divider before it was right and let me arm. 

Will update with more information on the GPS used. 

Setup:
Holybro X500 v2
Flight controller: Cube Orange Plus
Radio: Sik telemetry 915mhz, IRC Ghost
Controller: Radiomaster Tx/Ghost Module
Position: N/A
ESC: dshot600
Battery: 4s 3300mah
Setup

QGC: v4.4.4
PX4: v1.16.0-rc1 *

FETTec 45A 4in1 ESC on FMU PWM 1-4 (DSHOT600)
SIk Telem on Telem1
Ghost Rxon SBus
GPS on GPS 2

Logs

https://review.px4.io/plot_app?log=16c369fb-8907-4b33-af2c-82787a2eb407

Video

Will follow once I have better bandwidth connection

mrpollo, 2025-05-22:
@DroneWuKong is this the first time you fly this vehicle in its current configuration? Was GPS working before?

mrpollo, 2025-05-22:
Reported by @julianoes, no logs shared

> Just following up on v1.16 testing:
>
> I've set up:
> - a 650 quad frame (this [Hexsoon one](https://irlock.com/products/hexsoon-edu-650-teardrop-> energy-saving-aerodynamic-rotorcraft-v2))
> - with Cube Orange+
> - Here3 GPS/mag (via DroneCAN)
> - Herelink (using Telem1 and SBUS), using [QGC v4.4.3 sideloaded](https://github.com/CubePilot/qgroundcontrol-herelink/releases/tag/v4.4.3-herelink)
> - flashed PX4 from the release/1.16 branch ([1ed406c4](https://github.com/PX4/PX4-Autopilot/commit/1ed406c41d1db21376b7191775e60961f50331ef))
> 
> This flew fine, no issues for me.
> 
> I just had to tune down attitude roll/pitch slightly from the default 4001 frame:
> - MC_ROLL_P: 6.5 -> 5
> - MC_PITCH_P: 6.5 -> 5
>

dk7xe, 2025-05-31:
Tested PX4 1.16 RC0 on a Sonicmodel Skyhunter with NXP MR-VMU-RT1176 (Pixhawk 6X-RT equivalent)
Three flights with over 32min flighttime overall.

Stabilized, mission mode, Loiter, RTL, Geofence has been tested.

GPD05 - Skyhunter w NXP VMU-RT1176 PX4 1.16 RC0 - F1
https://logs.px4.io/plot_app?log=4fe84c0d-f473-4a66-b501-5f8aec7e1714
Flight time 0:15:45
Airspeed sensor failed after takeoff
Power module failed after 2min
MAG1 timeout after 3min

GPD05 - Skyhunter w NXP VMU-RT1176 PX4 1.16 RC0 - F2
https://logs.px4.io/plot_app?log=da692222-e3ec-4453-86b1-11105a24a757
Flight time 0:10:33
Airspeed sensor failed 20sec after takeoff
MAG1 timeout after 3min
Power module failed after 3min

GPD05 - Skyhunter w NXP VMU-RT1176 PX4 1.16 RC0 - F3
https://logs.px4.io/plot_app?log=ced32f14-cd64-4f2f-a54a-91ee1845448c
Flight time 0:07:24
Airspeed sensor failed 150sec after takeoff

mrpollo, 2025-05-31:
@sfuhrer can you please take a look at the errors reported by @dk7xe on fixed wing? There's also another issue with dshot on VTOL that we should take a look at, I'll tag you on that one.

sfuhrer, 2025-06-02:
> airspeed failure
There is a very large error (scale error) on the airspeed data, likely due to placement of the pitot in proximity to the fuselage/wing. PX4 is learning the correct scale factor over time (see attached screenshot), but it takes a while and at the end of the flight, the value is only stored if the airspeed sensor is not declread invalid (which was the case here).
--> set ASPD_SCALE_1 to 1.7 for this plane.

[image]

[image]

> mag1 timeout
The mag1 data times out indeed, then comes back for a little while before timing out again. In my experience that usually points to a wiring issue.
[image]

> power module timeout
The power module times out at the same time as the mag1, so likely related (wiring? power?)
[image]

Is there anything else you'd want me to have a look in the log file?

dk7xe, 2025-06-04:
But as a consequence i will do shorter flights next time to have the airspeen scaling saved.
As well i have mounted a 2nd 5V Regulator to supply the FMU from both power ports.

mrpollo, 2025-07-16:
Thanks everyone that helped with testing!

closed 2025-07-16 by mrpollo

afwilkin, 2025-07-21:
RC3 Logs w/ Ark ESC

1: https://review.px4.io/plot_app?log=fdf4ac53-c738-44b4-94da-0ae03ea44971
2: https://review.px4.io/plot_app?log=4037346f-fa5c-4b5b-9f16-3b34c5d35195
3: http://review.px4.io/plot_app?log=c74fd354-b694-4424-9ce6-75292c68a6c6