PX4/PX4-Autopilot #24192: [Bug] "brew install px4-sim-gazebo" generated 2 errors

ytl001, 2025-01-10:
### Describe the bug

Errors generated (see screenshot below) in "brew install px4-sim-gazebo" while following the steps in installing a new gazebo classic simulation on https://docs.px4.io/main/en/dev_setup/dev_env_mac.html.  This is the first time px4 has been installed on this computer.

MacOS Ventura 13.7.2
intel quadcore i5
16gb RAM

Command Line Tools were updated using "sudo xcode-select --install", even though it still complained that a newer version is available.

### Screenshot / Media

Yings-iMac:~ ytl$ brew install px4-sim-gazebo
==> Auto-updating Homebrew...
Adjust how often this is run with HOMEBREW_AUTO_UPDATE_SECS or disable with
HOMEBREW_NO_AUTO_UPDATE. Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
==> Auto-updated Homebrew!
Updated 3 taps (osrf/simulation, homebrew/core and homebrew/cask).
==> New Formulae
codesnap                                                           netatalk
==> New Casks
font-nanum-square         font-nanum-square-neo     font-nanum-square-round   macai                     serial-studio

You have 33 outdated formulae installed.

==> Fetching dependencies for px4/px4/px4-sim-gazebo: libpng, nettle, glib-networking, doxygen and osrf/simulation/gazebo11
==> Fetching libpng
==> Downloading https://ghcr.io/v2/homebrew/core/libpng/manifests/1.6.45
############################################################################################################################# 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/libpng/blobs/sha256:0a59a94ee138a9903a2cf0f7a06b448d267647f245dec672ae56c92d27fd695
############################################################################################################################# 100.0%
==> Fetching nettle
==> Downloading https://ghcr.io/v2/homebrew/core/nettle/manifests/3.10.1
############################################################################################################################# 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/nettle/blobs/sha256:7d924af8b68729fbd75a279173b03da71219ee53c8c660198360dd4c8c0cde5
############################################################################################################################# 100.0%
==> Fetching glib-networking
==> Downloading https://ghcr.io/v2/homebrew/core/glib-networking/manifests/2.80.1
############################################################################################################################# 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/glib-networking/blobs/sha256:051b59d9c1a7d2403a8d34628d6c0acad7c25f50e9d25d4756095c
############################################################################################################################# 100.0%
==> Fetching doxygen
==> Downloading https://ghcr.io/v2/homebrew/core/doxygen/manifests/1.13.2
############################################################################################################################# 100.0%
==> Downloading https://ghcr.io/v2/homebrew/core/doxygen/blobs/sha256:4dadc96efe27d3d615ae06a1196c444dd260201122a1b11612147615bd2fd0
############################################################################################################################# 100.0%
==> Fetching osrf/simulation/gazebo11
==> Downloading https://gist.githubusercontent.com/scpeters/9199370/raw/afe595587e38737c537124a3652db99de026c272/brew_python_fix.pat
Already downloaded: /Users/ytl/Library/Caches/Homebrew/downloads/12171d5787bf5709495604c1591fd598a9cffc3a055eb509bea588d33019cabc--brew_python_fix.patch
==> Downloading https://osrf-distributions.s3.amazonaws.com/gazebo/releases/gazebo-11.15.0.tar.bz2
Already downloaded: /Users/ytl/Library/Caches/Homebrew/downloads/e21841426e32b22934a46365b5def3c1390d536ea4c9fdfb4b1c008b593c3251--gazebo-11.15.0.tar.bz2
==> Fetching px4/px4/px4-sim-gazebo
==> Downloading https://raw.githubusercontent.com/PX4/PX4-Autopilot/master/Tools/px4.py
Already downloaded: /Users/ytl/Library/Caches/Homebrew/downloads/be4aff9b473b8b0c6bae0ff08a6611885de9d56ee3a477c05453bb6bbcaab246--px4.py
==> Installing px4-sim-gazebo from px4/px4
Warning: A newer Command Line Tools release is available.
Update them from Software Update in System Settings.

If that doesn't show you any updates, run:
  sudo rm -rf /Library/Developer/CommandLineTools
  sudo xcode-select --install

Alternatively, manually download them from:
  https://developer.apple.com/download/all/.
You should download the Command Line Tools for Xcode 15.2.

==> Installing dependencies for px4/px4/px4-sim-gazebo: libpng, nettle, glib-networking, doxygen and osrf/simulation/gazebo11
==> Installing px4/px4/px4-sim-gazebo dependency: libpng
==> Downloading https://ghcr.io/v2/homebrew/core/libpng/manifests/1.6.45
Already downloaded: /Users/ytl/Library/Caches/Homebrew/downloads/45e044f04a9c9003b3fbe1ee536c125e6d68a8509f2d30cf7308fb6421d3b980--libpng-1.6.45.bottle_manifest.json
[... 48 lines cut ...]
make[1]: *** [plugins/rest_web/CMakeFiles/RestUiPlugin.dir/all] Error 2
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/osrf/homebrew-simulation/issues

These open issues may also help:
gazebo11 fails to build on macos15.2 m1pro https://github.com/osrf/homebrew-simulation/issues/2927

Error: A newer Command Line Tools release is available.
Update them from Software Update in System Settings.

If that doesn't show you any updates, run:
  sudo rm -rf /Library/Developer/CommandLineTools
  sudo xcode-select --install

Alternatively, manually download them from:
  https://developer.apple.com/download/all/.
You should download the Command Line Tools for Xcode 15.2.

Yings-iMac:~ ytl$ 

### Vehicle type

None

mrpollo, 2025-01-20:
Have you tried installing the Command Line Tools for Xcode?

ytl001, 2025-01-21:
Yes, I followed the instructions to first remove the directory then install
xcode command line tools.  It didn't give any errors during installation.

ytl001, 2025-03-07:
> ~Have you tried installing the Command Line Tools for Xcode?~ (sorry didn't ready)
> 
> Did you upgrade recently from from an intel to an m-based mac? this sometimes leads to issues with previous dependencies

No, I have been trying to install on the same intel based mac all along.