PX4/PX4-Autopilot #24200: [Bug] PX4 msgs are no longer being synchronized

beniaminopozzan, 2025-01-10:
### Describe the bug

Although there is a Jenkin action for synchronizing the PX4 msgs and services on https://github.com/PX4/px4_msgs :
https://github.com/PX4/PX4-Autopilot/blob/208d37e703f1bdb52f0b134fad685879a6e1d5cf/Jenkinsfile#L223-L251
The messages have not being synchronized since the beginning of December.

Previously the messages where synchronized by the github actions but that was removed in https://github.com/PX4/PX4-Autopilot/commit/652bb82603c5f54b611b8a705123bc1e526b07d1

### Expected behavior

Messages should be synchronized on every push on `main`

### Vehicle type

None

TheMightiestCarrot, 2025-01-22:
Distilled quickfix commands :
```
rm -f src/px4_msgs/msg/*.msg               2>/dev/null || true && \
    rm -f src/px4_msgs/msg/versioned/*.msg     2>/dev/null || true && \
    rm -f src/px4_msgs/srv/*.srv               2>/dev/null || true && \
    rm -f src/px4_msgs/srv/versioned/*.srv     2>/dev/null || true && \
    mkdir -p src/px4_msgs/msg/versioned src/px4_msgs/srv/versioned && \
    cp /home/$USERNAME/PX4-Autopilot/msg/*.msg                     src/px4_msgs/msg/         2>/dev/null || true && \
    cp /home/$USERNAME/PX4-Autopilot/msg/versioned/*.msg           src/px4_msgs/msg/versioned 2>/dev/null || true && \
    cp /home/$USERNAME/PX4-Autopilot/srv/*.srv                     src/px4_msgs/srv/         2>/dev/null || true && \
    cp /home/$USERNAME/PX4-Autopilot/srv/versioned/*.srv           src/px4_msgs/srv/versioned 2>/dev/null || true
```

bkueng, 2025-02-21:
Fixed in https://github.com/PX4/PX4-Autopilot/pull/24364

closed 2025-02-21 by bkueng