PX4/PX4-Autopilot #24206: [Bug] Multi EKF vehicle local positions starting with large offset on quad

patrikpordi, 2025-01-13:
### Describe the bug

I’ve been facing significant issues with quads using optical flow for localization indoors(no GPS and EKF2_HGT_REF=RANGE). The estimators are starting from a large offset from each other(20-500 m), if the estimator switches the drones tries to compensate. It happens occasionally.

### To Reproduce

Fly indoors with no GPS and EKF2_HGT_REF=RANGE in the latest main in position/offboard.

### Expected behavior

If the estimator switches the drone should should stay where it is since the estimates should align.

### Screenshot / Media

[image]
[image]

### Flight Log

https://review.px4.io/plot_app?log=6c072e4a-e0aa-4a46-9833-78db51355d37
https://review.px4.io/plot_app?log=6e490f75-ed41-477a-9d91-771896215a7d

### Software Version

Main

### Flight controller

ARKV6X Flight Controller

### Vehicle type

Multicopter

### Additional context

I am also having issues with dist_bottom_valid behaving weirdly and invoking fail safe even though the drone seemed fine otherwise.

patrikpordi, 2025-01-13:
This is an offboard example, the drone is trying to takeoff, then the estimator switches and the drone jumps to the side
https://github.com/user-attachments/assets/d08212b3-de05-443a-b41e-0f8d2d4f6da9

patrikpordi, 2025-01-14:
I have investigated it more I think something is probably wrong with the optical flow math. I walked around with two different drones in the office.
[image]
[image]
Logs:
https://review.px4.io/plot_app?log=3c96872f-acd1-4f5f-bb4e-ae06cf65c081
https://review.px4.io/plot_app?log=2777fc99-72a0-4338-886a-ff27403c66d5

I also recorded a log with the GPS on, the same behavior is visible while the GPS is setting up, but it eventually gets resolved once the GPS got the signals.
 
[image]
Log:
https://review.px4.io/plot_app?log=3fc313f0-193a-405f-8798-7eadbbb54054

patrikpordi, 2025-01-15:
I have recorded three logs based on @dagar suggestions.
I rebooted the flight controller and did not move the drone before flying to get clean logs
Setups:
EKF2_HGT_REF=RANGE and EKF2_RNG_CTRL=1
https://review.px4.io/plot_app?log=7f4b6b0c-074d-4512-b66e-8edef373ce94
EKF2_HGT_REF=RANGE and EKF2_RNG_CTRL=2
https://review.px4.io/plot_app?log=e9694d80-3fc4-40c7-a3c7-ea718ac1f4a6
EKF2_HGT_REF=GPS and EKF2_RNG_CTRL=1
https://review.px4.io/plot_app?log=89041761-4655-4135-a31f-811554fd1ae2

beniaminopozzan, 2025-03-05:
@patrikpordi any clue regarding those log dropouts? There have been a few in all logs you uploaded

patrikpordi, 2025-03-05:
Not so sure about the reason, but using this commit they were reduced:
https://github.com/PX4/PX4-Autopilot/commit/0b259face5cfc522121a4a339e6d91dd18f6ca4f

Also this issue moved a bit forward since it was updated here:
Reason behind the problem based on input from @dagar :
For optical flow to start we need a valid terrain estimate or another valid horizontal aid source https://github.com/PX4/PX4-Autopilot/blob/96461cdd7d9f576e17f61f093dd110ec17a23610/src/modules/ekf2/EKF/aid_sources/optical_flow/optical_flow_control.cpp#L156

In range height mode the terrain isn't valid.

Solutions could be:
Making terrain valid while in range height mode.
OR
Forcing people to use baro as the main height ref if gps is not available and making sure that range aid is used, especially when the drone does not move horizontally. 

@dakejahl is working on the second one, first in the sim and if it can be reliable tested there it should resolve the problem on real vehicles.

I am not so sure how the estimator switching should be handled still and also whether we should make the terrain valid in range height for now. Probably @dagar or @bresch has some thoughts on this.

beniaminopozzan, 2025-03-05:
Thanks for the update @patrikpordi !

> In range height mode the terrain isn't valid.

Is that a design decision? It seems counter intuitive to me.

> Forcing people to use baro as the main height ref if gps is not available and making sure that range aid is used, especially when the drone does not move horizontally.

Wouldn't this cause other issues like: https://github.com/PX4/PX4-Autopilot/issues/24058? There the solution would have been switching to range height mode for example.

closed 2026-02-01 by github-actions[bot]