PX4/PX4-Autopilot #24230: [Bug] On Generic Standard Plane DShot mixing get called twice

PetervdPerk-NXP, 2025-01-20:
### Describe the bug

When configuring the autopilot to be in Generic Standard Plane the actuator outputs seemed to be glitchy. When putting the scope on the signal I noticed this.

[image]

As you can see shortly after `DShot::Run()` is called it get called a second time and then it operates @ 400Hz.

When using multicopter (MC) I don't see this behaviour

### To Reproduce

Reset FMU, select "Generic Standard Plane" airframe. 
In actuators select dshot, and set MAIN1 to Motor 1.
Then connect scope on the Pin and you see the double output

### Expected behavior

A periodic pulse of the Dshot output without bursts

### Software Version

Main

### Flight controller

V6X-RT

### Vehicle type

Fixed Wing

dagar, 2025-01-22:
If you look at `work_queue status` how fast is the dshot module cycling?

PetervdPerk-NXP, 2025-01-23:
Quadrotor multicopter, no glitching, single Dshot output motor 1
```
Work Queue: 13 threads                          RATE        INTERVAL
|__ 1) wq:rate_ctrl
|   |__ 1) control_allocator                789.8 Hz         1266 us
|   |__ 2) dshot                            789.8 Hz         1266 us
|   |__ 3) mc_rate_control                  789.8 Hz         1266 us
|   \__ 4) vehicle_angular_velocity         789.9 Hz         1266 us
```

Generic standard plane, glitching, single dshot motor 1

[image]

```
Work Queue: 13 threads                          RATE        INTERVAL
|__ 1) wq:rate_ctrl
|   |__ 1) control_allocator                789.7 Hz         1266 us
|   |__ 2) dshot                            789.7 Hz         1266 us
|   \__ 3) vehicle_angular_velocity         394.8 Hz         2533 us

```
Hm most likely I think it has to do with the different rates of `dshot`, `control_allocator` and `vehicle_angular_velocity`

dk7xe, 2025-03-03:
@dagar @mrpollo this is one of the examples i have mentioned in the board meeting last week

mrpollo, 2025-05-28:
We need to investigate if this was fixed with #24727 in v1.16.0

MaEtUgR, 2025-05-28:
@dk7xe It would be awesome if you could give 1.16 beta a try which contains the fix https://github.com/PX4/PX4-Autopilot/pull/24727 and when we have it confirmed we could close this issue with a notice.

PetervdPerk-NXP, 2025-05-28:
@MaEtUgR I still see it on main with 029e0a7740d33bcdd92cdaa9b63f7748f5046871

```
nsh> ver all
HW arch: PX4_FMU_V6XRT
HW type: V6XRT
HW FMUM ID: 0x000
HW BASE ID: 0x003
PX4 git-hash: 029e0a7740d33bcdd92cdaa9b63f7748f5046871
PX4 version: 1.16.0 c0 (17825984)
PX4 git-branch: main
OS: NuttX
OS version: Release 12.9.0 (201916671)
OS git-hash: db3dc7b9e60b464c759d15f69fc3ee192aefa816
Build datetime: May 28 2025 17:28:34
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000900000000000000008292807c1b2b800e
MCU: i.MX RT1170 rB0, rev. 0
nsh> work_queue status

Work Queue: 13 threads                          RATE        INTERVAL
|__ 1) wq:rate_ctrl
|   |__ 1) control_allocator                789.7 Hz         1266 us
|   |__ 2) dshot                            695.3 Hz         1438 us
|   \__ 3) vehicle_angular_velocity         395.0 Hz         2532 us
|__ 2) wq:SPI0
|   \__ 1) i2c_launcher                       1.1 Hz       906660 us (1000000 us)
|__ 3) wq:SPI1
|   \__ 1) icm42686p                        393.8 Hz         2539 us
|__ 4) wq:SPI2
|   \__ 1) icm42688p                        400.9 Hz         2494 us
|__ 5) wq:SPI3
|   |__ 1) bmi088_accel                     399.1 Hz         2506 us (2500 us)
|   \__ 2) bmi088_gyro                      398.4 Hz         2510 us
|__ 6) wq:I2C0
|   \__ 1) i2c_launcher                       1.1 Hz       915990 us (1000000 us)
|__ 7) wq:I2C2
|   |__ 1) bmp390                            23.2 Hz        43118 us (43300 us)
|   \__ 2) ina226                             9.9 Hz       100681 us
|__ 8) wq:I2C3
|   |__ 1) bmm150                            20.4 Hz        49006 us (50000 us)
|   \__ 2) bmp388                            23.1 Hz        43250 us (43300 us)
|__ 9) wq:nav_and_controllers
|   |__ 1) airspeed_selector                 10.1 Hz        99168 us (100000 us)
|   |__ 2) fw_att_control                   197.4 Hz         5065 us
|   |__ 3) fw_lat_lon_control                98.7 Hz        10131 us
|   |__ 4) fw_mode_manager                   50.2 Hz        19910 us
|   |__ 5) fw_rate_control                  394.8 Hz         2533 us
|   |__ 6) land_detector                     92.6 Hz        10795 us
|   |__ 7) sensors                          197.3 Hz         5068 us
|   |__ 8) vehicle_acceleration             197.8 Hz         5055 us
|   |__ 9) vehicle_air_data                  23.3 Hz        42872 us
|   |__10) vehicle_gps_position               3.3 Hz       298953 us
|   \__11) vehicle_magnetometer              27.0 Hz        37082 us
|__ 10) wq:INS0
|   |__ 1) ekf2                             197.4 Hz         5066 us
|   |__ 2) vehicle_imu                      201.1 Hz         4973 us
|   |__ 3) vehicle_imu                      259.8 Hz         3850 us
|   \__ 4) vehicle_imu                      197.5 Hz         5062 us
|__ 11) wq:hp_default
|   |__ 1) battery_status                   100.1 Hz         9991 us
|   |__ 2) board_adc                         99.5 Hz        10053 us (10000 us)
|   |__ 3) fw_autotune_attitude_control     395.0 Hz         2532 us
|   |__ 4) manual_control                     5.0 Hz       199653 us
|   |__ 5) pwm_out                            3.5 Hz       282451 us (300000 us)
|   |__ 6) rc_update                          0.0 Hz            0 us
|   |__ 7) safety_button                     30.3 Hz        32996 us (33000 us)
|   \__ 8) tone_alarm                         2.3 Hz       432304 us
|__ 12) wq:ttyS4
|   \__ 1) rc_input                         250.1 Hz         3999 us (4000 us)
\__ 13) wq:lp_default
    |__ 1) cdcacm_autostart                   2.1 Hz       482079 us
    |__ 2) gyro_calibration                  50.0 Hz        20005 us (20000 us)
    |__ 3) load_mon                           2.1 Hz       483250 us (500000 us)
    |__ 4) mag_bias_estimator                50.2 Hz        19906 us (20000 us)
    |__ 5) parameters                         0.2 Hz      3686879 us
    \__ 6) send_event                        30.0 Hz        33319 us (33333 us)

```

mrpollo, 2025-05-31:
@sfuhrer can you please take a look here, it seems like something weird is going on.

sfuhrer, 2025-06-02:
@PetervdPerk-NXP can you check with https://github.com/PX4/PX4-Autopilot/pull/24955?

mrpollo, 2025-06-04:
@PetervdPerk-NXP @dk7xe the fix was just merged into main and v1.16, we will be publishing v1.16.0-rc2 shortly.

PR #24958 (2025-06-04): [backport] Control Allocation: only run on torque setpoint, not thrust
backport of #24955 fixes #24230
closed 2025-06-04 by PetervdPerk-NXP