PX4/PX4-Autopilot #24234: [Bug] Offboard no signal when using real drone

Dobryng, 2025-01-20:
### Describe the bug

I am a tertiary student doing a project with my team about ros2 and we wanted to use the Livox MID360 lidar on a UAV. Previously, we were using ROS 1 and Mavros to Offboard our UAV, but our supervisor decided that we should upgrade our UAV to be compatible with ROS2. After reading the ROS2-compatible PX4 packages, our team downloaded px4_ros_com with px4_msg packages and used the micro XRCE-DDS agent as a bridge to get the uORB topic into Ros2. We were able to simulate the drone in the gazebo simulator and we were able to code specific nodes for the simulation. Right now, we want to run the code for a real drone using a companion computer, in our case a Jetson Orin NX, and a Pixhawk 6C flight controller. But whenever we offboard, the QGC kept saying there wasn't any signal. It would be a great help if someone with experience in this field lend a helping hand. 

### Screenshot / Media

### Vehicle type

None

Jaeyoung-Lim, 2025-01-20:
@Dobryng In order to get help, you need to provide more information than your setup is not working.

Please provide flight logs, information of what you are doing, code snippets to explain what you are doing

Dobryng, 2025-01-21:
Basically, i tried doing offboard using the default takeoff code given you first do the gitclone. 

[image]

We connected the jetson to the px4 via UART connection and ran the micro agent with the uart connection command, which was successful

[image]

After doing some research, we realise that we needed local position data, so we use our Lidar odom and publish to the /fmu/in/vehicle_visual_odometry and was successful and we were able to run the listener command in the Marvlink console and get a feedback from px4 indicating that the publishing was a success. But the px4 odometry still uses the vehicle odometry, so I am not to sure how to proceed next