PX4/PX4-Autopilot #24239: [Bug][Compiling] Compiling failed due to the incompatibility of protobuf installed from /Tools/setup/ubuntu.sh in wsl2 of ubuntu22.04 on Win10

Kolt1911, 2025-01-22:
### Describe the bug

I installed and ran make the px4_sitl in wsl2 of ubuntu22.04 on Win10, by following the steps from the link [https://docs.px4.io/main/en/dev_setup/dev_env_windows_wsl.html](url). All is right until the make instruction.
There are many compiling errors like this:
```
In file included from /usr/include/gz/msgs10/gz/msgs/discovery.pb.h:12,
                 from /usr/include/gz/transport13/gz/transport/Discovery.hh:59,
                 from /usr/include/gz/transport13/gz/transport.hh:26,
                 from /home/admin/PX4-Autopilot/src/modules/simulation/gz_bridge/GZMixingInterfaceServo.hpp:38,
                 from /home/admin/PX4-Autopilot/src/modules/simulation/gz_bridge/GZMixingInterfaceServo.cpp:34:
/usr/include/gz/msgs10/gz/msgs/details/discovery.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
   17 | #error This file was generated by an older version of protoc which is
      |  ^~~~~
compilation terminated due to -Wfatal-errors
```
Seem like the version of protobuf. Then I checked the `/usr/include/gz/msgs10/gz/msgs/details/discovery.pb.h` and located the error statement. 
```
#include <google/protobuf/port_def.inc>
#if PROTOBUF_VERSION < 3012000
#error This file was generated by a newer version of protoc which is
#error incompatible with your Protocol Buffer headers. Please update
#error your headers.
#endif
#if 3012004 < PROTOBUF_MIN_PROTOC_VERSION
#error This file was generated by an older version of protoc which is
#error incompatible with your Protocol Buffer headers. Please
#error regenerate this file with a newer version of protoc.
#endif
```
Seem like the imported lib `google/protobuf/port_def.inc` includes the `PROTOBUF_VERSION` and `PROTOBUF_MIN_PROTOC_VERSION`. Then I searched this file to check if the `PROTOBUF_VERSION` and `PROTOBUF_MIN_PROTOC_VERSION` was correct. However, the value of these two defined are 3012004 and 3012000, respectively. The condition `3012004 < PROTOBUF_MIN_PROTOC_VERSION`  should not be satisfied and may not trigger the error. 
```
// Shared google3/opensource definitions. //////////////////////////////////////
#define PROTOBUF_VERSION 3012004
#define PROTOBUF_MIN_HEADER_VERSION_FOR_PROTOC 3012000
#define PROTOBUF_MIN_PROTOC_VERSION 3012000
#define PROTOBUF_VERSION_SUFFIX ""
```
And I have tried to regenerate these pb.h files with a current version of protoc but still had this error. 
Could someone give me some suggestions please?

### To Reproduce

Try to install and make the px4_sitl in wsl2 of ubuntu22.04 on Win10, by following the steps from the link [https://docs.px4.io/main/en/dev_setup/dev_env_windows_wsl.html](url).

### Expected behavior

There are many compiling errors about the older version of protobuf, as the same like the description in **Describe the bug**.

### Screenshot / Media

```
In file included from /usr/include/gz/msgs10/gz/msgs/discovery.pb.h:12,
                 from /usr/include/gz/transport13/gz/transport/Discovery.hh:59,
                 from /usr/include/gz/transport13/gz/transport.hh:26,
                 from /home/admin/PX4-Autopilot/src/modules/simulation/gz_bridge/GZMixingInterfaceServo.hpp:38,
                 from /home/admin/PX4-Autopilot/src/modules/simulation/gz_bridge/GZMixingInterfaceServo.cpp:34:
/usr/include/gz/msgs10/gz/msgs/details/discovery.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
   17 | #error This file was generated by an older version of protoc which is
      |  ^~~~~
compilation terminated due to -Wfatal-errors
```

### Software Version

px4 1.14.0

### Vehicle type

None

Kolt1911, 2025-01-22:
Well, I finally find the root cause: the compiling instruction import the protobuf v3.20.3 in anaconda in the Win10. There is an option like "-isystem /mnt/d/anaconda/Library/include". Change the dir name of anaconda in D disk temporarily, and make will be succcessful.

closed 2025-01-22 by Kolt1911

Jacob-Gunderson-GA, 2025-04-01:
Another way to fix this error is to disable the import of Windows environment variables in WSL like so:
`sudo nano /etc/wsl.conf`
Add these lines:
```
[interop]
appendWindowsPath = false
```