PX4/PX4-Autopilot #24273: Building for NuttX problem ( ninja: build stopped: subcommand failed)

AR-01-25, 2025-01-30:
### Describe the bug

Hello,

> Iam trying to build using px4 

> I followed the instructions here:

https://github.com/PX4/PX4-user_guide/blob/main/en/dev_setup/building_px4.md

> I am getting the error after running the above code.

ninja: build stopped: subcommand failed.
make: *** [Makefile:227: px4_fmu-v5_default] Error 1

> I got this error:

test@test-PC:~/PX4/PX4-Autopilot$ make px4_fmu-v5_default
[0/1] Re-running CMake...
-- PX4_GIT_TAG: v1.16.0-alpha2-276-g3b828e157a
CMake Warning (dev) at CMakeLists.txt:176 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- PX4 config file: /home/test/PX4/PX4-Autopilot/boards/px4/fmu-v5/default.px4board
-- PLATFORM nuttx
-- TOOLCHAIN arm-none-eabi
-- ARCHITECTURE cortex-m7
-- ROMFSROOT px4fmu_common
-- IO px4_io-v2_default
-- SERIAL_GPS1 /dev/ttyS0
-- SERIAL_TEL1 /dev/ttyS1
-- SERIAL_TEL2 /dev/ttyS2
-- SERIAL_TEL4 /dev/ttyS3
-- ROOT_PATH /fs/microsd
-- PARAM_FILE /fs/mtd_params
-- UAVCAN_INTERFACES 2
-- UAVCAN_TIMER_OVERRIDE 6
-- PX4 config: px4_fmu-v5_default
-- PX4 platform: nuttx
-- Enabling double FP precision hardware instructions
-- cmake build type: MinSizeRel
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- drivers/px4io: ROMFS including px4_io-v2_default
CMake Warning (dev) at src/drivers/uavcan/libdronecan/libuavcan/CMakeLists.txt:24 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10") 
-- Release build type: MinSizeRel
-- ROMFS: ROMFS/px4fmu_common
-- ROMFS:  Adding platforms/nuttx/init/stm32f7/rc.board_arch_defaults -> /etc/init.d/rc.board_arch_defaults
-- ROMFS:  Adding boards/px4/fmu-v5/init/rc.board_defaults -> /etc/init.d/rc.board_defaults
-- ROMFS:  Adding boards/px4/fmu-v5/init/rc.board_sensors -> /etc/init.d/rc.board_sensors
-- ROMFS:  Adding boards/px4/fmu-v5/extras/px4_fmu-v5_bootloader.bin -> /etc/extras/px4_fmu-v5_bootloader.bin
-- ROMFS:  Adding boards/px4/fmu-v5/extras/px4_io-v2_default.bin -> /etc/extras/px4_io-v2_default.bin
-- Configuring done (4.3s)
-- Generating done (0.5s)
-- Build files have been written to: /home/test/PX4/PX4-Autopilot/build/px4_fmu-v5_default
[2/1336] git submodule src/modules/uxrce_dds_client/Micro-XRCE-DDS-Client
[7/1336] git submodule platforms/nuttx/NuttX/nuttx
[8/1336] git submodule src/drivers/gps/devices
[15/1336] git submodule src/drivers/uavcan/libdronecan/dsdl
[... 73 lines cut ...]
                 from /home/test/PX4/PX4-Autopilot/src/modules/uxrce_dds_client/Micro-XRCE-DDS-Client/include/uxr/client/transport.h:24,
                 from /home/test/PX4/PX4-Autopilot/src/modules/uxrce_dds_client/Micro-XRCE-DDS-Client/include/uxr/client/profile/transport/serial/serial_transport.h:31,
                 from /home/test/PX4/PX4-Autopilot/src/modules/uxrce_dds_client/Micro-XRCE-DDS-Client/include/uxr/client/profile/transport/serial/serial_transport_platform.h:23,
                 from /home/test/PX4/PX4-Autopilot/src/modules/uxrce_dds_client/Micro-XRCE-DDS-Client/src/c/profile/transport/serial/serial_transport_posix.c:2:
/home/test/PX4/PX4-Autopilot/platforms/nuttx/NuttX/nuttx/include/sys/socket.h: In function '__cmsg_nxthdr':
/home/test/PX4/PX4-Autopilot/platforms/nuttx/NuttX/nuttx/include/sys/socket.h:342:11: warning: cast increases required alignment of target type [-Wcast-align]
  342 |   __ptr = (FAR struct cmsghdr *)
      |           ^
[26/29] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/session/session.c.obj
[27/29] Building C object CMakeFiles/microxrcedds_client.dir/src/c/core/serialization/xrce_types.c.obj
ninja: build stopped: subcommand failed.

-- stderr output is:

CMake Error at /home/test/PX4/PX4-Autopilot/build/px4_fmu-v5_default/src/modules/uxrce_dds_client/src/libmicroxrceddsclient_project-stamp/libmicroxrceddsclient_project-build-MinSizeRel.cmake:47 (message):
  Stopping after outputting logs.

[149/1336] Generating uORB topic ucdr headers
ninja: build stopped: subcommand failed.
make: *** [Makefile:227: px4_fmu-v5_default] Error 1
test@test-HP-280-G4-MT-Business-PC:~/PX4/PX4-Autopilot$ 

### Vehicle type

None

Officially-Siddhant, 2025-04-05:
Faced the same issue. Turns out, the cmake version is where the problem lies.

1.  Firstly, check your cmake version:

`cmake --version`
If it is not equal to `3.24` then here's what you can do (mind you, this can be lengthy but it is still a clean fix):
`sudo apt remove --purge cmake cmake-data`
`sudo apt autoremove`
`sudo rm /usr/local/bin/cmake`

2. Now, download the version files best compatible with PX4:

`wget https://github.com/Kitware/CMake/releases/download/v3.24.4/cmake-3.24.4.tar.gz`
`tar -zxvf cmake-3.24.4.tar.gz`
`cd cmake-3.24.4`

3. Build them:
`./bootstrap`
`make -j$(nproc)`

4. Install them:
` sudo make install`

6. Verify version again using:  'cmake --version'

If this prompts an error along the lines of No such file or directory, then that's because we removed the path earlier while fixing. Map the version you've downloaded using symlink for it to be accessed by Ubuntu globally using:
`sudo ln -s /usr/local/bin/cmake /usr/bin/cmake`

Check your version, and you should be good to go.

dakejahl, 2025-04-05:
3.22.1 is the official ubuntu 22.04 cmake version. I recently had issues with cmake and ended up reverting back to that. But purging it from your system will break the installation of other applications that depend on it. Not a big deal but something to be aware of, you might have to fix some other installs like ros2

closed 2025-04-05 by dakejahl

peterkappelt, 2025-04-11:
In case someone is encountering the same issue: I managed to fix this issue locally by patching the relevant sections, since I didn't feel like downgrading CMake on my system.

The relevant changes are in `src/modules/uxrce_dds_client/Micro-XRCE-DDS-Client/src/c/profile/transport/ip/udp/udp_transport_posix.c`:

```
diff --git a/src/c/profile/transport/ip/udp/udp_transport_posix.c b/src/c/profile/transport/ip/udp/udp_transport_posix.c
index 2801c23..746d8ac 100644
--- a/src/c/profile/transport/ip/udp/udp_transport_posix.c
+++ b/src/c/profile/transport/ip/udp/udp_transport_posix.c
@@ -43,7 +43,10 @@ bool uxr_init_udp_platform(
         }
         hints.ai_socktype = SOCK_DGRAM;

+#pragma GCC diagnostic push
+#pragma GCC diagnostic ignored "-Wimplicit-function-declaration"
         if (0 == getaddrinfo(ip, port, &hints, &result))
+#pragma GCC diagnostic pop
         {
             for (ptr = result; ptr != NULL; ptr = ptr->ai_next)
             {
@@ -54,7 +57,10 @@ bool uxr_init_udp_platform(
                     break;
                 }
             }
+#pragma GCC diagnostic push
+#pragma GCC diagnostic ignored "-Wimplicit-function-declaration"
             freeaddrinfo(result);
+#pragma GCC diagnostic pop
         }
     }
     return rv;
```

Take this patch with a grain of salt since it only disables the two warnings being treated as errors. I'm not sure though how the different CMake versions cause this problem (maybe some changed implicit defaults? Or wanted behaviour?)

tuupola, 2025-09-04:
Problem also exists with cmake 3.28.2 and 3.30.8. IMO this needs some real fix and not require developers to run specific version of cmake.

marko-castral, 2025-09-15:
I agree that downgrading of cmake is not a valid option.

I will also apply the patch for now but looking forward for fix in the future.

esmaeelE, 2025-11-01:
> In case someone is encountering the same issue: I managed to fix this issue locally by patching the relevant sections, since I didn't feel like downgrading CMake on my system.
>
> The relevant changes [...]

Applying these change manually will fix the same error on kalilinux 2025.3

tuupola, 2025-11-01:
Another workaround is to enable a feature such as `CONFIG_NETDB_HOSTFILE=y` which then automatically sets `CONFIG_LIBC_NETDB=y` and compiling then succeeds.

Or if you are not using ROS you could just disable the uxrce_dds_client module with:

`CONFIG_MODULES_UXRCE_DDS_CLIENT=n`