PX4/PX4-Autopilot #24279: Airspeed Sensor Failure on PX4 with STM32F743VIT6 During Flight, Causing Mode Switch to Manual.

sahil-enercomp, 2025-01-31:
### Describe the bug

When using a HolyBro MS4525DO Digital Airspeed Sensor with the Flamingo H7 flight controller (STM32F743VIT6) running PX4 firmware, the airspeed sensor intermittently fails during flight. This issue triggers a switch to Manual mode, making the vehicle unsafe for autonomous operation. The sensor works without issues when using ArduPilot firmware, indicating the problem is specific to PX4.

Error Messages in Logs:

-     [airspeed_selector] Airspeed sensor failure detected. Return to launch (RTL) is advised.
-     [health_and_arming_checks] Preflight Fail: Airspeed invalid.
-     [airspeed_selector] Airspeed sensor healthy, start using again (-1, 1).

### To Reproduce

Steps to reproduce the behavior:

- Power on the drone with Flamingo H7.
- Connect HolyBro MS4525DO airspeed sensor to I2C bus 4.
- Upload a mission using QGroundControl (QGC).
- Arm the drone and take off in Manual Mode or Mission Mode.
- Monitor QGC for airspeed sensor failure messages and unexpected mode switches to Manual Mode.

### Expected behavior

- The airspeed sensor should provide continuous, accurate readings throughout the flight without triggering any failure warnings.
- The vehicle should remain in Mission Mode or the user-selected mode unless manually overridden.
- PX4 should not trigger RTL or mark the airspeed as invalid unless there's an actual hardware failure.

### Screenshot / Media

[https://drive.google.com/file/d/1sBz55kQPmTCDc664naUQkbcuxILzafD-/view?usp=drive_link](url)
Video is showing failure but drone not goes to disarm because we set this parameter COM_DISARM_LAND = -1

### Flight Log

Log - 1
[https://review.px4.io/plot_app?log=305d9730-b6ab-44a8-bc72-bb7bd0c191bc](url)
Log - 2
[https://review.px4.io/plot_app?log=134363c1-cd59-4095-a3f9-e08ac2b4c5d0](url)

### Software Version

PX4 Firmware: v1.15.2

### Flight controller

Hardware: Flamingo H7 (STM32F743VIT6) Source: (Indian made flight controller tested with ardupilot and its working fine. Flight controller not available online, Purchased from: Local Distributor)

### Vehicle type

Fixed Wing

### How are the different components wired up (including port information)

Airspeed Sensor: HolyBro MS4525DO Digital Airspeed Sensor
-     Connected to I2C Bus 4
-     No external pull-up resistors added (using internal)
-     100nF capacitor not initially placed; added later with no impact on issue

### Additional context

- The airspeed sensor works flawlessly with ArduPilot firmware, pointing towards a PX4-specific issue.
- Increasing COM_DISARM_LAND parameter delayed auto-disarming but did not resolve airspeed sensor failures.
- Attempted solutions included modifying ASPD_SCALE_APPLY, disabling stall speed checks, and fine-tuning ASPD_FS_INNOV.
- I2C communication errors were observed, especially after arming and during full throttle, suggesting possible interference or driver handling issues in PX4.

sahil-enercomp, 2025-01-31:
In the above logs, the values of airspeed parameters during the airspeed sensor failure issue were:

ASPD_BETA_GATE - 1  
ASPD_BETA_NOISE - 0.15  
ASPD_DO_CHECKS - 7  
ASPD_FALLBACK_GW - 0  
ASPD_FS_INNOV - 5  
ASPD_FS_INTEG - 10  
ASPD_FS_T_START - 1  
ASPD_FS_T_STOP - 2  
ASPD_PRIMARY - 1  
ASPD_SCALE_1 - 1  
ASPD_SCALE_2 - 1  
ASPD_SCALE_3 - 1  
ASPD_SCALE_APPLY - 2  
ASPD_SCALE_NSD - 0.0001  
ASPD_TAS_GATE - 4  
ASPD_TAS_NOISE - 1.4  
ASPD_WERR_THR - 0.55  
ASPD_WIND_NSD - 0.1  

These were the parameter values when the airspeed sensor failure issue occurred.

Please advise:
    Do we need to modify these values?
    What is the correct parameter setup for Fixed-Wing/VTOL with the MS4525DO airspeed sensor to ensure that airspeed failure does not occur during setup and flight?"

dirksavage88, 2025-01-31:
Hello, the logs posted only show the aircraft in stabilized mode. Also a GPS is required for mission mode, whereas an airspeed sensor is _completely optional in px4 for FW auto modes_ (though it can be [helpful](https://docs.px4.io/main/en/advanced_config/tuning_the_ecl_ekf.html#airspeed)). 

 But it looks like your airspeed sensor is not causing an RTL (it is only warning and advising the user to RTL).

Perhaps @sfuhrer can help with the airspeed specific preflight checks

sahil-enercomp, 2025-02-01:
Hey @dirksavage88, thanks for your reply.

The above log is from indoor testing, where we were just observing the airspeed sensor values and behavior. We need the airspeed sensor values, but the airspeed sensor failure occurs during flight.

We also tested this in the field, and the same issue happened. Below are the details of our outdoor field testing:

    Log file: [Link](https://drive.google.com/file/d/1ttGrZQQsjJWgBo31pW3X4gkH2pYaj3XA/view?usp=drive_link)
    PX4 log review: [Link](https://review.px4.io/plot_app?log=33bd7917-222b-43a1-b033-bbf33f195675)
    VTOL flying video: [Link](https://drive.google.com/file/d/1uBx0EtVSe0xjvXFm67Z7kqZsrUpKCNkK/view?usp=drive_link)
    QGC video screen during flight: [Link](https://drive.google.com/file/d/1uWaXem5JYPMyntk_0EvysmncNr12zCHG/view?usp=drive_link)

@dirksavage88, @sfuhrer, if you can provide any support on this issue, it would be very helpful for us.

Thanks!

mrpollo, 2025-02-01:
Hey @sahil-enercomp I'm not familiar with this hardware, did you port this to PX4 in v1.15?

sahil-enercomp, 2025-02-01:
Hey @mrpollo , thanks for your response.

Yes, we ported this hardware to PX4 v1.15, including modifications to support our flight controller and airspeed sensor.
PX4 also supports the MS4525DO airspeed sensor, and we know its driver is already available in PX4.

sfuhrer, 2025-02-20:
The airspeed measurement times out (there is no new differential pressure measurement published for more than 1s). It's a [hard-coded limit of 1s.](https://github.com/PX4/PX4-Autopilot/blob/24da87db1218fe0e3270d8025e2851923139b0b9/src/modules/airspeed_selector/airspeed_selector_main.cpp#L419C3-L421C2)
[image]

Instead of changing the airspeed validator parameters I would focus on finding the root issue (sensor hardware, transmission lines, driver, ..)

closed 2025-05-22 by sahil-enercomp