PX4/PX4-Autopilot #24284: [Bug] ver1.15.2 doesn't let me arm when powered by battery

0i0i0i, 2025-02-01:
### Describe the bug

I have 1.15.2 running on a pixhawk2.4.8 clone fc. And I found that after I have setting everything up in the qgc software and it I can arm the drone when the usb cable of the fc is connected to pc no matter the battery is pluged or not. But if powered by just the battery, the fc blinks red led and don't let me arm it and beeps whenever I try to arm it. I tested the ardupilot firmware on the same hardware and it works fine. Also I built another drone with the same model (different batch and source) fc hardware, different escs, different motors and different gps, I got the same issue. 

### To Reproduce

use the 1.15.2, then choose the frame (generic X shape quad drone), then calibrate all the sensors, then set the priority of the mag0 to disable, since otherwise, it shows a 1X0 degree inconsistence error. then set everything just following the standard way. 

### Expected behavior

let me arm it

### Software Version

1.15.2

### Flight controller

pixhawk 2.4.8

### Vehicle type

Multicopter

bresch, 2025-02-03:
Hi, do you see an error message when trying to arm?

If you don't have telemetry and all what you see is "blinking red", you could set [SDLOG_MODE](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#SDLOG_MODE) to "from boot until shutdown", then re-do your test with the battery only, then re-connect the USB cable and pull the log to see the error message.

You can also get some info if you type `dmesg` in the mavlink console of QGC after plugging the USB cable again.

0i0i0i, 2025-02-03:
Thank you very much! Yes, I have ordered a telemetry and waiting for its arrival and will test your methods, too.